espressif/esp-zigbee-sdk #147: Can't set measurement att val in Carbon Dioxide Cluster with esp_zb_zcl_set_attribute_val

felipejfc, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-277-gc8243465e4

### esp-zigbee-lib version.

1.0.3

### esp-zboss-lib version.

1.0.2

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

I want to be able to set the value like:

`esp_zb_zcl_set_attribute_val(HA_AIRQ_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID, &co2_level, false);
`

### What is the actual behavior?

Value is not set to the value inside the pointer co2_level (this is a float_t). I tried to check the value with:

 float_t check= *((float_t*) esp_zb_zcl_get_attribute(HA_AIRQ_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID)->data_p);

but check returns 0.0... this happens even if I hardcode the "co2_level".

### Steps to reproduce.

This is the relevant parts of my code:

`
static void add_temphum_endpoint(esp_zb_attribute_list_t *esp_zb_basic_cluster, esp_zb_attribute_list_t *esp_zb_identify_cluster,
                               esp_zb_attribute_list_t *esp_zb_groups_cluster, esp_zb_attribute_list_t *esp_zb_scenes_cluster,
                               esp_zb_attribute_list_t *esp_zb_on_off_cluster, esp_zb_ep_list_t *esp_zb_ep_list){

    esp_zb_cluster_list_t *esp_zb_cluster_list_airqsensor = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list_airqsensor, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list_airqsensor, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list_airqsensor, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_attribute_list_t * esp_zb_temperature_measurement_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_measurement_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &temperature_int);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_measurement_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_ID, &temperature_int);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_measurement_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_ID, &temperature_int);
    esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list_airqsensor, esp_zb_temperature_measurement_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_attribute_list_t * esp_zb_relative_humidity_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT);
    esp_zb_humidity_meas_cluster_add_attr(esp_zb_relative_humidity_cluster, ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID, &humidity_int);
    esp_zb_humidity_meas_cluster_add_attr(esp_zb_relative_humidity_cluster, ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_MIN_VALUE_ID, &humidity_int);
    esp_zb_humidity_meas_cluster_add_attr(esp_zb_relative_humidity_cluster, ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_MAX_VALUE_ID, &humidity_int);
    esp_zb_cluster_list_add_humidity_meas_cluster(esp_zb_cluster_list_airqsensor, esp_zb_relative_humidity_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list_airqsensor, HA_TEMPHUM_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_METER_INTERFACE_DEVICE_ID);
}

static void add_airq_endpoint(esp_zb_attribute_list_t *esp_zb_basic_cluster, esp_zb_attribute_list_t *esp_zb_identify_cluster,
                               esp_zb_attribute_list_t *esp_zb_groups_cluster, esp_zb_attribute_list_t *esp_zb_scenes_cluster,
                               esp_zb_attribute_list_t *esp_zb_on_off_cluster, esp_zb_ep_list_t *esp_zb_ep_list){

[... 119 lines cut ...]

Both temperature and humidity are being successfully set and sent to zigbee coordinator, however, the carbon dioxide measurement is not being reported to the coordinator, and this line:

        float_t res = *((float_t*) esp_zb_zcl_get_attribute(HA_AIRQ_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID)->data_p);

returns 0, even while the "co2_level" set previously is != 0.

I can only send "initial values" to this attr, like, if I initialize it like this:

    co2_level = 0.000999f;
    esp_zb_carbon_dioxide_measurement_cluster_add_attr(esp_zb_carbon_dioxide_concentration_cluster, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID, &co2_level);

The right value is sent to coordinator upon boot, however, subsequent esp_zb_zcl_set_attribute_val will also produce no effect in changing the value.

### More Information.

Uppon settings values I have the following log lines:

I (10579) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 1026 cluster_role 1
I (10579) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 1026 cluster_role 1  
I (10589) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 1029 cluster_role 1
I (10589) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 1029 cluster_role 1
I (10609) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 1037 cluster_role 1 

for 1037 cluster (co2) the write attr hook is not being triggered, which I believe means that internally the value set is failing? what am I doing wrong?

felipejfc, 2023-11-15:
update, this "hack" works for setting the value, and then the reporting will also work:

`        *((float_t*) esp_zb_zcl_get_attribute(HA_AIRQ_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID)->data_p) = co2_level;
`

so there must really be some bug in the set function

xieqinan, 2023-11-21:
@felipejfc ,

The following suggestions may be helpful for you to address the issue.

Firstly, try printing the return value of `esp_zb_zcl_set_attribute_val()` to check if it is valid. Refer to [esp_zb_zcl_status_t](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L169) for possible error codes.

Secondly, I noticed that you are using the same value to set the max and min values of carbon_dioxide, which may lead to corner issues. Consider enlarging the allowed range for carbon_dioxide, and then use `esp_zb_zcl_set_attribute_val()` to set the desired value within the allowed range.
```
esp_zb_carbon_dioxide_measurement_cluster_add_attr(esp_zb_carbon_dioxide_concentration_cluster, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID, &co2_level);
esp_zb_carbon_dioxide_measurement_cluster_add_attr(esp_zb_carbon_dioxide_concentration_cluster, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MIN_MEASURED_VALUE_ID, &co2_level);
esp_zb_carbon_dioxide_measurement_cluster_add_attr(esp_zb_carbon_dioxide_concentration_cluster, ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MAX_MEASURED_VALUE_ID, &co2_level);
```

g-chevalier, 2023-12-01:
Hi @felipejfc and Zigbee SDK team,

I had the same problem, and I tried your solution *esp_zb_zcl_get_attribute(…) = value
It works partially: I mean the attribute is actually written in the stack and can be read manually from the coordinator by sending a read attribute command (using Z2M console for instance).
However, the automatic reporting is not working as it is working for instance with the temperature cluster I also implemented.
The esp_zb_zcl_set_attribute_val() function is probably managing some reporting stuff in addition to the write value itself…

For information, I checked the return value of esp_zb_zcl_set_attribute_val() and I got code=135 (which translates to "ERROR" with esp_err_to_name()).

Does anybody know a temporary (up to this bug is being fixed) work around for telling the stack that the value changed and should be reported?

For information, I had also the exact same problem with the PM2.5 cluster. I'll create a new bug for that.

felipejfc, 2023-12-02:
I confirm that following @xieqinan suggestion of properly setting min and max cluster values fixed the issue for me. It maybe worth documenting somewhere that unlike other clusters, there are validations based on min and max values set for this one. thanks!

g-chevalier, 2023-12-04:
Hi @felipejfc 

Thank you for your answer.

As explained into the xx issue, it worked.

However, for Carbon Dioxide Cluster, my code was:
```esp_zb_carbon_dioxide_measurement_cluster_cfg_t carbon_dioxide_measurement_cluster_cfg;
carbon_dioxide_measurement_cluster_cfg.measured_value = ESP_ZB_ZCL_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_UNKNOWN;
carbon_dioxide_measurement_cluster_cfg.min_measured_value = ESP_ZB_ZCL_CARBON_DIOXIDE_MEASUREMENT_MIN_MEASURED_VALUE_UNDEFINED;
carbon_dioxide_measurement_cluster_cfg.max_measured_value = ESP_ZB_ZCL_CARBON_DIOXIDE_MEASUREMENT_MAX_MEASURED_VALUE_UNDEFINED;
esp_zb_attribute_list_t *esp_zb_carbon_dioxide_measurement_cluster = esp_zb_carbon_dioxide_measurement_cluster_create(&carbon_dioxide_measurement_cluster_cfg);
esp_zb_cluster_list_add_carbon_dioxide_measurement_cluster(esp_zb_cluster_list, esp_zb_carbon_dioxide_measurement_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```
xxxx, what do you think about that? Shouldn't those constants prevent any check of the value itself when setting attribute val? For temperature, this is working correctly.
I set min and max to:
```temperature_meas_cluster_cfg.min_value =  0x8000;
temperature_meas_cluster_cfg.max_value =  0x8000;
```
I did not find UNKNOWN constants in .h for temperature, so I put 0x8000 as said in the standard § 4.4.2.2.1.2 (A MeasuredValue of 0x8000 indicates that the temperature measurement is unknown).