espressif/esp-zigbee-sdk #148: sleepy_end_device example crashes

blastrock, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### esp-zigbee-lib version.

~1.0.0

### esp-zboss-lib version.

~1.0.0

### Espressif SoC revision.

ESP32-C6-DevKitC-1

### What is the expected behavior?

Not crash

### What is the actual behavior?

The device somehow fails to connect and crashes with the following log:

```
I (526) main_task: Returned from app_main()
I (527) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (536) ESP_ZB_SLEEP: Zigbee stack initialized
I (542) ESP_ZB_SLEEP: Start network steering
I (9148) ESP_ZB_SLEEP: Zigbee can sleep
I (9538) ESP_ZB_SLEEP: Zigbee can sleep
I (9666) ESP_ZB_SLEEP: Zigbee can sleep
I (19175) ESP_ZB_SLEEP: ZDO signal: ZDO Leave (0x3), status: ESP_OK

abort() was called at PC 0x4201867d on core 0
0x4201867d: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x4080998a  SP      : 0x4087e160  GP      : 0x4080fbf4  
0x4080063a: panic_abort at /home/blastrock/esp/esp-idf/components/esp_system/panic.c:452

0x4080998a: __ubsan_include at /home/blastrock/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4086d9bc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4087e1c4  A0      : 0x4087e18c  A1      : 0x4087e1c2  
A2      : 0x00000000  A3      : 0x4087e1b9  A4      : 0x00000001  A5      : 0x40817000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x00000773  
S4      : 0x00000038  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
4087e160: 0x4087f214 0x40880000 0x4087e1c0 0x4080e2be 0x4206d0f0 0x4206ecb4 0x00000003 0x40810680
0x4080e2be: abort at /home/blastrock/esp/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4087e180: 0x4087e1c4 0x4081069c 0x4087e1c0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4087e1a0: 0x43502074 0x34783020 0x38313032 0x20643736 0x63206e6f 0x2065726f 0x00000030 0x42050000
0x42050000: esp_ota_get_running_partition at /home/blastrock/esp/esp-idf/components/app_update/esp_ota_ops.c:576

4087e1c0: 0x00000030 0x31303234 0x64373638 0x00000700 0x00000000 0x00000000 0x00000038 0x42018680
0x42018680: zb_globals_init at ??:?

4087e1e0: 0x00000000 0x00000000 0x40812050 0x42026d40 0x00000000 0x00000000 0x00000000 0x4201d2be
0x42026d40: zb_multimac_mlme_reset_request_proxy_send_req at nwk_multimac_proxy.c.obj:?

0x4201d2be: zb_sched_loop_iteration at ??:?

4087e200: 0x00000000 0x42026d2c 0x00000773 0x423b391e 0x0000003c 0x00000004 0x00000000 0x00000000
0x42026d2c: zb_multimac_debug_transaction_stuck_handler at nwk_multimac_proxy.c.obj:?

4087e220: 0x00000000 0x00000000 0x00000000 0x42039692 0x00000000 0x00000000 0x4087f1f4 0x4200f80e
0x42039692: zboss_main_loop_iteration at ??:?

0x4200f80e: esp_zb_main_loop_iteration at ??:?

4087e240: 0x00000000 0x00000000 0x00000000 0x4200af26 0x00000000 0x00000008 0x00000000 0x00000000
0x4200af26: esp_zb_task at /tmp/esp-zigbee-sdk/examples/esp_zigbee_sleep/sleepy_end_device/main/esp_zb_sleepy_end_device.c:149 (discriminator 2)

4087e260: 0x00000000 0x00000002 0x00060000 0x00000fa0 0x00000000 0x00000000 0x00000000 0x4080be26
0x4080be26: vPortTaskWrapper at /home/blastrock/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4087e280: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e2a0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087e090
4087e2c0: 0x000070c1 0x408110f0 0x408110f0 0x4087e2bc 0x408110e8 0x00000014 0x4087f2b4 0x4087f2b4
4087e2e0: 0x4087e2bc 0x00000000 0x00000005 0x4087d2b8 0x6267695a 0x6d5f6565 0x006e6961 0x00000000
4087e300: 0x00000000 0x4087e2b0 0x00000005 0x00000000 0x00000000 0x00000000 0x00000009 0x40818268
4087e320: 0x408182d0 0x40818338 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
4087e340: 0x4205ec1a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4205ec1a: _cleanup_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e360: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e380: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e400: 0x00000000 0x00000000 0x00000000 0x40000000 0x00000020 0x00000000 0x00000000 0x00000003
0x40000000: _start in ROM

4087e420: 0x420719bc 0x4081b090 0x00000001 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4087e438
4087e440: 0x00000000 0x00000000 0x00000000 0x4087e450 0xffffffff 0x4087e450 0x4087e450 0x00000000
4087e460: 0x4087e464 0xffffffff 0x4087e464 0x4087e464 0x00000001 0x00000001 0x00000000 0x0000ffff
4087e480: 0x00000000 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4087e490 0x00000000 0x00000000
4087e4a0: 0x00000000 0x4087e4a8 0xffffffff 0x4087e4a8 0x4087e4a8 0x00000000 0x4087e4bc 0xffffffff
4087e4c0: 0x4087e4bc 0x4087e4bc 0x00000001 0x00000001 0x00000000 0x4000ffff 0x00000000 0xb33fffff
0x4000ffff: ppRxFragmentProc in ROM

4087e4e0: 0x00000000 0x00000054 0x00000000 0x4087e4e8 0x00000000 0x00000000 0x00000000 0x4087e500
4087e500: 0xffffffff 0x4087e500 0x4087e500 0x00000000 0x4087e514 0xffffffff 0x4087e514 0x4087e514
4087e520: 0x00000001 0x00000001 0x00000000 0x1d00ffff 0x00000000 0xb33fffff 0x00000000 0x00000028
4087e540: 0xffff0000 0x00002000 0x70030000 0x4081a0e0 0x4081a0e0 0x4081a0e0 0x4081a0e0 0x00000000

ELF file SHA256: 58a8ea514d53f62c

Rebooting...
```

### Steps to reproduce.

1. Clone this repo, I'm using commit 9d016af06ad2af163b7fbc448455cc2fd52380fe from the current main branch
2. Open examples/esp_zigbee_sleep/sleepy_end_device in vscode
3. Set the target to ESP32-C6 and the tty to ttyUSB0, use the UART bridge
4. Erase the device's flash memory (just in case)
5. Build, flash and monitor
6. Set your zigbee network to allow new devices to connect

### More Information.

This is a regression. I didn't have this issue with esp-zboss-lib 0.6.2 and esp-zigbee-lib 0.9.5.

xieqinan, 2023-11-16:
Hello @blastrock @Bosemani ,

The ESP32C6 currently does not support the sleep mode, but support is anticipated within this month. Any updates regarding the support will be communicated to you. In the meantime, kindly consider using the ESP32H2 as an alternative.

Bosemani, 2023-11-16:
Hai @xieqinan 
 I'm using ESP32-H2 and [esp-zigbee-sdk-v1.0.3](https://github.com/espressif/esp-zigbee-sdk/commit/9d016af06ad2af163b7fbc448455cc2fd52380fe)
I haven't changed anything on code, Just Build the SLEEPY_END_DEVICE example. Erased the chip & loaded the program.

Still I'm facing this issue. Here is the Log
```c
ELF file SHA256: bc4e2adeca07391e

Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xc10
load:0x4083efd0,len:0x2da4
load:0x408460e8,len:0x177c
entry 0x4083cfd0
I (25) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (26) boot: compile time Nov 16 2023 12:16:10
I (27) boot: chip revision: v0.1
I (28) boot.esp32h2: SPI Speed      : 64MHz
I (33) boot.esp32h2: SPI Mode       : DIO
I (37) boot.esp32h2: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 000a2800
I (81) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (88) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=114f0h ( 70896) map
I (140) esp_image: segment 1: paddr=00021518 vaddr=40800000 size=06b00h ( 27392) load
I (156) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=67310h (422672) map
I (346) esp_image: segment 3: paddr=0008f338 vaddr=40806b00 size=0726ch ( 29292) load
I (363) esp_image: segment 4: paddr=000965ac vaddr=4080dd70 size=0154ch (  5452) load
I (368) esp_image: segment 5: paddr=00097b00 vaddr=50000000 size=00004h (     4) load
I (377) boot: Loaded app from partition at offset 0x10000
I (378) boot: Disabling RNG early entropy source...
I (392) cpu_start: Unicore app
I (393) cpu_start: Pro cpu up.
W (402) clk: esp_perip_clk_init() has not been implemented yet
I (409) cpu_start: Pro cpu start user code
I (409) cpu_start: cpu freq: 96000000 Hz
I (409) cpu_start: Application information:
I (412) cpu_start: Project name:     sleepy_end_device
I (417) cpu_start: App version:      1
I (422) cpu_start: Compile time:     Nov 16 2023 12:14:58
I (428) cpu_start: ELF file SHA256:  bc4e2adeca07391e...
I (434) cpu_start: ESP-IDF:          v5.1.1
I (439) cpu_start: Min chip rev:     v0.0
I (443) cpu_start: Max chip rev:     v0.99
I (448) cpu_start: Chip rev:         v0.1
I (453) heap_init: Initializing. RAM available for dynamic allocation:
I (460) heap_init: At 40815BF0 len 00037790 (221 KiB): D/IRAM
I (467) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (474) spi_flash: detected chip: generic
I (478) spi_flash: flash io: dio
W (482) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (495) sleep: Enable automatic switching of GPIO sleep configuration
I (503) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (511) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (519) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (526) sleep_sys_periph: HP System sleep retention initialization
I (533) sleep_sys_periph: TEE/APM sleep retention initialization
I (540) sleep_sys_periph: UART sleep retention initialization
I (546) sleep_sys_periph: Timer Group sleep retention initialization
I (553) sleep_sys_periph: IO Matrix sleep retention initialization
[... 42 lines cut ...]
4084ef30: 0x00000000 0x00000002 0x00060000 0x00000fa0 0x00000000 0x00000000 0x00000000 0x4080a936
4084ef50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084ef70: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4084ed60 0x000070ab
4084ef90: 0x4080f4cc 0x4080f4cc 0x4084ef88 0x4080f4c4 0x00000014 0x40817d84 0x40817d84 0x4084ef88
4084efb0: 0x00000000 0x00000005 0x4084df84 0x6267695a 0x6d5f6565 0x006e6961 0x00ffffff 0x00000000
4084efd0: 0x4084ef80 0x00000005 0x00000000 0x00000000 0x00000000 0x00000009 0x408165a4 0x4081660c
4084eff0: 0x40816674 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4205a484
4084f010: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f030: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f050: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f0b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4084f0d0: 0x00000000 0x00000000 0x40000000 0x00000020 0x00000000 0x00000000 0x00000003 0x420717f8
4084f0f0: 0x40819390 0x00000001 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4084f104 0x00000000
4084f110: 0x00000000 0x00000000 0x4084f11c 0xffffffff 0x4084f11c 0x4084f11c 0x00000000 0x4084f130
4084f130: 0xffffffff 0x4084f130 0x4084f130 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000
[... cut ...]
4084f170: 0x4084f174 0xffffffff 0x4084f174 0x4084f174 0x00000000 0x4084f188 0xffffffff 0x4084f188
4084f190: 0x4084f188 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff 0x00000000
4084f1b0: 0x00000020 0x01c3f8d9 0x00000000 0x000f4240 0x00000000 0x4205487a 0x00000000 0x00000000
4084f1d0: 0x40815a54 0x00000054 0x00000000 0x4084f1d8 0x00000000 0x00000000 0x00000000 0x4084f1f0
4084f1f0: 0xffffffff 0x4084f1f0 0x4084f1f0 0x00000000 0x4084f204 0xffffffff 0x4084f204 0x4084f204
4084f210: 0x00000001 0x00000001 0x00000000 0x3600ffff 0x00000000 0xb33fffff 0x00000000 0x00000028
```

nalves23, 2023-11-16:
I'm facing the same issue with the esp_HA-Costumized_light and esp_HA-Costumized_switch example.

I try it with other examples that I know that are working on the past and the same thing!

xieqinan, 2023-11-17:
@Bosemani,

I tested the examples [esp-zigbee-HA-switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) and [esp-zigbee-sleep](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device) from the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk), and there were no crashes. Could you please provide details about your test environment? Additionally, could you specify the device you expect the sleep device to join?

xieqinan, 2023-11-17:
@nalves23 ，

I also utilized the master branch of [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) to test the [esp_zigbee_customized_devices](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices), and it performed reliably. Could you please provide the crash log for your test in the example?

Bosemani, 2023-11-17:
> @Bosemani,
> 
> I tested the examples [esp-zigbee-HA-switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) and [esp-zigbee-sleep](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device) from the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk), and there were no crashes. Could you please provide details about your test environment? Additionally, could you specify the device you expect the sleep device to join?
> 
> I recommend updating the ESP-IDF version to v5.1.2 for testing, as it provides enhanced stability for sleep support.

Hai @xieqinan 
 After updating to ESP-IDF version v5.1.2. It is working fine without any Crash.
Thank you for your support.

nalves23, 2023-11-17:
@xieqinan ,

I update the esp-idf to last version 5.1.2, and the crash remains.

This are my specifications:
IDF version.
v5.1.2

esp-zigbee-lib version.
~1.0.3

esp-zboss-lib version.
~1.0.2

Espressif SoC revision.
ESP32-H2-Mini1

When the assert occurs, that is the reason:

I (482) spi_flash: flash io: dio
W (486) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (500) app_start: Starting scheduler on CPU0
I (505) main_task: Started on CPU0
I (505) main_task: Calling app_main()
I (515) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2

**assert failed: zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL)**
Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x408068dc  SP      : 0x4084f860  GP      : 0x4080cee0
0x4080063a: panic_abort at C:/Users/nalves/esp-idf-v5.1.2/components/esp_system/panic.c:452

Bosemani, 2023-11-18:
> @xieqinan ,
>
> I update the esp-idf to last version 5.1.2, and the crash remains.
>
> This are my specifications: IDF version. v5.1.2
>
> esp-zigbee-lib version. ~1.0.3
>
> esp-zboss-lib version. ~1.0.2 [...]

Hai @nalves23 

Its look like you made changes in partition table.

nalves23, 2023-11-20:
Hi @Bosemani 

No, I kept the partition configuration as it was in the example:

I kept the partition configuration as it was in the example:

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 650K,
zb_storage, data, fat,      0xb3000, 16K,
zb_fct,     data, fat,      0xb7000, 1K,

Regards.
Nalves

kelin6, 2023-11-21:
@nalves23 According to the log you provided, assert failed at `zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL).` There are only two possibilities: First, there is no `zb_storage space` in the partition table. Second, an error occurred during the search for zb_storage using `esp_partition_find_first`. Please refer to [sleepy_end_device](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/sleepy_end_device/partitions.csv ) ensure that this example can run.

If the partition table is configured correctly, you can try `idf.py fullclean` and recompile.

blastrock, 2023-12-03:
@chshu following here on your comment on #99 

I have just tested the light_sleep example from c46797a. I followed my reproduction steps from my first post in this issue. I have changed the idf_component.yml to:

```yml
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zigbee-lib:
    version: "main"
    path: "components/esp-zigbee-lib"
    git: "https://github.com/espressif/esp-zigbee-sdk"
  espressif/esp-zboss-lib:
    version: "master"
    git: "https://github.com/espressif/esp-zboss-lib"
  # Required IDF version
  idf:
    version: ">=5.0.0"
```

Now the device doesn't seem to crash, but every time it goes to sleep (before it evens joins the network?), it reboots after a couple seconds without reason:

```
I (17) boot: ESP-IDF v5.3-dev-741-g03414a1550 2nd stage bootloader
I (18) boot: compile time Dec  3 2023 11:16:46
I (19) boot: chip revision: v0.0
I (22) boot.esp32c6: SPI Speed      : 80MHz
I (27) boot.esp32c6: SPI Mode       : DIO
I (32) boot.esp32c6: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 000e1000
I (75) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (82) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=122c8h ( 74440) map
I (118) esp_image: segment 1: paddr=000222f0 vaddr=40800000 size=05d28h ( 23848) load
I (124) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=63aech (408300) map
I (208) esp_image: segment 3: paddr=0008bb14 vaddr=40805d28 size=0a9b4h ( 43444) load
I (219) esp_image: segment 4: paddr=000964d0 vaddr=408106e0 size=01e64h (  7780) load
I (221) esp_image: segment 5: paddr=0009833c vaddr=50000000 size=00058h (    88) load
I (228) boot: Loaded app from partition at offset 0x10000
I (230) boot: Disabling RNG early entropy source...
I (247) cpu_start: Unicore app
W (257) clk: esp_perip_clk_init() has not been implemented yet
I (264) cpu_start: Pro cpu start user code
I (264) cpu_start: cpu freq: 160000000 Hz
I (264) cpu_start: Application information:
I (267) cpu_start: Project name:     sleepy_end_device
I (273) cpu_start: App version:      c46797a-dirty
I (278) cpu_start: Compile time:     Dec  3 2023 11:16:38
I (284) cpu_start: ELF file SHA256:  9d8a62d8d...
I (289) cpu_start: ESP-IDF:          v5.3-dev-741-g03414a1550
I (296) cpu_start: Min chip rev:     v0.0
I (301) cpu_start: Max chip rev:     v0.99 
I (305) cpu_start: Chip rev:         v0.0
I (310) heap_init: Initializing. RAM available for dynamic allocation:
I (317) heap_init: At 40819990 len 00062C80 (395 KiB): RAM
I (323) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (330) heap_init: At 50000058 len 00003F90 (15 KiB): RTCRAM
I (337) spi_flash: detected chip: generic
I (341) spi_flash: flash io: dio
W (344) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (358) sleep: Enable automatic switching of GPIO sleep configuration
I (365) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (373) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (381) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (389) sleep_sys_periph: HP System sleep retention initialization
I (396) sleep_sys_periph: TEE/APM sleep retention initialization
I (402) sleep_sys_periph: UART sleep retention initialization
I (409) sleep_sys_periph: Timer Group sleep retention initialization
I (416) sleep_sys_periph: IO Matrix sleep retention initialization
I (423) sleep_sys_periph: SPI Mem sleep retention initialization
I (429) sleep_sys_periph: SysTimer sleep retention initialization
I (447) coexist: coex firmware version: a3b19de
I (447) coexist: coexist rom version 5b8dcfa
I (448) main_task: Started on CPU0
I (450) main_task: Calling app_main()
I (461) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 160, APB_MIN: 160, Light sleep: ENABLED
I (465) esp_zb_sleep: Enable ieee802154 zigbee light sleep, the wake up source is ESP timer
I (474) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (525) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (526) btbb_init: btbb sleep retention initialization
I (527) ieee802154: ieee802154 mac sleep retention initialization
I (545) main_task: Returned from app_main()
I (546) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (547) ESP_ZB_SLEEP: Zigbee stack initialized
I (554) ESP_ZB_SLEEP: Start network steering
I (9166) ESP_ZB_SLEEP: Zigbee can sleep
I (17) boot: ESP-IDF v5.3-dev-741-g03414a1550 2nd stage bootloader
...
```

It does this in a loop, it retries to join the network, goes to sleep, reboots.

blastrock, 2023-12-08:
I am testing that light sleep example you linked, this is not integrated in my project, I opened the example directly with vscode.

I have just tested again with commit 796eb80 and it seems to work! It connects and does not reboot. I see however that it wakes up quite often. I tried changing ED_KEEP_ALIVE to 40000 but it still wakes up at least once every 5s, some times multiple times per second. Is this some technical limitation of the zigbee protocol?

```
 (321973) ESP_ZB_SLEEP: Zigbee can sleep
I (327051) ESP_ZB_SLEEP: Zigbee can sleep
I (332129) ESP_ZB_SLEEP: Zigbee can sleep
I (335290) ESP_ZB_SLEEP: Zigbee can sleep
I (335787) ESP_ZB_SLEEP: Zigbee can sleep
I (335912) ESP_ZB_SLEEP: Zigbee can sleep
I (336029) ESP_ZB_SLEEP: Zigbee can sleep
I (336146) ESP_ZB_SLEEP: Zigbee can sleep
I (336295) ESP_ZB_SLEEP: Zigbee can sleep
I (336535) ESP_ZB_SLEEP: Zigbee can sleep
I (336882) ESP_ZB_SLEEP: Zigbee can sleep
I (337398) ESP_ZB_SLEEP: Zigbee can sleep
I (338160) ESP_ZB_SLEEP: Zigbee can sleep
I (339291) ESP_ZB_SLEEP: Zigbee can sleep
I (340975) ESP_ZB_SLEEP: Zigbee can sleep
```

kelin6, 2023-12-11:
@blastrock The configuration of the long poll interval through `zb_zdo_pim_set_long_poll_interval` should be done after the join process because during the join process, the long poll interval will be set to the default value ZB_PIM_default_Long_Poll_interval (5 seconds).

Therefore, it is recommended to call `zb_zdo_pim_set_long_poll_interval` after the connection is established. 

Please add the following code in the Zigbee sleep example:

```
#include "zboss_api.h"
...
...
...
case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());

                     zb_zdo_pim_set_long_poll_interval(ED_KEEP_ALIVE);

```

blastrock, 2023-12-23:
Thanks again for the help!

Just a note that may help other people reading: I noticed that when we set this interval to a value, say 40s, it actually does 40s cycles where:

- for 17s it wakes up every few hundred milliseconds/seconds
- for 23s it sleeps completely

The 17s seem to be non-compressible, but we can always set the interval very high so that it stays awake 17s and then sleeps 100s or more.

This is fine for my use, and this issue is fixed, I'm closing it.

closed 2023-12-23 by blastrock

kelin6, 2023-12-25:
@blastrock  Could you provide the environment to reproduce this issue and ?
 Based on the esp-zigbee-sdk version (796eb801), the described phenomenon did not occur.
 
 I have only change this:
 
 
[image]

> Thanks again for the help!
>
> Just a note that may help other people reading: I noticed that when we set this interval to a value, say 40s, it actually does 40s cycles where:
>
> * for 17s it wakes up [...]

blastrock, 2023-12-25:
I applied the same patch as you on the light sleep example, and reproduce the behavior:

```
I (3378) ESP_ZB_SLEEP: Joined network successfully (Extended PAN ID: 00:12:4b:00:25:8c:5c:86, PAN ID: 0x1a62, Channel:11, Short Address: 0xc654)
I (3420) ESP_ZB_SLEEP: Zigbee can sleep
I (3495) ESP_ZB_SLEEP: Zigbee can sleep
I (3885) ESP_ZB_SLEEP: Zigbee can sleep
I (3997) ESP_ZB_SLEEP: Zigbee can sleep
I (4123) ESP_ZB_SLEEP: Zigbee can sleep
I (4238) ESP_ZB_SLEEP: Zigbee can sleep
I (4340) ESP_ZB_SLEEP: Zigbee can sleep
I (4503) ESP_ZB_SLEEP: Zigbee can sleep
I (4742) ESP_ZB_SLEEP: Zigbee can sleep
I (5089) ESP_ZB_SLEEP: Zigbee can sleep
I (5604) ESP_ZB_SLEEP: Zigbee can sleep
I (6366) ESP_ZB_SLEEP: Zigbee can sleep
I (7496) ESP_ZB_SLEEP: Zigbee can sleep
I (8515) ESP_ZB_SLEEP: Zigbee can sleep
I (9185) ESP_ZB_SLEEP: Zigbee can sleep
I (9604) ESP_ZB_SLEEP: Zigbee can sleep
I (11702) ESP_ZB_SLEEP: Zigbee can sleep
I (14722) ESP_ZB_SLEEP: Zigbee can sleep
I (17741) ESP_ZB_SLEEP: Zigbee can sleep
I (19405) ESP_ZB_SLEEP: Zigbee can sleep
I (20766) ESP_ZB_SLEEP: Zigbee can sleep
I (44218) ESP_ZB_SLEEP: Zigbee can sleep
I (44714) ESP_ZB_SLEEP: Zigbee can sleep
I (44837) ESP_ZB_SLEEP: Zigbee can sleep
I (44954) ESP_ZB_SLEEP: Zigbee can sleep
I (45071) ESP_ZB_SLEEP: Zigbee can sleep
I (45233) ESP_ZB_SLEEP: Zigbee can sleep
I (45473) ESP_ZB_SLEEP: Zigbee can sleep
I (45820) ESP_ZB_SLEEP: Zigbee can sleep
I (46336) ESP_ZB_SLEEP: Zigbee can sleep
I (47098) ESP_ZB_SLEEP: Zigbee can sleep
I (48228) ESP_ZB_SLEEP: Zigbee can sleep
I (49912) ESP_ZB_SLEEP: Zigbee can sleep
I (52425) ESP_ZB_SLEEP: Zigbee can sleep
I (55445) ESP_ZB_SLEEP: Zigbee can sleep
I (58465) ESP_ZB_SLEEP: Zigbee can sleep
I (60113) ESP_ZB_SLEEP: Zigbee can sleep
I (61490) ESP_ZB_SLEEP: Zigbee can sleep
I (84942) ESP_ZB_SLEEP: Zigbee can sleep
```

The device wakes up every 40s (at 44000 and 84000) but wake up regularly for 17s each time. I'm using the example code from 926a2fa and esp-zboss and esp-zigbee 1.0.7. I'm using esp-idf master's branch from 3 weeks ago https://github.com/espressif/esp-idf/tree/f0607f6d3b4b3bfcbebdd9c580c6a034a183aad6 .

The device connects to my actual home network, which has a few sensors, lights and switches. The orchestrator is zigbee2mqtt.

kelin6, 2023-12-26:
@blastrock Thank you very much for the information you provided. Based on the environment you shared, I couldn't reproduce it using the HA_on_off_switch and light_sleep examples. Therefore, the difference should be related to the coordinator.  

Could you provide packet capture files when you reproduce this scenario? If possible, it would be best to provide them in a format that can be opened with Wireshark software. Thanks a lot.

Suxsem, 2024-01-10:
@kelin6 I'm experiencing the same exact issue of the first post.

Just to be clear, is ESP32-**C6** light sleep support in ESP-IDF **5.1.2** and current master zboss and zigbee sdk?

Suxsem, 2024-01-10:
I can answer my question... 5.1.2 does not supports esp32-c6 light sleep. Support starts with 5.2 (not yet stable). I'm cloning the 5.2 branch right now, I'm sure that's the problem.

kelin6, 2024-01-11:
@Suxsem The corresponding support for light sleep in versions based on esp-zigbee-sdk 1.0.7 is as follows:

V5.1.2 (commit: [482a8fb](https://github.com/espressif/esp-idf/commit/482a8fb2d78e3b58eb21b26da8a5bedf90623213)): ESP32H2 supports light sleep; **ESP32C6 does not support light sleep**.
V5.2-beta2 (commit: [b4a6123](https://github.com/espressif/esp-idf/commit/b4a612345be8fa70be8321a6a491baf813a74c4d)): ESP32H2 supports light sleep; ESP32C6 supports light sleep.

Suxsem, 2024-01-11:
@kelin6 regarding the power profiling image that we can find in the light sleep example, I suppose it is related to the esp32-h2.
Can we expect the C6 to have a similar power consumption? I'm interested particularly in the average current over at lest 30 seconds of working with a 5 seconds keep alive interval, exactly like shown in the image. Any chance you can power profile the C6 as well?

kelin6, 2024-01-11:
@Suxsem Yes, This is the power consumption of esp32h2 in light sleep working with a 4 seconds keep-alive interval. 
[ESP32H2-light-sleep-power-consumption.png](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/image/ESP32H2-light-sleep-power-consumption.png)

Compared to the esp32h2, the power consumption of esp32c6 in light sleep is slightly higher, as shown in the following figure.
Note: The following image shows my test of esp32c6 light sleep working with a 5 seconds keep-alive interval. 
[image]

> @kelin6 regarding the power profiling image that we can find in the light sleep example, I suppose it is related to the esp32-h2. Can we expect the C6 to have a similar power consumption? I'm [...]

Suxsem, 2024-01-11:
Much much appreciated, thank you!!!