espressif/esp-zigbee-sdk #149: On/Off switch device as control in Home Assistant

aaaaahaaaaa, 2023-11-19:
### Question

I realise that the following question might be a bit out of scope here as it relates to Home Assistant, but I'm a complete newbie when it comes to the ESP SDK, and I only have a single C6 device and therefore Home Assistant is currently the only tool I feel confortable with for debugging.

I'm trying to setup a simple on/off switch (not a light), and for it to appear as an actual switch control in Home Assistant. With the setup below, I can see the `light` (just for testing) showing up in HA as an actual **control**, but not the `switch`. I've tried a bunch of different things, but nothing works. Maybe I'm misunderstanding the concept of how a switch can be represented as an end  device control 🤷.

Help would be greatly appreciated.

```c
static void esp_zb_task(void* pvParameters)
{
    esp_zb_init(&zb_device_config);

    esp_zb_on_off_switch_cfg_t switch_config = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_cluster_list_t* switch_cluster_list = esp_zb_on_off_switch_clusters_create(&switch_config);

    esp_zb_on_off_light_cfg_t light_config = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_cluster_list_t* light_cluster_list = esp_zb_on_off_light_clusters_create(&light_config);

    esp_zb_ep_list_t* endpoint_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(endpoint_list, switch_cluster_list, 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
    esp_zb_ep_list_add_ep(endpoint_list, light_cluster_list, 2, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);

    esp_zb_device_register(endpoint_list);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

[image]

valdisz, 2023-11-22:
@aaaaahaaaaa I have an even worse situation - my device is not connecting to the network at all. Can you share the `esp_zb_app_signal_handler` code that you are using?

aaaaahaaaaa, 2023-11-22:
@valdisz The rest of my code is identical to the [HA_on_off_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) example. Without changing the example at all, it does connect to my existing network.

aaaaahaaaaa, 2023-11-22:
@xieqinan Any chance you could shade some light with regards to my question above? All the endpoints I'm testing are showing up as expected, except the default on/off switch. I don't understand why.

```c
    esp_zb_on_off_switch_cfg_t switch_config = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_cluster_list_t* switch_cluster_list = esp_zb_on_off_switch_clusters_create(&switch_config);

    esp_zb_on_off_light_cfg_t light_config = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_cluster_list_t* light_cluster_list = esp_zb_on_off_light_clusters_create(&light_config);

    esp_zb_door_lock_cfg_t door_lock_config = ESP_ZB_DEFAULT_DOOR_LOCK_CONFIG();
    esp_zb_cluster_list_t* door_lock_cluster_list = esp_zb_door_lock_clusters_create(&door_lock_config);

    esp_zb_shade_cfg_t shade_config = ESP_ZB_DEFAULT_SHADE_CONFIG();
    esp_zb_cluster_list_t* shade_cluster_list = esp_zb_shade_clusters_create(&shade_config);

    esp_zb_temperature_sensor_cfg_t temperature_sensor_config = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();
    esp_zb_cluster_list_t* temperature_sensor_cluster_list = esp_zb_temperature_sensor_clusters_create(&temperature_sensor_config);

    esp_zb_ep_list_t* endpoint_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(endpoint_list, switch_cluster_list, 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
    esp_zb_ep_list_add_ep(endpoint_list, light_cluster_list, 2, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
    esp_zb_ep_list_add_ep(endpoint_list, door_lock_cluster_list, 3, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_DOOR_LOCK_DEVICE_ID);
    esp_zb_ep_list_add_ep(endpoint_list, shade_cluster_list, 4, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_SHADE_DEVICE_ID);
    esp_zb_ep_list_add_ep(endpoint_list, temperature_sensor_cluster_list, 5, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_TEMPERATURE_SENSOR_DEVICE_ID);
```

[image]

aaaaahaaaaa, 2023-11-22:
I updated to `1.0.4` and idf `5.1.2` and the issue is resolved. The switch appears as an end device as expected.

closed 2023-11-22 by aaaaahaaaaa