espressif/esp-zigbee-sdk #151: esp_zb_core_action_handler_register(zb_action_handler) doesn't work in the recent sdk version

Metatorn, 2023-11-21:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### esp-zigbee-lib version.

1.0.3

### esp-zboss-lib version.

1.0.3

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I want to use the action handler, obviously is needed to every command request in a Zigbee project.

If I send a command request, set, get attributes, an OTA notification, so on, the system must be able to handle every trigger and act in consecuence.

### What is the actual behavior?

The callback function never reach. I call the function esp_zb_core_action_handler_register(zb_action_handler) expecting the trigger of the callback but it never happens, I tested the program in older Zigbee sdk versions and works perfectly. 

I've updated the libraries and this behavior start to appear averytime since 1.0.0 versions and earlier.

Don't know if I'm doing something wrong but as I said, in older versions the sdk works fine.

### Steps to reproduce.

1. Call the esp_zb_core_action_handler_register(zb_action_handler) function.
2. Make any request for server to client
3. Wait for the callback execution, never happens

xieqinan, 2023-11-22:
Hello,

It may have been caused by a failure in the pipeline yesterday. The esp-zigbee-lib has been updated to v1.0.4. Please update it and try again.

Metatorn, 2023-11-22:
Mmm no, same thing, I have updated the Zigbee lib to 1.0.4 and Zboss lib to 1.0.3 and the callback never reach, If I use the version 1.0.0 works fine.

I created a new project and copy the OTA example and doesn't work, the rest of examples avoid the action handler so I cannot validate if they work

xieqinan, 2023-11-29:
Hello @Metatorn,

- The `esp_zb_core_action_handler_register()` is used in many examples in the esp-zigbee-sdk, such as [esp_zb_light](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c#L117). You can refer to these examples for its usage.

- The `.bin` file is specific to the esp32h2 chip. If you flash it into an esp32c6 without any modifications, you'll encounter errors like the following:
  ```
  I (13415) ESP_OTA_CLIENT: -- OTA Client receives data: progress [8/159616]
  E (13415) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x6d)
  E (13415) ESP_OTA_CLIENT: zb_ota_upgrade_status_handler(83): Failed to write OTA data to partition, status: ESP_ERR_OTA_VALIDATE_FAILED
  I (13435) ESP_OTA_CLIENT: OTA status: 4
  ```

- To update the firmware, only need to replace it with the `esp32c6 firmware` and please refer to the OTA [README.md](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota/ota_server).

- If you want to verify whether `esp_zb_core_action_handler_register()` works or not, replace the esp32c6 with the esp32h2 and try again. The correct log should look like(the test was done in  Nov 29 2023):
  ```
  I (390) ESP_OTA_CLIENT: OTA example 1.0 is running
  I (420) phy: phy_version: 200,0, 1cef4f4, May 22 2023, 11:57:13
  I (430) main_task: Returned from app_main()
  I (540) ESP_OTA_CLIENT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
  I (11890) ESP_OTA_CLIENT: Network steering was not successful (status: ESP_FAIL)
  I (27900) ESP_OTA_CLIENT: -- OTA upgrade start
  I (28030) ESP_OTA_CLIENT: Joined network successfully (Extended PAN ID: 60:55:f9:ff:fe:f7:2d:d8, PAN ID: 0x242b, Channel:13, Short Address: 0xe5ad)
  I (28180) ESP_OTA_CLIENT: -- OTA Client receives data: progress [8/159616]
  I (28470) ESP_OTA_CLIENT: -- OTA Client receives data: progress [72/159616]
  I (28980) ESP_OTA_CLIENT: -- OTA Client receives data: progress [136/159616]
  I (30010) ESP_OTA_CLIENT: -- OTA Client receives data: progress [200/159616]
  I (31030) ESP_OTA_CLIENT: -- OTA Client receives data: progress [264/159616]
  I (32050) ESP_OTA_CLIENT: -- OTA Client receives data: progress [328/159616]
  ```

Metatorn, 2023-11-30:
> Hello @Metatorn,
>
> * The `esp_zb_core_action_handler_register()` is used in many examples in the esp-zigbee-sdk, such as [...]

Thanks for your recommendations, I validate that the problem was a bad use of non static local variables into additional functions that call the attributes when you begin to configure the clusters.

I have now the OTA working but I cannot trigger the request for the Electrical Measurement Clusters, I'm working on that.

I'm going to close the issue

closed 2023-11-30 by Metatorn