espressif/esp-zigbee-sdk #152: Working of Lock state attribute in Door lock cluster.

mallaprashant, 2023-11-22:
### Question

I was going through the ZCL documentation and noticed that the Lock State attribute is read-only. According to the docs, the Lock State attribute changes only when the door lock gets a lock/unlock command and subsequently executing the corresponding physical lock/unlock operation. 

I am seeking clarification on the mechanism through which the Lock State attribute determines the status of the physical lock, specifically how it figures out whether the lock is in a locked or unlocked state. I am interested in understanding the inherent relationship facilitating this determination and the methods by which synchronization between the Lock State attribute and the physical lock is established.

Your help in explaining this would be really helpful.

xieqinan, 2023-11-22:
Hello @mallaprashant ,

Regarding this issue, I believe the Zigbee report mechanism can meet your requirements.

Firstly, the `esp_zb_zcl_lock_door_cmd_req()` / `esp_zb_zcl_unlock_door_cmd_req()` provides a way to lock or unlock the door for the `Door Lock Controller`.

Secondly, the report attribute of `lock state` and `door state` provides a way to report their physical state to the `Door Lock Controller` by calling `esp_zb_zcl_set_attribute_val()` when it executes the corresponding physical lock/unlock operation.

Furthermore, it is necessary to ultilize `esp_zb_zcl_config_report_cmd_req()` and `esp_zb_zdo_device_bind_req()` to bind the attribute report.

mallaprashant, 2023-11-24:
Hi @xieqinan,

Thank you for your response. I have conducted tests based on the instructions provided, specifically utilizing the `esp_zb_zcl_set_attribute_val()` function. However, the function call resulted in the return value:

`ESP_ZB_ZCL_STATUS_READ_ONLY = 0x88U`,

indicating that the operation is constrained by a read-only status. Your further guidance on how to address this limitation would be greatly appreciated.

> Secondly, the report attribute of lock state and door state provides a way to report their physical state to the Door Lock Controller by calling esp_zb_zcl_set_attribute_val() when it executes the corresponding physical lock/unlock operation.

aaaaahaaaaa, 2023-11-25:
@xieqinan I'm getting the same return value (`ESP_ZB_ZCL_STATUS_READ_ONLY`) as @mallaprashant when trying to calling `esp_zb_zcl_set_attribute_val` on `ESP_ZB_ZCL_ATTR_DOOR_LOCK_LOCK_STATE_ID`.

Furthermore, I'm wondering how to configure the default lock state value upon initialization of the door lock cluster. The following config doesn't change the state to `locked` when the device joins the network. It remains `unlocked`.
```c
esp_zb_door_lock_cluster_cfg_t door_lock_config;
door_lock_config.lock_state = 0x01; // 0x01: Locked, 0x02 :Unlocked
door_lock_config.lock_type = 0x02; // 0x02: Other
door_lock_config.actuator_enabled = ESP_ZB_ZCL_ATTR_DOOR_LOCK_ACTUATOR_ENABLED_DEFAULT_VALUE;
esp_zb_attribute_list_t* door_lock_cluster = esp_zb_door_lock_cluster_create(&door_lock_config);
```

xieqinan, 2023-11-27:
@mallaprashant  @aaaaahaaaaa ,

Please use the mandatory way to set the attribute value.

Such as:
```
 esp_zb_zcl_set_attribute_val(endpoint, cluster_id, cluster_role, attr_id, false);
```

mallaprashant, 2023-11-27:
Thank you @xieqinan,

Previously, I set the 'check access' method to true.

closed 2023-11-27 by mallaprashant

aaaaahaaaaa, 2023-11-27:
@xieqinan What about the default value upon initialization?

aaaaahaaaaa, 2023-11-27:
@xieqinan I'm confused, who do you refer to by "the user"?

The config example above [I shared above](https://github.com/espressif/esp-zigbee-sdk/issues/152#issuecomment-1826397917) doesn't work for the lock. The default `lock_state ` value doesn't change in HA when the device joins the network for the first time. But it does for the default `on_off` of a switch.

aaaaahaaaaa, 2023-11-27:
@xieqinan Thank you for the clarification. So if I understand correctly, we can actually expect the following setting to  eventually work once it's supported by the SDK?
```c
esp_zb_door_lock_cluster_cfg_t door_lock_config;
door_lock_config.lock_state = 0x01; // 0x01: Locked, 0x02 :Unlocked
esp_zb_attribute_list_t* door_lock_cluster = esp_zb_door_lock_cluster_create(&door_lock_config);
```

xieqinan, 2023-11-28:
@aaaaahaaaaa ,

Yes, you can try it now.