espressif/esp-zigbee-sdk #157: `zb_assert()` is useless in a core dump

nomis, 2023-11-25:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### esp-zigbee-lib version.

1.0.0

### esp-zboss-lib version.

1.0.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Core dumps that occur because of a `zb_assert()` include either the original PC or a filename and line number.

### What is the actual behavior?

Core dumps capture the PC of `zb_assert()` which is of no use for debugging.

### Steps to reproduce.

1. Enable writing of core dumps to flash.
2. Introduce a deliberate `zb_assert()` failure.
3. Check that the location of the assertion appears in the code dump that was written to flash.

### More Information.

PC 0x42021767 is just the location of the `zb_assert` function, so I know there has been crash but have no idea why or where it is:

```
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x40828d2c, name: 'zigbee_main', GDB name: 'process 1082297644'

================== CURRENT THREAD REGISTERS ===================
ra             0x40807d4a	0x40807d4a <prvCopyDataToQueue>
sp             0x40828b30	0x40828b30
gp             0x4080f9a0	0x4080f9a0 <__c.4+48>
tp             0x40813980	0x40813980 <s_coredump_stack+1008>
t0             0x37363534	926299444
t1             0x7271706f	1920036975
t2             0x33323130	858927408
fp             0x40828b5c	0x40828b5c
s1             0x40828b5c	1082297180
a0             0x40828b68	1082297192
a1             0x40828b4a	1082297162
a2             0x0	0
a3             0x40828b95	1082297237
a4             0x1	1
a5             0x4081c000	1082245120
a6             0x0	0
a7             0x76757473	1987409011
s2             0xbb	187
s3             0x1	1
s4             0x1	1
s5             0xbb	187
s6             0x0	0
s7             0x0	0
s8             0x0	0
s9             0x0	0
s10            0x0	0
s11            0x0	0
t3             0x6e6d6c6b	1852664939
t4             0x6a696867	1785292903
t5             0x66656463	1717920867
t6             0x62613938	1650538808
pc             0x40800668	0x40800668 <panic_abort+20>

==================== CURRENT THREAD STACK =====================
#0  0x40800668 in panic_abort (details=0x40828b68 "abort() was called at PC 0x42021767 on core 0") at /home/simon/src/esp-idf/components/esp_system/panic.c:452
#1  0x40807d4a in __ubsan_include () at /home/simon/src/esp-idf/components/esp_system/ubsan.c:313
Backtrace stopped: frame did not save the PC

======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1082297644 0x40800668 in panic_abort (details=0x40828b68 "abort() was called at PC 0x42021767 on core 0") at /home/simon/src/esp-idf/components/esp_system/panic.c:452
  2    process 1082264448 esp_gpio_reserve_pins (mask=4648282267607748692) at /home/simon/src/esp-idf/components/esp_hw_support/esp_gpio_reserve.c:16
  3    process 1082313732 0x4080b044 in xTaskGenericNotify (xTaskToNotify=0x4087dba4, uxIndexToNotify=<optimized out>, ulValue=0, eAction=<optimized out>, pulPreviousNotificationValue=<optimized out>) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:6029
  4    process 1082302604 0x4080b0b6 in xTaskGenericNotify (xTaskToNotify=0x0, uxIndexToNotify=1082302608, ulValue=<optimized out>, eAction=<optimized out>, pulPreviousNotificationValue=0x600c5000) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5963
  5    process 1082251068 0x4080b044 in xTaskGenericNotify (xTaskToNotify=0x4081c000 <s_rx_frame+12408>, uxIndexToNotify=<optimized out>, ulValue=1082245120, eAction=<optimized out>, pulPreviousNotificationValue=<optimized out>) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:6029
  6    process 1082307144 0x4080b0b6 in xTaskGenericNotify (xTaskToNotify=0x0, uxIndexToNotify=1082307148, ulValue=<optimized out>, eAction=<optimized out>, pulPreviousNotificationValue=0x600c5000) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5963

       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x40828d2c      zigbee_main      5/5         592/7592
0x40820b80             IDLE      0/0         208/1316
0x4082cc04          ui_uart      1/1         464/5664
0x4082a08c      device_main    19/19         336/3752
0x4081d73c        esp_timer    22/22         240/3840
0x4082b248          ui_main      2/2         336/3756

==================== THREAD 1 (TCB: 0x40828d2c, name: 'zigbee_main') =====================
#0  0x40800668 in panic_abort (details=0x40828b68 "abort() was called at PC 0x42021767 on core 0") at /home/simon/src/esp-idf/components/esp_system/panic.c:452
#1  0x40807d4a in __ubsan_include () at /home/simon/src/esp-idf/components/esp_system/ubsan.c:313
Backtrace stopped: frame did not save the PC

==================== THREAD 2 (TCB: 0x40820b80, name: 'IDLE') =====================
#0  esp_gpio_reserve_pins (mask=4648282267607748692) at /home/simon/src/esp-idf/components/esp_hw_support/esp_gpio_reserve.c:16
#1  0x40809272 in vListInsert (pxList=0x4081d054, pxNewListItem=0x40820418) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:180
Backtrace stopped: frame did not save the PC

==================== THREAD 3 (TCB: 0x4082cc04, name: 'ui_uart') =====================
#0  0x4080b044 in xTaskGenericNotify (xTaskToNotify=0x4087dba4, uxIndexToNotify=<optimized out>, ulValue=0, eAction=<optimized out>, pulPreviousNotificationValue=<optimized out>) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:6029
#1  0x00000008 in ?? ()

==================== THREAD 4 (TCB: 0x4082a08c, name: 'device_main') =====================
#0  0x4080b0b6 in xTaskGenericNotify (xTaskToNotify=0x0, uxIndexToNotify=1082302608, ulValue=<optimized out>, eAction=<optimized out>, pulPreviousNotificationValue=0x600c5000) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5963
#1  0xffffffff in ?? ()

==================== THREAD 5 (TCB: 0x4081d73c, name: 'esp_timer') =====================
#0  0x4080b044 in xTaskGenericNotify (xTaskToNotify=0x4081c000 <s_rx_frame+12408>, uxIndexToNotify=<optimized out>, ulValue=1082245120, eAction=<optimized out>, pulPreviousNotificationValue=<optimized out>) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:6029
#1  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC

==================== THREAD 6 (TCB: 0x4082b248, name: 'ui_main') =====================
#0  0x4080b0b6 in xTaskGenericNotify (xTaskToNotify=0x0, uxIndexToNotify=1082307148, ulValue=<optimized out>, eAction=<optimized out>, pulPreviousNotificationValue=0x600c5000) at /home/simon/src/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5963
#1  0xffffffff in ?? ()

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x50000000 0x0 RW  
.rtc.force_fast 0x50000000 0x0 RW  
.rtc_noinit 0x50000000 0x0 RW  
.rtc.force_slow 0x50000000 0x0 RW  
.iram0.text 0x40800000 0xf06c R XA
.iram0.text_end 0x4080f06c 0x0 RW  
.iram0.bss 0x4080f070 0x0 RW  
.dram0.data 0x4080f070 0x3768 RW A
.flash.text 0x42000020 0x6ffec R XA
.flash.appdesc 0x42080020 0x100 R  A
.flash.rodata 0x42080120 0x153a0 RW A
.eh_frame 0x420954c0 0xf70 R  A
.flash.rodata_noload 0x42096430 0x0 RW  
.dram0.heap_start 0x4081c830 0x0 RW  
.coredump.tasks.data 0x40828d2c 0x160 RW 
.coredump.tasks.data 0x40828a90 0x290 RW 
.coredump.tasks.data 0x40820b80 0x160 RW 
.coredump.tasks.data 0x40820aa0 0xd0 RW 
.coredump.tasks.data 0x4082cc04 0x160 RW 
.coredump.tasks.data 0x4082ca20 0x1d0 RW 
.coredump.tasks.data 0x4082a08c 0x160 RW 
.coredump.tasks.data 0x40829f30 0x150 RW 
.coredump.tasks.data 0x4081d73c 0x160 RW 
.coredump.tasks.data 0x4081efe0 0xf0 RW 
.coredump.tasks.data 0x4082b248 0x160 RW 
.coredump.tasks.data 0x4082b0f0 0x150 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
```

nomis, 2023-11-26:
Even with the UART, by default the function has no output and all you get is the address of `zb_assert`:
```
I (66634) nutt.ZigbeeDevice: Leave network
E (66834) nutt.ZigbeeDevice: Device rejoin

abort() was called at PC 0x42023b6b on core 0
Core  0 register dump:
MEPC    : 0x408004e4  RA      : 0x408085d2  SP      : 0x4082c9d0  GP      : 0x40811050  
TP      : 0x40814c5c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4082c9fc  S1      : 0x4082c9fc  A0      : 0x4082ca08  A1      : 0x4082c9ea  
A2      : 0x00000000  A3      : 0x4082ca35  A4      : 0x00000001  A5      : 0x4081d000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x000010ff  
S4      : 0x00000038  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  
```

The `abort()` function works better than this and should be used directly instead.

nomis, 2024-02-07:
Do you mean you've fixed it? There have been no recent commits mentioning "assert".

This has nothing to do with any tools for accessing core dumps or console output. The string `abort() was called at PC 0x42021767 on core 0` is embedded in the core dump itself. Without a longer stack trace the caller of `zb_assert()` is lost.

nomis, 2024-02-25:
This kind of output doesn't retain any information about where it crashed:
```
==================== THREAD 1 (TCB: 0x4082ca6c, name: 'zigbee_main') =====================
#0  panic_abort (details=details@entry=0x4082c898 "abort() was called at PC 0x42023c7f on core 0") at /home/simon/src/esp-idf/components/esp_system/panic.c:472
#1  0x408085d2 in esp_system_abort (details=details@entry=0x4082c898 "abort() was called at PC 0x42023c7f on core 0") at /home/simon/src/esp-idf/components/esp_system/port/esp_system_chip.c:93
#2  0x4080f666 in abort () at /home/simon/src/esp-idf/components/newlib/abort.c:38
#3  0x42023c82 in ?? ()
```

```
42023c76 <zb_assert>:
42023c76:       1141                    addi    sp,sp,-16
42023c78:       c606                    sw      ra,12(sp)
42023c7a:       fe7ec097                auipc   ra,0xfe7ec
42023c7e:       96e080e7                jalr    -1682(ra) # 4080f5e8 <abort>
```

Version 1.0.5 was running for 88.7 days before it crashed, and then crashed again 3h40m later.

I see no evidence that this has been fixed because `zb_assert()` is still a function when it needs to be a macro that calls `abort()`:
```
ZB_NORETURN void zb_assert(const zb_char_t *file_name, zb_int_t line_number);
```

nomis, 2024-02-27:
I've tested this with 1.1.2 by calling `zb_assert()` and nothing has changed. It still indirectly calls `abort()` so the location is not captured by the core dump.

kelin6, 2024-03-01:
@nomis 
* In the upcoming 1.1.3 release, there will be support for printing the specific assert location, for example, like this:

[image]

* Additionally, it should be noted that the backtrace interpretation using idf monitor may not be accurate. You can try using other terminal tools such as cutecom to collect logs. The logs will output information like the following example. 

[image]

* Afterward, you can use `riscv32-esp-elf-addr2line` to try resolving the addresses printed above.

[image]

* Alternatively, you can use `riscv32-esp-elf-objdump -S ./on_off_light_bulb.elf > debug.S` to generate a static analysis file, and then search for the function corresponding to the addresses in the ` debug.S` file.

nomis, 2024-03-01:
>     * In the upcoming 1.1.3 release, there will be support for printing the specific assert location, for example, like this:

Not acceptable because it requires capturing all the UART output.

>     * Additionally, it should be noted that the backtrace interpretation using idf monitor may not be accurate. You can try using other terminal tools such as cutecom to collect logs. The logs will output information like the following example.

Not acceptable because it requires capturing all the UART output.

> Another crucial point is that, in order to reduce the size of the compiled libraries

Then you need to distribute libraries with debugging information so that the full backtrace is available, or make it all open source like other providers of the ZBOSS libraries do.

chshu, 2024-07-26:
There are some enhancements recently:
- some fixes in the tracing logging feature, it can output desired logs stably
- release separate debugg version libs with more logs, and produt lib for production
- mode docs for debugging purpose: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging

If you encounter any other unreasonable asserts, feel free file new issues with debugging infomation.

closed 2024-07-26 by chshu

nomis, 2024-07-26:
More logging is not a fix. Does useful information make it into the core dump or not?

chshu, 2024-07-26:
If it's not an assert, the trace logging would help.
If it's an assert in Zboss stack, since its source code is not public, user need to provide the core dump info to us for analysis [assertion-failures](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#assertion-failures)

nomis, 2024-07-26:
That documentation states that the log output at the time of the core dump must be provided, i.e. the saved core dump itself is not sufficient.

I've provided core dumps in other issues where you then ask for the log output from it.