espressif/esp-zigbee-sdk #159: Fail to send attribute report (`esp_zb_zcl_report_attr_cmd_req` crashes)

aaaaahaaaaa, 2023-11-25:
### Question

I have a simple on/off switch connected to Home Assistant via a gateway (I haven't implemented any client, it's the default HA Zigbee integration). I'm trying to manually set the on/off attribute of a switch and report back the change to Home Assistant. Setting the attribute works but the report via `esp_zb_zcl_report_attr_cmd_req` crashes. I'm guessing the report cmd payload is incorrect, I'm unsure how to configure it.

```c
esp_zb_zcl_status_t status = esp_zb_zcl_set_attribute_val(
    DEVICE_ENDPOINT,
    ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
    &switch_state,
    false
);

if (status == ESP_ZB_ZCL_STATUS_SUCCESS) {
    ESP_LOGI(TAG, "Set attribute successful. Sending config report...");
    esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
    report_attr_cmd.zcl_basic_cmd.src_endpoint = DEVICE_ENDPOINT;
    report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
    report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
    report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
    esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
}
```

Logs:
```
I (16565) ESP_ZB_DOOR_BUZZER: Set attribute successful. Sending config report...

abort() was called at PC 0x4201590f on core 0
0x4201590f: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x40807c12  SP      : 0x4087e030  GP      : 0x4080dd50  
0x4080063a: panic_abort at /Users/g/Dev/esp/esp-idf/components/esp_system/panic.c:452

0x40807c12: __ubsan_include at /Users/g/Dev/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4086f1cc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4087e094  A0      : 0x4087e05c  A1      : 0x4087e092  
A2      : 0x00000000  A3      : 0x4087e089  A4      : 0x00000001  A5      : 0x40815000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000006  S3      : 0x00000087  
S4      : 0x00000036  S5      : 0x4087e84c  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
4087e030: 0x00000000 0x4087e088 0x4087e090 0x4080c41e 0x7b422af5 0x00000000 0x40817710 0x4080e614
0x4080c41e: abort at /Users/g/Dev/esp/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4087e050: 0x4087e094 0x4080e630 0x4087e090 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4087e070: 0x43502074 0x34783020 0x35313032 0x20663039 0x63206e6f 0x2065726f 0x00000030 0x40870000
4087e090: 0x42060030 0x31303234 0x66303935 0x4205f800 0x408156ec 0x0000ffff 0x00000000 0x42015912
0x4205f800: esp_log_writev at /Users/g/Dev/esp/esp-idf/components/log/log.c:191

0x42015912: zb_globals_init at ??:?

4087e0b0: 0x00000006 0x0000ffff 0x00000000 0x4202c728 0x00000000 0x00000000 0x00000000 0x0000000b
0x4202c728: zb_zcl_get_attr_desc_a at ??:?

4087e0d0: 0x4087e108 0x4087e17c 0x42063000 0x4200e352 0x4200d38a 0x4087e17c 0x00000036 0x0000000b
0x4200e352: esp_zb_zcl_report_attr_cmd_req at ??:?

0x4200d38a: zcl_device_attribute_update_handler at esp_zigbee_core.c.obj:?

4087e0f0: 0x4200d38a 0x4087e17c 0x42063000 0x4200799c 0x4200d38a 0x4087e17c 0x00000000 0x42007a16
0x4200d38a: zcl_device_attribute_update_handler at esp_zigbee_core.c.obj:?

0x4200799c: unlock_door at /Users/g/Dev/esp/zigbee-door-buzzer/main/main.c:140

0x4200d38a: zcl_device_attribute_update_handler at esp_zigbee_core.c.obj:?

0x42007a16: zb_attribute_handler at /Users/g/Dev/esp/zigbee-door-buzzer/main/main.c:147 (discriminator 13)

4087e110: 0x00004d08 0x00000000 0x40870006 0x00000000 0x42063624 0x0000004d 0x4087e17c 0x42007aec
0x42007aec: zb_attribute_handler at /Users/g/Dev/esp/zigbee-door-buzzer/main/main.c:156 (discriminator 1)

4087e130: 0x00000001 0x40000001 0x00000000 0x000034f3 0x42063624 0x00000177 0x00000000 0x42007b6a
0x40000001: _start in ROM

0x42007b6a: zb_action_handler at /Users/g/Dev/esp/zigbee-door-buzzer/main/main.c:215

4087e150: 0x40810818 0x4087ebde 0x00000040 0x4200d5e0 0x40810818 0x0000000b 0x00000000 0x4202f6ec
0x4200d5e0: zcl_device_attribute_update_handler at esp_zigbee_core.c.obj:?

0x4202f6ec: zb_zcl_handle at ??:?

4087e170: 0x0000000b 0x40810818 0x00000036 0x00000000 0x0006d34d 0x00100000 0x0c010000 0x42408108
4087e190: 0x0000004d 0x0000000b 0x408107fc 0x4200963a 0x0000004d 0x00000001 0x408107fc 0x42030d5c
0x4200963a: zb_zcl_device_callback at esp_zigbee_core.c.obj:?

0x42030d5c: zb_zcl_call_on_off_attr_device_cb at zcl_on_off_commands.c.obj:?

4087e1b0: 0x00000000 0x00000000 0x0000000b 0x00000372 0x00000000 0x00000000 0x0000000b 0x42030fac
0x42030fac: zb_zcl_on_off_invoke_user_app at ??:?

4087e1d0: 0x00000000 0x00000000 0x00000000 0x01e3a900 0x0600004d 0x01010400 0x0000cb00 0x01000000
4087e1f0: 0x00000000 0x00000000 0x00000001 0x4201a560 0x00000000 0x42030f68 0x000b0000 0x42393600
0x4201a560: zb_sched_loop_iteration at ??:?

0x42030f68: zb_zcl_on_off_invoke_user_app at ??:?

4087e210: 0x0000003c 0x00000004 0x4087e898 0x4087e8e4 0x4087e9c0 0x4087ebf8 0x4087ea0c 0x4203691e
0x4203691e: zboss_main_loop_iteration at ??:?

4087e230: 0x00000000 0x00000000 0x4087ed0c 0x4200d800 0x4087e9c0 0x4087ebf8 0x4087ea0c 0x4200757a
0x4200d800: esp_zb_main_loop_iteration at ??:?

0x4200757a: esp_zb_task at /Users/g/Dev/esp/zigbee-door-buzzer/main/main.c:276 (discriminator 2)

4087e250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e270: 0x00000000 0x00000000 0x00000000 0x40809fa6 0x00000000 0x00000000 0x00000000 0x00000000
0x40809fa6: vPortTaskWrapper at /Users/g/Dev/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4087e290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4087e2b0: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087e060 0x00000653 0x4080f0b0 0x4080f0b0 0x4087e2bc
4087e2d0: 0x4080f0a8 0x00000014 0x4087edcc 0x4087edcc 0x4087e2bc 0x00000000 0x00000005 0x4087d2b8
4087e2f0: 0x6267695a 0x6d5f6565 0x006e6961 0x00000000 0x00000000 0x4087e2b0 0x00000005 0x00000000
4087e310: 0x00000000 0x00000000 0x00000009 0x40816218 0x40816280 0x408162e8 0x00000000 0x00000000
4087e330: 0x00000001 0x00000000 0x00000000 0x00000000 0x4205cb48 0x00000000 0x00000000 0x00000000
0x4205cb48: _cleanup_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4087e350: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e390: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
0x40000000: _start in ROM

4087e410: 0x00000060 0x00000000 0x00000101 0x00000024 0x00000000 0x00000000 0x4087e478 0xb33fffff

ELF file SHA256: 3c956d7a676baa79

Rebooting...
```

nomis, 2023-11-26:
You should only be using `esp_zb_zcl_set_attribute_val()`. It will automatically report the change using the configured reporting interval.

xieqinan, 2023-11-27:
@aaaaahaaaaa ,

If you intend to send the `report attribute` to the HA, you need to use `esp_zb_zdo_device_bind_req()` to bind the HA's address to the device that needs to report. Additionally, utilize `esp_zb_zcl_config_report_cmd_req()` to configure the report. The attribute will be reported automatically if it changes on the device.

aaaaahaaaaa, 2023-11-28:
@xieqinan I'm sure it's a very simple thing but I sadly can't get it to work. I followed the `customized_switch` example to try to implement the bind via `esp_zb_zdo_device_bind_req`, but I feel like the use case is different so I'm lost. In the example, the binding happens from the client side and the server is discovered via `esp_zb_zdo_find_on_off_light`.

In my case, I'm trying to report back the change from the server to the client, which was created automatically by Home Assistant when joining the network. Specifically, the functionality i'm trying to reproduce is the following: 
1. switch is toggled in HA.
2. device handles the action, wait 3 sec, toggle back the switch programatically.
3. status change is reflected in HA.

A short example on how to implement `esp_zb_zdo_device_bind_req` from the server side would be an immense help if that's not too much to ask.

aaaaahaaaaa, 2023-11-28:
What I tried last is the code below which results in the following timeout:

```
I (33585) ESP_ZIGBEE_API_ZDO: ZDO binding request timeout!
W (33585) ESP_ZIGBEE_API_ZDO: The tsn of callback is not found
W (33585) ESP_ZIGBEE_API_ZDO: The tsn of callback is not found and can't remove
```

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "string.h"
#include "driver/gpio.h"
#include "esp_zigbee_core.h"
#include "ha/esp_zigbee_ha_standard.h"

#define DEVICE_ENDPOINT 1
#define BUZZER_GPIO 4
#define BUZZ_DURATION 3000

char model_id[] = { 7, 'E', 'S', 'P', '3', '2', 'C', '6' };
char manufacturer_name[] = { 17, 'b', 'l', 'a', 'c', 'k', '_', 'c', 'a', 't', 's', '_', 's', 't', 'u', 'd', 'i', 'o' };

static const char* TAG = "ESP_ZB_buzzer";
static bool switch_state = false;

static esp_zb_platform_config_t zb_platform_config = {
    .radio_config.radio_mode = RADIO_MODE_NATIVE,
    .host_config.host_connection_mode = HOST_CONNECTION_MODE_NONE,
};

static esp_zb_cfg_t zb_device_config = {
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,
    .install_code_policy = false,
    .nwk_cfg.zed_cfg = {
        .ed_timeout = ESP_ZB_ED_AGING_TIMEOUT_64MIN,
        .keep_alive = 3000,
    },
};

typedef struct device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} device_params_t;

device_params_t remote_device_params;

void buzz() {
    ESP_LOGI(TAG, "Buzzing for %d ms...", BUZZ_DURATION);
    gpio_set_level(BUZZER_GPIO, 1);
    vTaskDelay(BUZZ_DURATION / portTICK_PERIOD_MS);
    gpio_set_level(BUZZER_GPIO, 0);

    ESP_LOGI(TAG, "Setting switch state back to false");
    esp_zb_zcl_set_attribute_val(
        DEVICE_ENDPOINT,
        ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
        &switch_state,
        false
    );
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void* user_ctx) {
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
[... 169 lines cut ...]
        }
        else {
            ESP_LOGI(TAG, "Network steering was not successful");
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
    default:
        ESP_LOGI(TAG, "Unhandled signal: %s (%s)", esp_zb_zdo_signal_to_string(signal_type), esp_err_to_name(error_status));
        break;

    }
}

void buzzer_driver_init() {
    gpio_reset_pin(BUZZER_GPIO);
    gpio_set_direction(BUZZER_GPIO, GPIO_MODE_OUTPUT);
}

void app_main(void) {
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&zb_platform_config));
    buzzer_driver_init();
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}
```

xieqinan, 2023-11-30:
@aaaaahaaaaa ,

A lot of commands and functions may be confusing for you; we will try to provide a more convenient way for the esp-zigbee-sdk.

Regarding your issue, the bind request should look like:

```c
        esp_zb_zdo_bind_req_param_t bind_req;
        esp_zb_ieee_addr_t ieee_addr;
        esp_zb_get_long_address(ieee_addr);
        memcpy(&(bind_req.src_address), ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.src_endp = DEVICE_ENDPOINT;
        bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        bind_req.dst_address_u.addr_long = HA_IEEE_ADDRESS; // use the `esp_zb_zdo_ieee_addr_req()` to get it.
        bind_req.dst_endp = HA_ENDPOINT; // use the `esp_zb_zdo_active_ep_req()` to get it.
        bind_req.req_dst_addr = HA_SHORT_ADDRESS; // if the HA is coordinator, it is 0x0000 
        esp_zb_zdo_device_bind_req(&bind_req, bind_cb, NULL);
```

nomis, 2023-12-03:
If you're using HA, for newly created devices it will already automatically bind to the on/off (switch) endpoint and configure it to report any changes.

I don't think the end device should try to do this by itself, and if you remove the `esp_zb_zdo_ieee_addr_req()`, `esp_zb_zcl_config_report_cmd_req()` it should just work.

You can use https://github.com/mdeweerd/zha-toolkit#binds_get-get-binding-table-from-the-device to read the binding table, and https://github.com/mdeweerd/zha-toolkit#conf_report_read-read-configured-reporting to read the reporting configuration.

aaaaahaaaaa, 2023-12-04:
@nomis That's not the case AFAICT unfortunately. If I set the attribute using `esp_zb_zcl_set_attribute_val`, change doesn't reflect in HA automatically.

nomis, 2023-12-04:
> I was starting to suspect that this might be the issue. But you're saying this scenario is working as well on the branch you're mentioning or just the binding in general?

No, in my example I am using `esp_zb_scheduler_alarm()`, probably because changing it didn't work from the callback.

I have taken the approach that the only safe place to call the Zigbee functions is from the Zigbee thread, so I use `esp_zb_scheduler_alarm()` to do that from other threads. I also don't change any attribute values from attribute callbacks.

(Your code is sleeping for 3 seconds inside the Zigbee thread, which may cause you other problems.)

aaaaahaaaaa, 2024-01-01:
Sorry for the late feedback. I just tested with `esp_zb_scheduler_alarm`, it works perfectly. The issue was definitely because I was setting the attribute from the main thread when handling the command.

closed 2024-01-01 by aaaaahaaaaa