espressif/esp-zigbee-sdk #160: random crashes during Zigbee operations.

vritzka, 2023-11-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### esp-zigbee-lib version.

1.0.5

### esp-zboss-lib version.

1.0.5

### Espressif SoC revision.

ESP32-S3 + ESP32-C6

### What is the expected behavior?

I expected the system to not crash, or at least find out the reason why it crashes.

### What is the actual behavior?

My system crashes at different, unexpected times.

I am reading from a temperature sensor.  You can see in the console below that it reads the value of 25.65 deg C.

But then suddenly it crashes.

The crashes happen at different times. 

Here is the serial output during a crash:

```
I (38732) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (38762) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (38792) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (38812) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (38832) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (38852) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (41342) sensorTask.c: Sensor 0x3673
I (41352) general.c: Fanspeed: 0%
I (42482) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2565)
I (46342) sensorTask.c: Sensor 0x3673
I (46352) general.c: Fanspeed: 0%
I (49952) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(2568)

abort() was called at PC 0x42050b92 on core 0
0x42050b92: zb_assert at ??:?

Backtrace: 0x40375cda:0x3fcc7f40 0x4037e43d:0x3fcc7f60 0x40385be6:0x3fcc7f80 0x42050b92:0x3fcc7ff0 0x42050517:0x3fcc8010 0x4204f944:0x3fcc8030 0x4206d05e:0x3fcc8050 0x4206d40c:0x3fcc8090 0x42054e2a:0x3fcc80c0 0x4206d7f3:0x3fcc80f0 0x42049c2b:0x3fcc8110 0x4200d579:0x3fcc8130 0x40380e8d:0x3fcc8170
0x40375cda: panic_abort at /Users/volkerritzka/esp/esp-idf/components/esp_system/panic.c:452

0x4037e43d: esp_system_abort at /Users/volkerritzka/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40385be6: abort at /Users/volkerritzka/esp/esp-idf/components/newlib/abort.c:38

0x42050b92: zb_assert at ??:?

0x42050517: zb_bufpool_storage_free at ??:?

0x4204f944: zb_buf_free_func at ??:?

0x4206d05e: zb_af_handle_apsde_data_confirm at af_rx.c.obj:?

[... 397 lines cut ...]
extern void vCreateZigbeeTask(void)
{
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

```

I am running this code in a separate RTOS task every 5 seconds to read the temperature:

```
           	        esp_zb_zcl_read_attr_cmd_t cmd_req;
			cmd_req.zcl_basic_cmd.dst_endpoint = 1;
			cmd_req.zcl_basic_cmd.src_endpoint = HA_THERMOMETER_ENDPOINT	;
			cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
			cmd_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
			uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID};
			cmd_req.attr_number = sizeof(attributes) / sizeof(uint16_t);;
			cmd_req.attr_field = attributes;
			cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = my_tent.thermometer_short_addr;                                                         
			esp_zb_zcl_read_attr_cmd_req( &cmd_req); 
```

### Steps to reproduce.

You could clone and flash my app from https://github.com/vritzka/TomatoTent/tree/v2

felipejfc, 2023-12-12:
same happens here with esp32-h2

```
abort() was called at PC 0x4201db99 on core 0
0x4201db99: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x408085c0  SP      : 0x4084f080  GP      : 0x4080e7b4
0x4080063a: panic_abort at C:/Users/fjfca/esp/esp-idf/components/esp_system/panic.c:452

0x408085c0: __ubsan_include at C:/Users/fjfca/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4083c644  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x4084f0e4  A0      : 0x4084f0ac  A1      : 0x4084f0e2
A2      : 0x00000000  A3      : 0x4084f0d9  A4      : 0x00000001  A5      : 0x40816000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x40814410  S3      : 0x00000008
S4      : 0x0000003a  S5      : 0x4084f8c4  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4084f080: 0x40819850 0x408197d8 0x4084f0e0 0x4080ce8c 0x00000000 0x4084f0e8 0x00000008 0x4080f00c
0x4080ce8c: abort at C:/Users/fjfca/esp/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4084f0a0: 0x4084f0e4 0x4080f028 0x4084f0e0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4084f0c0: 0x43502074 0x34783020 0x64313032 0x20393962 0x63206e6f 0x2065726f 0x00000030 0x00000000
4084f0e0: 0x40810030 0x31303234 0x39396264 0x4202d000 0x00000021 0x00000000 0x00000019 0x4201db9c
0x4202d000: zb_nwk_neighbor_ext_to_base at ??:?

0x4201db9c: zb_globals_init at ??:?

4084f100: 0x00000000 0x00000000 0x00000019 0x420298ac 0x00000001 0x00000019 0x408137ad 0x08028090
0x420298ac: remove_parent_from_potential_parents at ??:?

4084f120: 0x00000019 0x40811910 0x00000000 0x00000334 0x00000000 0x00000000 0x00000019 0x420281e0
0x420281e0: state_rejoin_handle_scan_confirm at nwk_discovery.c.obj:?

4084f140: 0x00000000 0x00000001 0x00000019 0x420284b2 0x00000001 0x00000001 0x00000001 0x42022836
0x420284b2: zb_mlme_scan_confirm at ??:?

0x42022836: zb_sched_loop_iteration at ??:?

4084f160: 0x00000000 0x4202845e 0x00190000 0x423b3a19 0x00000000 0x00000000 0x4084f7e8 0x4084f7c0
0x4202845e: zb_mlme_scan_confirm at ??:?

4084f180: 0x4084f798 0x4084f734 0x4084f910 0x420460d4 0x00000010 0x00000004 0x4081c31c 0x42015a56
0x420460d4: zboss_main_loop_iteration at ??:?

0x42015a56: esp_zb_main_loop_iteration at ??:?

4084f1a0: 0x4084f798 0x4084f734 0x4084f910 0x4200a244 0x00000000 0x00000002 0x00000000 0x00000006
```

chshu, 2023-12-19:
The current Zboss **macsplit** based zigbee gateway+rcp solution is an experimental feature, we do see some stability issues recently.
 
We are working on a new zigbee gateway solution which will base on the [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp), some stress tests are on going, will push to github in 2024 Q1.

chshu, 2024-02-04:
@vritzka The [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) based Zigbee Gateway example is available now: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway

Note:
- Please use the esp-idf master branch ([Commit ad3e1e3](https://github.com/espressif/esp-idf/commit/ad3e1e3daf7037e58c650c735e1f6218ee886651) or later). It will be supported in the coming IDF v5.1.3 and later release versions.
- Enable `OPENTHREAD_NCP_VENDOR_HOOK` option before building [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) example.

Feel free to try out the new feature, and let us know if any questions or suggestions.

closed 2024-02-04 by chshu