espressif/esp-zigbee-sdk #163: esp_zigbee_sleep\deep_sleep example Crashes and not able to join Network.

Bosemani, 2023-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### esp-zigbee-lib version.

V.1.0.5

### esp-zboss-lib version.

V.1.0.5

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

**As per Readme File**

I (414) sleep: Configure to isolate all GPIO pins in sleep state
I (420) sleep: Enable automatic switching of GPIO sleep configuration
I (427) main_task: Started on CPU0
I (427) main_task: Calling app_main()
I (437) ESP_ZB_DEEP_SLEEP: Not a deep sleep reset
I (437) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 20s

I (447) phy: phy_version: 211,0, 5857fe5, Nov  1 2023, 11:31:09
I (447) phy: libbtbb version: ce629d6, Nov  1 2023, 11:31:19
I (457) main_task: Returned from app_main()
I (587) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (587) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (587) ESP_ZB_DEEP_SLEEP: Start network steering
I (10557) ESP_ZB_DEEP_SLEEP: Joined network successfully (Extended PAN ID: 60:55:f9:ff:fe:f7:2f:3e, PAN ID: 0xe4c4, Channel:13, Short Address: 0x5e1c)
I (10567) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep
I (15567) ESP_ZB_DEEP_SLEEP: Enter deep sleep

### What is the actual behavior?

On the First Boot Crash was not Happening but Not Joining the network Here is the Log
I'm using Zigbee2mqtt as a Coordinator. In V1.0.5 I'm Facing this issue.

```c
I (12153) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (14883) ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 10
I (14883) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (14893) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (15923) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (16453) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xf455 on channel 11, dev 0x0, iface_id 0
I (18823) ZBOSS: mac/mac_associate.c:448   received association response
I (18833) ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xa3fd, iface_id 0, ch 11, rejoin 0
I (18833) ZBOSS: zdo/zdo_app_join.c:773   xpanid 8b:b5:d8:bc:da:21:56:ce
I (28893) ZBOSS: zdo/zdo_app.c:595   Have not got nwk key - authentication failed
I (28893) ZBOSS: nwk/nwk_main.c:4409   zb_nwk_do_leave param 3 rejoin 0
I (28893) ZBOSS: common/zb_address.c:1063   zb_address_delete ref 27
I (28913) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (28913) ZBOSS: nwk/nwk_join.c:251   No dev for join
I (28913) ZBOSS: nwk/nwk_join.c:251   No dev for join
I (28923) ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 0
I (28933) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (28943) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (29973) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (30493) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xf455 on channel 11, dev 0x0, iface_id 0
I (32863) ZBOSS: mac/mac_associate.c:448   received association response
I (32873) ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xa3fd, iface_id 0, ch 11, rejoin 0
I (32873) ZBOSS: zdo/zdo_app_join.c:773   xpanid 8b:b5:d8:bc:da:21:56:ce
I (42913) ZBOSS: zdo/zdo_app.c:595   Have not got nwk key - authentication failed
I (42913) ZBOSS: nwk/nwk_main.c:4409   zb_nwk_do_leave param 6 rejoin 0
I (42913) ZBOSS: common/zb_address.c:1063   zb_address_delete ref 27
I (42933) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (42933) ZBOSS: nwk/nwk_join.c:251   No dev for join
I (42933) ZBOSS: nwk/nwk_join.c:251   No dev for join
``` 

Then I Reset the device. Now Crash was happening.  Here is the Log

```c
I (533) phy: phy_version: 211,0, 5857fe5, Nov  1 2023, 11:31:09
I (533) phy: libbtbb version: ce629d6, Nov  1 2023, 11:31:19
I (543) ZBOSS: common/zb_init_default.c:181   ED build
I (543) ZBOSS: common/zb_init_default.c:187   sizes: g_zb 19304 sched 1312 bpool 9856 nwk 1712 aps 3500 addr 660 zdo 836
I (553) ZBOSS: common/zb_init_default.c:196   sec 2
I (563) ZBOSS: common/zb_init_default.c:198   zcl 600
I (573) ZBOSS: common/zb_init_default.c:201   zll 376
I (573) ZBOSS: common/zb_init_default.c:207   nvram 140
I (583) ZBOSS: common/zb_init_default.c:210   buttons 140
I (583) ZBOSS: common/zb_init_default.c:212   err_ind 4
I (593) ZBOSS: common/zb_init_default.c:218   scheduler q size 60
I (603) ZBOSS: common/zb_init_default.c:221   g_mac 652 g_imac 28
I (603) ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 60 ZB_NWK_IN_Q_SIZE 30 ZB_MAC_PENDING_QUEUE_SIZE 15 ZB_APS_DST_BINDING_TABLE_SIZE 
64 ZB_APS_BIND_TRANS_TABLE_SIZE 16 ZB_N_APS_RETRANS_ENTRIES 20
I (623) ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 5 ZB_IEEE_ADDR_TABLE_SIZE 32 ZB_NEIGHBOR_TABLE_SIZE 32 ZB_NWK_ROUTING_TABLE_SIZE 0 ZB_APS_DUPS_TABLE_SIZE 32
I (643) ZBOSS: zdo/af_descriptor.c:357   ep_count 1
I (663) ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
I (663) ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
I (673) main_task: Returned from app_main()
I (683) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (683) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (693) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (703) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
I (713) ESP_ZB_DEEP_SLEEP: Start network steering
I (723) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (1243) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xf455 on channel 11, dev 0x0, iface_id 0
I (3613) ZBOSS: mac/mac_associate.c:448   received association response
I (3613) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0xf455 on channel 11, dev 0x0, iface_id 0
I (3683) ZBOSS: common/zb_debug.c:64   Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/mac/mac_data.c:854

abort() was called at PC 0x4201b685 on core 0
0x4201b685: zb_assert at ??:?

Stack dump detected
Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x40807f30  SP      : 0x4084eef0  GP      : 0x4080deb4  
0x4080063a: panic_abort at C:/Users/jithi/esp/esp-idf/components/esp_system/panic.c:452

0x40807f30: __ubsan_include at C:/Users/jithi/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4081ffd0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4084ef54  A0      : 0x4084ef1c  A1      : 0x4084ef52  
A2      : 0x00000000  A3      : 0x4084ef49  A4      : 0x00000001  A5      : 0x40816000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x000000ef  
S4      : 0x00000038  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000
``` 

### Steps to reproduce.

1. Update to esp-zigbee-sdk V1.0.5
2. Build the `"examples\esp_zigbee_sleep\deep_sleep"` example
3. Observe the Log
...

kelin6, 2023-12-07:
@Bosemani 
So are there  two questions here？

1) Deep sleep example cannot join the network where the coordinator is located
2) When the network connection is not successful, resetting the device will crash

However, based on the environment you provided, using ESP32H2 as the coordinator, the deep sleep example can successfully connect to the network without crashing But we haven't tried using Zigbee2mqtt as a Coordinator.

So can you provide a packet capture file using Wireshark when runing the deep sleep example?

Bosemani, 2023-12-08:
Hai @kelin6 

   I tested using esp-idf stable release, at that time crash was happen.
Now I Changed to Master Branch. 
Now there is no crash. Its working Fine.

Thank you.

closed 2023-12-11 by Bosemani