espressif/esp-zigbee-sdk #164: esp_zb_zcl_set_attribute_val is not working with PM2.5 cluster

g-chevalier, 2023-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.0

### esp-zigbee-lib version.

1.0.5

### esp-zboss-lib version.

1.0.5

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

esp_zb_zcl_set_attribute_val() should work with PM2.5 cluster as it works for instance with the Temperature measurement cluster.

### What is the actual behavior?

esp_zb_zcl_set_attribute_val() reports an error when used with PM2.5 cluster.

### Steps to reproduce.

esp_err_t err;
float_t pm25_level = (float_t) pm25;

ESP_LOGI(TAG, "Request PM25 reporting, val=%u", pm25);
err = esp_zb_zcl_set_attribute_val(HA_ESP_MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_PM2_5_MEASUREMENT,
          ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_PM2_5_MEASUREMENT_MEASURED_VALUE_ID, &pm25_level, false);
if (err != ESP_OK) {
  ESP_LOGE(TAG, "esp_zb_zcl_set_attribute_val / PM25 error, code=%d (%s)", err, esp_err_to_name(err));
} else {
  ESP_LOGI(TAG, "esp_zb_zcl_set_attribute_val / PM25 was OK");
}

### More Information.

The above code reports:
esp_zb_zcl_set_attribute_val / PM25 error, code=135 (ERROR)

The workaround given into https://github.com/espressif/esp-zigbee-sdk/issues/147 works partially
*((float_t*) esp_zb_zcl_get_attribute(HA_ESP_MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_PM2_5_MEASUREMENT,
       ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_PM2_5_MEASUREMENT_MEASURED_VALUE_ID)->data_p) = pm25_level;

See my comment in the aforementioned issue (147).

I think it is the same bug and perhaps due to a mishandling of the float type ("Volume" unit in Zigbee spec for those clusters) into esp_zb_zcl_set_attribute_val().

felipejfc, 2023-12-02:
worked for me after I properly set the min and max attributes

xieqinan, 2023-12-06:
Hi @g-chevalier ,

> My understanding is that the ESP-Zigbee -SDK should not check against min max if those are set to ESP_ZB_ZCL_PM2_5_MEASUREMENT_MIN_MEASURED_VALUE_UNDEFINED / ESP_ZB_ZCL_PM2_5_MEASUREMENT_MAX_MEASURED_VALUE_UNDEFINED

Or the definition of those constants is wrong or not the one expected by the library?

I disagree with your perspective. The values `ESP_ZB_ZCL_PM2_5_MEASUREMENT_MIN_MEASURED_VALUE_UNDEFINED` and `ESP_ZB_ZCL_PM2_5_MEASUREMENT_MAX_MEASURED_VALUE_UNDEFINED` are solely initialized as invalid values, and the stack does not prevent their assignment. However, according to the specification, the `measure value` is required to fall within the range defined by MinMeasuredValue and MaxMeasuredValue. It is conflict issue that a valid value is set for `measure value` within the range of these invalid values. Therefore, I believe that implementing a check for the value setting is a more stringent and appropriate behavior.

g-chevalier, 2023-12-06:
Hi @xieqinan,

Thank you for your answer.

It's OK for me as it is working when setting min and max, and being sure to interpret correctly the Zigbee Standard documentation is beyond my skills.

I was just wondering why it is working with the temperature cluster (I mean I set min and max to unknown 0x8000 as per the standard § 4.4.2.2.1.2, and the reports work)…

xieqinan, 2023-12-06:
@g-chevalier 

The temperature cluster will also verify whether the measured value falls within the specified range from the minimum to the maximum. I believe the measured value will not be set successfully if it is not equal to 0x8000 and exceeds the specified range of minimum and maximum values.

chshu, 2024-01-08:
@g-chevalier Could you please close the issue if no futher questions?