espressif/esp-zigbee-sdk #165: Can't create Window Covering Cluster

atmelmicro, 2023-12-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### esp-zigbee-lib version.

0.5.0

### esp-zboss-lib version.

0.4.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I want to create a window covering cluster on my ESP32-C6, but I'm having issues doing it. There are no built-in functions for it. I have tried using a custom cluster with no luck.

```
esp_zb_attribute_list_t *window_covering_attr = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING);

    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_WINDOW_COVERING_TYPE_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_PHYSICAL_CLOSED_LIMIT_LIFT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_PHY_CLOSED_LIMIT_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_LIFT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_NUMBER_OF_ACTUATIONS_LIFT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_NUMBER_OF_ACTUATIONS_TILT_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_CONFIG_STATUS_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_LIFT_PERCENTAGE_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);
    esp_zb_custom_cluster_add_custom_attr(window_covering_attr, ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &test_attr);

 esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, window_covering_attr, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

I have extracted the types for the clusters from the zboss lib. I wanted it to import it but there were type collisions.

### What is the actual behavior?

Create the cluster succesfully.

### Steps to reproduce.

1. put this into the main zigbee task

```c
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    // esp_zb_set_network_channel(11);

    /* set the on-off light device config */
    uint8_t test_attr, test_attr2, test_attr_3;

    test_attr = 0;
    test_attr2 = 4;
    test_attr_3 = 1;
    /* basic cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr2);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
[... 59 lines cut ...]
    ESP_LOGI(TAG, "Start With ESP_ZB_HA_COLOR_DIMMABLE_LIGHT_DEVICE_ID");
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_SHADE_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_device_add_set_attr_value_cb(attr_cb);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
```

### More Information.

Output from the ESP

```
E (427) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (427) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (437) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (447) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (447) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (457) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (467) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (477) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (477) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (487) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
E (497) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
```

xieqinan, 2023-12-06:
Hello @atmelmicro ,

The API for adding attributes to the window covering cluster can be found [here](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_attribute.h#L366). I recommend utilizing these APIs to create your window covering cluster.

Note that `esp_zb_custom_cluster_add_custom_attr()` is designed for custom clusters, not specifically for window covering clusters. The esp-zigbee-sdk also offers a general-purpose `esp_zb_cluster_add_attr()` API that can be used for adding attributes to any clusters.

atmelmicro, 2023-12-06:
Hey, I was running a old version of the zigbee sdk, I can see all the window covering functions now. Thanks a lot!

closed 2023-12-06 by atmelmicro