espressif/esp-zigbee-sdk #166: Network steering without Permit Join

TJvV-DT, 2023-12-04:
### Question

Hi,

Using 

I'm running a modified version of the `HA_color_dimmable_light` example as part of a mesh (configured as Router).
One thing I've noticed, is that when the device was already connected to a network and reboots, it will send a Permit Join request, opening up the network to devices when I don't want that yet.
I believe this comes from the call to `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);`.

Is there a way to have the node reconnect to the network on reboot without sending this Permit Join request?

xieqinan, 2023-12-08:
Hello @TJvV-DT,

You can use the following code to resolve your issue.

```
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device restarted");
            }
        } else {
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        
```

BaWe155, 2023-12-08:
Hello Xieqinan,

Thank you for your replay, together with TjvV, I test our Zigbee network.

I have added your code, but it didn't solve the problem unfortunately.
The behaver is when I power up the node and the Coordinator/control bridge already knows the node. the Control bridge opened the network for other nodes (permit join)

I hope too here soon from you.

Kind regards

xieqinan, 2023-12-11:
Hello @BaWe155 ,

As far as I understand, you anticipate that when the device is already connected to a network and undergoes a reboot (without erasing flash), the device should refrain from sending a `permit_join` request to the coordinator. If the aforementioned code is successfully added in the `HA_color_dimmable_light` example, upon pressing the boot button on the chip, the `HA_color_dimmable_light` should only print the following log in the console and take no further action.

```
I (477) main_task: Returned from app_main()                                                                                                 
I (487) ESP_ZB_COLOR_DIMM_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL                                                      
I (487) ESP_ZB_COLOR_DIMM_LIGHT: Zigbee stack initialized                                                                                   
W (497) ESP_ZB_COLOR_DIMM_LIGHT: Network(0x4d40) closed, devices joining not allowed.                                                       
I (497) ESP_ZB_COLOR_DIMM_LIGHT: Device restarted 

```

chshu, 2024-02-01:
@TJvV-DT @BaWe155 Please reopen if any follow up questions.

closed 2024-02-01 by chshu

eskomj, 2024-03-30:
Hello,
I have same kind of problem, but so that I need to joining and binding hapen from the beginning each time when I reboot. How can I do that? It seems that stack makes it automatically when it has done once. This call is used to test:

esp_zb_bdb_is_factory_new()

I am using zigbee example temperature sensor and I am trying to join Tuya gateway.

Esko