espressif/esp-zigbee-sdk #168: ESP32-C6 Unable to join Phlips Hue zigbee network

Paveldeg, 2023-12-05:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### esp-zigbee-lib version.

0.9.0

### esp-zboss-lib version.

0.7.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Using "HA_on_off_light" sample code from the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk), my Hue bridge should see HA_on_off_light available to be configured.

### What is the actual behavior?

Using "HA_on_off_light" sample code from the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk), my Hue bridge does not see the light, and I get repeating console messages like this
" ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)"

Is ESP32-C6 generally compatible with Philips Hue Zigbee bridge?

### Steps to reproduce.

1. Erase VRAM
2. Flash "HA_on_off_light" sample code to the chip
3. Search for available lights using Philips Hue bridge
4. The Hue bridge reports no lights found

### More Information.

I get repeating console messages like this
" ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)"

xieqinan, 2023-12-08:
Hello @Paveldeg,

The message "ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)" indicates that the light cannot find the coordinator.

Regarding this issue, I have two suggestions that might be valid:

1. Update the esp-zigbee-lib and esp-zboss-lib to the latest version.
2. Verify that the network of the Philips Hue bridge is open, which can be determined from Wireshark. Providing the Wireshark packet would be helpful for us to address this issue.

Paveldeg, 2023-12-08:
I upgraded the dependencies to v. 1.0.6, but still see same issue. My Hue network is open since I can easily add 3rd party smart bulbs to it (IKEA etc.). 
Full logs from ESP32:

`(378) main_task: Started on CPU0
I (378) main_task: Calling app_main()
I (388) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (388) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (428) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (448) main_task: Returned from app_main()
I (458) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (458) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (458) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (41138) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (82828) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)`

Paveldeg, 2023-12-08:
I tried to use code from another example: [https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c](url)
but still no luck:

` (384) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (424) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (444) main_task: Returned from app_main()
I (454) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (454) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (454) ESP_ZB_ON_OFF_LIGHT: Start network formation
I (464) ESP_ZB_ON_OFF_LIGHT: Formed network successfully (Extended PAN ID: 00:00:00:00:00:00:00:00, PAN ID: 0xffff, Channel:255, Short Address: 0xfffe)`

xieqinan, 2023-12-11:
@Paveldeg，

> I tried to use code from another example: [https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c](url)
> but still no luck:

This example is a Zigbee coordinator, it should not form a invalid network. Do you have some modifcations for it?

xieqinan, 2023-12-11:
> I upgraded the dependencies to v. 1.0.6, but still see same issue. My Hue network is open since I can easily add 3rd party smart bulbs to it (IKEA etc.).

xieqinan, 2023-12-11:
> I upgraded the dependencies to v. 1.0.6, but still see same issue. My Hue network is open since I can easily add 3rd party smart bulbs to it (IKEA etc.).

The sniffer packet is crucial for us to address this issue. Many situations can trigger the failure, including a mismatched channel, an excessively small RSSI threshold, limitations imposed by the black_list, conflicts in extend_panid, and the status of the network formed by the coordinator, among other factors. You can verify these aspects initially.

We also provide documentation on Wireshark usage. You can refer to [docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging).

Paveldeg, 2023-12-16:
I setup WireShark on same Mac machine where I run my ESP32-C6 board (connected via USB), but which interface I should be capturing on? I only see WiFi, but I don't see any relevant traffic there. I know my Hue gateway IP address, but again I don't see any relevant traffic from this IP in WireShark.

Paveldeg, 2023-12-17:
Would be great if Espressif engineers could provide examples witch actually work with any 3rd party Zigbee gateway, rather than relying on end users to debug connectivity issues.
Thanks.

nrw505, 2024-02-19:
@Paveldeg The Hue product line require authentication to join the network with a pre-configured distributed global link key (which you need to get from Phillips, signing an NDA etc.)

Once you have that key, you can configure it before starting network steering, e.g. after calling `esp_zb_init()`

```
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_enable_joining_to_distributed(true);
    esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);
```

chshu, 2024-03-01:
@nrw505 Thanks for the answering, closing the issue.

closed 2024-03-01 by chshu