espressif/esp-zigbee-sdk #169: Correct way to restart after failed network steering or Zigbee stack initialisation

AussieSusan, 2023-12-06:
### Question

ESP32-H2, IDF V5.1.2, Zigbee lib 1.0.3, ZBoss lib 1.0.2
Code is basically the 'On Off Light' demo code.
From the signal handler I get the message that the network steering is starting and after a while I normally get the 'Network steering was not successful (status -1)' message.
If I do successfully join the network I often get an NLME Status Indication with a status value of 0 and the network seems to have dropped after that (i.e. I can't get any response from the coordinator).
I also get messages at times saying "Failed to initialise Zigbee stack".
The coordinator says that the RSSI is -58 so I think the signal strength is not the issue.
My question is: when I get these failures, what is the correct way to restart the Zigbee end device?
Susan

xieqinan, 2023-12-08:
Hello, 

> From the signal handler I get the message that the network steering is starting and after a while I normally get the 'Network steering was not successful (status -1)' message.

This indicates that the end device cannot find or join the network, possibly triggered by the closed network of the coordinator.

> If I do successfully join the network I often get an NLME Status Indication with a status value of 0 and the network seems to have dropped after that (i.e. I can't get any response from the coordinator).

It indicates the response for the 'link status', which does not imply that the network has been dropped.

> I also get messages at times saying "Failed to initialise Zigbee stack".

It indicates that the non-factory-reset Zigbee end device cannot find its parent node when it reboots.

As a result, when the end device is factory new, it needs an open network to join. If the device is not factory new and has already joined the network, please ensure that its parent is online when it reboots. The 'NLME Status Indication' provides information about the network status.

AussieSusan, 2023-12-10:
Thank you for your comments.
Can you also please answer the question I asked - how to restart after I get those errors?
Susan

xieqinan, 2023-12-11:
@AussieSusan ，

> Can you also please answer the question I asked - how to restart after I get those errors?

No problem.

- `Failed to initialize Zigbee stack`: You need to power on the parent node of the device first, then restart the device.

- `Network steering was not successful (status -1)`: Place a network open coordinator around the device; it is not necessary to restart.

- `NLME Status Indication `: It is not necessary to restart.

chshu, 2024-01-08:
@AussieSusan please reopen if any further questions.

closed 2024-01-08 by chshu