espressif/esp-zigbee-sdk #170: OTA Update over zigbee2mqtt

vedatbotuk, 2023-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-582-gab03c2ea13

### esp-zigbee-lib version.

1.0.5

### esp-zboss-lib version.

1.0.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

I want to upgrade esp32h2 over zigbee2mqtt. 

### What is the actual behavior?

zigbee2mqtt does not send normal binary. Because file header is not supported.

An example header bytes of OTA file.
0000000 **f11e** 0bee 0100 0038 0000 128b 0106 0401

Binary file of esp32h2 application
0000000 **05e9** 1f02 0250 4080 00ee 0000 0010 0000

I think, binary file has to start with **0xf11e**.  Then this can be sent to me zigbee2mqtt. But esp32h2 should also accept that.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

[image]

g-chevalier, 2023-12-06:
Hi,

This issue is related to my question in https://github.com/espressif/esp-zigbee-sdk/issues/146, which by the way I should have written as a feature request.

I think a there is a need to have a script that can create a standard OTA file from an ESP-IDF binary.

At the moment, I did not have time to investigate the tool that @nomis mentioned into my issue, but for sure, I would prefer that Espressif provides this tool OOTB.

nomis, 2023-12-06:
> I think, binary file has to start with **0xf11e**. Then this can be sent to me zigbee2mqtt. But esp32h2 should also accept that.

What you're asking for is not possible. The format of the application binary running on the ESP32 and the format of the specification for Zigbee OTA files are different and cannot be changed.

You need to identify where the content of the upgrade image sub-element begins in the image and use that with the ESP OTA functions, as I've told you twice in https://github.com/nomis/candle-dribbler/discussions/2.

This is not a bug. A realistic feature request would be wrapper functions that can parse the first upgrade image sub-element from a stream for you (and do transparent zlib decompression, because ESP32 applications are always huge).

vedatbotuk, 2023-12-06:
Thank you for your prompt response.

I faced challenges with the C++ code, prompting me to seek alternative solutions. Currently, I am in the process of experimenting with the subelement, attempting to introduce a delay in the reception of bytes.

closed 2023-12-06 by vedatbotuk