espressif/esp-zigbee-sdk #172: Commercial end device really slow to respond when connected to ESP32-C6 switch on off command

djorr5, 2023-12-08:
### Question

Hi,

I have successfully connected a Zigbee power point (GPO) to my ESP32-C6 using the HA_on_off_switch example. However when I try to control the GPO, either switching on or off it takes a long time to complete the action. Sometimes more than 30 seconds. It does eventually turn on or off but the time taken to complete the action varies from 15 seconds to more than 30 seconds. When I used the HA_on_off_light with another C6 board it was instantaneous. Anyone else had any experience with this happening?

djorr5, 2023-12-10:
I noticed that when I send either an on or off command about 10 seconds later I always get a NLME Status Indication (0x32) message. And then 20 seconds after I get the message the end device turns on or off. Consistently on these timings.  What does the NLME message mean? Can't see it in the SDK documentation anywhere and a google search shows nothing as well. I can only assume that it is the end device reconnecting? If I toggle the on and off in quick succession, i.e. about 10 times really quickly, I get the NLME Status message after 10 seconds and then 20 seconds later the end devices cycles through the on and off commands. Any help on this would be appreciated. 

```c
I (221472) ESP_ZB_ON_OFF_SWITCH: Send 'off' toggle' command
I (232582) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (261002) ESP_ZB_ON_OFF_SWITCH: Send 'on' toggle' command
I (272122) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (300942) ESP_ZB_ON_OFF_SWITCH: Send 'off' toggle' command
I (311662) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (346712) ESP_ZB_ON_OFF_SWITCH: Send 'on' toggle' command
I (357282) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (358792) ESP_ZB_ON_OFF_SWITCH: Send 'off' toggle' command
I (369442) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (379882) ESP_ZB_ON_OFF_SWITCH: Send 'on' toggle' command
I (390732) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

xieqinan, 2023-12-12:
@djorr5 ,

As a temporary solution, you can add the following case to the `esp_zb_app_signal_handler()` to check the NLME status, which can be referenced from `zb_nwk_command_status_t`.
```
    case ESP_ZB_NLME_STATUS_INDICATION: {
        printf("%s, status: 0x%x\n", esp_zb_zdo_signal_to_string(sig_type), *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p));
    } break;
    
```

djorr5, 2023-12-12:
Awesome, thanks for the support. It showed up as a ZB_NWK_COMMAND_STATUS_NO_ROUTE_AVAILABLE and from that I reasoned that it still had the old light example registered in the network. I did an erase-flash and it now works as expected. Thanks for the support.

closed 2023-12-12 by djorr5