espressif/esp-zigbee-sdk #173: Espressif Zigbee HA Light Example not working with Tuya Gateway Smartlink app

munish-nuos, 2023-12-12:
### Question

Espressif Zigbee HA Light Example joining the network successfully, but not working with Tuya Gateway Smart Home app. On app it show a button and works only one time click to ON the led on Zigbee Router side and after that nothing happens. 
But same code works fine on SONOFF Gateway and its app.

xieqinan, 2023-12-12:
@munish-nuos ,

Could you please check whether the on/off command was sent by the Smart Home app?

wisemanny, 2023-12-13:
@munish-nuos please read tuya spec and stick to it. They require specific endpoint id to be used. I just built a 2 gang switch that works with tuya. Here are details on what they expect: https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/tuya-zigbee-multiple-switch-access-standard?id=K9ik6zvnqr09m

munish-nuos, 2023-12-13:
Hello xieqinan,

Yes, on/off command is send by Smart Home App and i received callback in esp32H2 Zigbee example.
Issue is that , when i send back the on/off command from esp32H2 Server Cluster to SONOFF gateway, Smarthome app not changing the state on/off in SmartHome app.

munish-nuos, 2023-12-13:
Hello wisemanny, 

Thank you for your valuable response. I will surely going to read these details sent by you.

munish-nuos, 2023-12-13:
> @munish-nuos please read tuya spec and stick to it. They require specific endpoint id to be used. I just built a 2 gang switch that works with tuya. Here are details on what they expect: [...]

Its Done, now light switches are working in Tuya gateway app. I just set the endpoints 1, 2 and its working now.

closed 2024-01-08 by chshu