espressif/esp-zigbee-sdk #174: Esp zigbee HA light example: Bidirectional communication with Esp32H2 Router example and any zigbee gateway

munish-nuos, 2023-12-12:
### Question

Could you help me with building a project having Zigbee Router on one side with light and switch on same board (Router as Server configuration + light endpoint) and connect it to any gateway like sonoff, tuya, etc. and operate it either from gateway app or from esp32he board switch press. Simply its a bidirectional communication between espressif router and any gateway coordinator. Now everything is working fine, join network is ok, device binding is ok, light is operated from gateway app, except it is not working from espressif switch side. on pressing switch, i am calling this function:

static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    /* By checking the button function pair to call different cmd send */
    switch (button_func_pair->func) {
    case SWITCH_ONOFF_TOGGLE_CONTROL:
        /* Send on-off toggle command to remote device */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;

    	cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = g_device_ctx.on_off_light.short_addr;
    	cmd_req.zcl_basic_cmd.dst_endpoint = g_device_ctx.on_off_light.endpoint;
    	cmd_req.zcl_basic_cmd.src_endpoint = HA_ESP_LIGHT_ENDPOINT;

        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command to address(0x%x) endpoint(%d))",
        		g_device_ctx.on_off_light.short_addr, g_device_ctx.on_off_light.endpoint);
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
        break;
    default:
        break;
    }
}  
command send to gateway but nothing happens on app like switch not changing its state in app GUI.

What I would like to achieve:
1. Bidirectional communication between Router and Gateway.
2. On sending command from Switch in router side, the app like ewelink light toggle switch should change its state.

xieqinan, 2023-12-12:
Hello @munish-nuos ,

Do you expect the on/off cluster server (light) and client (switch) to run on a single board, then the board joins the network formed by the Smart Home app, and finally, the coordinator can show the interaction between the light and switch. Is that correct?

As far as I understand, you may need to pay attention to two points.

Firstly, ensure that the `g_device_ctx` information is correct. Could you provide the log printed by the `ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command to address(0x%x) endpoint(%d))"`? If the light and switch are on the same board, `g_device_ctx.on_off_light.short_addr` should be equal to `esp_zb_get_short_address()`.

Secondly, if the command `esp_zb_zcl_on_off_cmd_req()` is sent to the board itself, it will not be sent to the air, so the gateway cannot receive the command. As a simple solution, the gateway should use the report mechanism to subscribe to the `ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID` attribute so that when the attribute changes, the light/switch can report the value to the gateway.

munish-nuos, 2023-12-12:
Hello xieqinan,

Thanks for the reply. 

No, I don't expect the on/off cluster server (light) and client (switch) to run on a single board. My board only have server(light) cluster. But board has the a light and a switch. Client cluster is created on gateway end and app associated with gateway, controls the light on my board. My esp32h2 board has binding information of client cluster from gateway. 
g_device_ctx.on_off_light.endpoint and g_device_ctx.on_off_light.short_add are information i got from Gateway during binding request from my board. Now gateway app is toggling the light on my board but when i call esp_zb_zcl_on_off_cmd_req(&cmd_req), it dont toggle the device light on my app. That is the challenge.

munish-nuos, 2023-12-12:
Mobile APP (Toggle Light) ------wifi---------> GATEWAY ---------zigbee--------->LIGHT (SERVER)  [Working OK]

LIGHT (SERVER) [OnOff Button Pressed (not an end point) (calling esp_zb_zcl_on_off_cmd_req(&cmd_req))] -----zigbee----> GATEWAY ------wifi----->Mobile APP(Light not toggling)  [Not Working]

munish-nuos, 2023-12-13:
Hi, can anyone help me with this issue.

chshu, 2024-02-01:
@mundevx Please reopen if any follow up questions.

closed 2024-02-01 by chshu

mundevx, 2024-02-01:
> @mundevx Please reopen if any follow up questions.

Thanks, i already solved this issue