espressif/esp-zigbee-sdk #175: Really not sure if it's the SDK or hardware, I really need to fix it sooner rather than later

teleger, 2023-12-13:
### Question

While I was using the zigbee main task, my console kept rebooting and I had to open the trace log, but I didn't know where to start.
Attach my log print and my main function running:

trace log :
ZBOSS: common/zb_debug.c:64   
Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/nwk/nwk_multimac_proxy.c:206

And part of the print is this:
ZBOSS: common/zb_debug.c:64   
Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/nwk/nwk_multimac_proxy.c:226

main:
xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
static void esp_zb_task(void *pvParameters)
{
#if 1
   esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_WARN, ESP_ZB_TRACE_SUBSYSTEM_MAC | ESP_ZB_TRACE_SUBSYSTEM_APP);
#endif
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
#if(CONFIG_ZB_RADIO_MACSPLIT_UART)
    esp_zb_add_rcp_failure_cb(rcp_error_handler);
#endif
    esp_zb_main_loop_iteration();
    esp_rcp_update_deinit();
    vTaskDelete(NULL);
}

The pattern I use is radio_mode = RADIO_MODE_UART_RCP;The hardware is esp32s3 + esp32h2.

This example uses the gateway and rcp demo.

I don't know. Why do I keep rebooting? Is it my hardware? Can you help me check the problem? I know you are running well,
but it does not mean that it is very good, I want to debug this code, but there is no source code, I am helpless now, can you help?

chshu, 2023-12-19:
@teleger The current Zboss **macsplit** based zigbee gateway+rcp solution is an experimental feature, we do see some stability issues recently.
 
We are working on a new zigbee gateway solution which will base on the [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp), some stress tests are on going, will push to github in 2024 Q1.

teleger, 2023-12-20:
> @teleger 当前的 Zboss **macsplit** 基于 zigbee gateway+rcp 的解决方案是一个实验性功能，我们最近确实看到了一些稳定性问题。
> 
> 我们正在开发一种新的 zigbee 网关解决方案，该解决方案将基于 [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp)，一些压力测试正在进行中，将推送到 github 2024 年第一季度。

I am looking forward to having a new and stable version to use Zigbee-gateway related functions. At present, I also want to use mcu(cpu) to connect esp32h2 as zigbee gateway function. Does your sdk provide similar functions? Simply put, it is the use of coprocessor to complete the development of zigbee gateway, thank you for your quick reply, thank you

chshu, 2024-02-04:
@teleger The [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) based Zigbee Gateway example is available now: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway

Note:
- Please use the esp-idf master branch ([Commit ad3e1e3](https://github.com/espressif/esp-idf/commit/ad3e1e3daf7037e58c650c735e1f6218ee886651) or later). It will be supported in the coming IDF v5.1.3 and later release versions.
- Enable `OPENTHREAD_NCP_VENDOR_HOOK` option before building [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) example.

Feel free to try out the new feature, and let us know if any questions or suggestions.

closed 2024-02-04 by chshu