espressif/esp-zigbee-sdk #177: Need help in executing esp_zb_zcl_set_attribute_val in project Esp32H2 Zigbee as Light with Server Cluster

munish-nuos, 2023-12-14:
### Question

Hi,

I am working to build a project as described below:
Firstly, I build a board ESp32H2 Zigbee server cluster using example code of esp_zigbee_customized_devices->customized_server.
In hardware, i am using a key press with led on/off. I set it to ZIGBEE ROUTER mode in server cluster configuration.
Secondly, i am using Tuya Gateway + Smart Life App. It works as ZIGBEE COORDINATOR.
Zigbee Finding, Binding, Discovering cluster and attribute matching is working fine.
If i change from Smart Life App, it reports to my Zigbee Router Esp32H2 device and my led is toggling fine on board.
But when i change from my keypress in my board Zigbee Router Esp32H2, and send on/off change command to destination endpoint (Tuya Gateway) using fxn esp_zb_zcl_on_off_cmd_req() , then nothing happens on SmartLife App, switch don't toggle state. 
Then i replaced esp_zb_zcl_on_off_cmd_req() with esp_zb_zcl_set_attribute_val and it works. But issue arise when it doesn't work for SONOFF Gateway. The Ewelink App switch don't toggle state with esp_zb_zcl_set_attribute_val  but it works for Tuya gateway App.
Also, when there are multiple endpoints, esp_zb_zcl_set_attribute_val  not works even in Tuya gateway. Need help on this topic.

xieqinan, 2023-12-29:
Hello @munish-nuos ,

> If i change from Smart Life App, it reports to my Zigbee Router Esp32H2 device and my led is toggling fine on board.

When changing from the Smart Life App, the app sends an on/off command to the Zigbee Router ESP32H2, altering the state of the on/off (client) attribute. If the on/off(client) attribute is bound, it generates a report to communicate its state. Consequently, it works fine.

> But when i change from my keypress in my board Zigbee Router Esp32H2, and send on/off change command to destination endpoint (Tuya Gateway) using fxn esp_zb_zcl_on_off_cmd_req() , then nothing happens on SmartLife App, switch don't toggle state.

When you press the key on the Zigbee Router ESP32H2 to trigger the `esp_zb_zcl_on_off_cmd_req()` to send an on/off command to the Tuya gateway, the switch state does not change. I assume that this behavior is solely meant to send a command, which does not alter the on/off (server) state, thereby preventing the on/off (server) state from being reported. In contrast, when you use the `esp_zb_zcl_set_attribute_val()` command, it changes the on/off (server) state, triggering the generation of a report.

> Then i replaced esp_zb_zcl_on_off_cmd_req() with esp_zb_zcl_set_attribute_val and it works. But issue arise when it doesn't work for SONOFF Gateway. The Ewelink App switch don't toggle state with esp_zb_zcl_set_attribute_val but it works for Tuya gateway App.

I am unsure if the `SONOFF Gateway` has configured the on/off (server) attribute. This could be a potential reason for it not working. Please verify this configuration again.

> Also, when there are multiple endpoints, esp_zb_zcl_set_attribute_val not works even in Tuya gateway.

The report-bound relationship is from one endpoint to another endpoint. I believe it should have an endpoint with an on/off attribute switch that can function across multiple endpoints in TuYa gateway.

chshu, 2024-03-08:
Closing as the question is answered.

closed 2024-03-08 by chshu