espressif/esp-zigbee-sdk #178: I am using time cluster to update time from router(server) to end device(client)

omtrivedi-bacancy, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### esp-zigbee-lib version.

v1.0.6

### esp-zboss-lib version.

v1.0.6

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

To read and write particular attribute with write access.

### What is the actual behavior?

I am unable to read/write attribute from router(server) to End device(client) .
I am using time , occupancy and temperature cluster.

Also, can I report from client to server? 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

grebenyuk51, 2024-03-09:
It's not a bug

1. Subscribe on core actions
`esp_zb_core_action_handler_register(zb_action_handler);`
2. Handle read attribute callback
```
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t *)message);
        break;
    default:
        ESP_LOGE(TAG, "No handler for action(0x%x)", callback_id);
        break;
    }
    return ret;
}
```
3. write needed behavior
```
static esp_err_t zb_read_attr_resp_handler(const esp_zb_zcl_cmd_read_attr_resp_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)", message->info.status);

    esp_zb_zcl_read_attr_resp_variable_t *variable = message->variables;
    ESP_LOGI(TAG, "Read attribute response: status(%d), cluster(0x%x), attribute(0x%x), type(0x%x), value(%d)", 
            message->info.status,
            message->info.cluster,
            variable->attribute.id,
            variable->attribute.data.type,
            variable->attribute.data.value ? *(uint8_t *)variable->attribute.data.value : 0);

    if (message->info.dst_endpoint == ENDPOINT)
    {
        switch (message->info.cluster) 
        {
        case ESP_ZB_ZCL_CLUSTER_ID_TIME:
            switch(variable->attribute.id)
            {
                case ESP_ZB_ZCL_ATTR_TIME_LOCAL_TIME_ID:
                    time_t t = *(uint32_t*) variable->attribute.data.value + 946684800;
                    //TODO
                    break;
                default:
                    ESP_LOGE(TAG, "Message data: cluster(0x%x), attribute(0x%x)  ", message->info.cluster, variable->attribute.id);
            }
            break;
        default:
            ESP_LOGE(TAG, "Message data: cluster(0x%x), attribute(0x%x)  ", message->info.cluster, variable->attribute.id);
        }
    }
    return ESP_OK;
}
```

Call this method after network steering on end device
```
void read_server_time()
{
    uint16_t attributes[] = 
    {
        ESP_ZB_ZCL_ATTR_TIME_LOCAL_TIME_ID
    };

    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.attr_field = attributes;
    read_req.attr_number = sizeof(attributes) / sizeof(uint16_t);
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TIME;
    read_req.zcl_basic_cmd.dst_endpoint = ENDPOINT;
    read_req.zcl_basic_cmd.src_endpoint = ENDPOINT;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;  //coordinator
    ESP_LOGI(TIME_TAG, "Request read time attr");
    esp_zb_zcl_read_attr_cmd_req(&read_req);
}
```

xieqinan, 2024-06-17:
@omtrivedi-bacancy,

Do you have some updates for this issue? If not, please consider closing it.

closed 2024-06-17 by omtrivedi-bacancy

mundevx, 2025-03-06:
> It's not a bug
>
> 1. Subscribe on core actions
>    `esp_zb_core_action_handler_register(zb_action_handler);`
> 2. Handle read attribute callback
>
> ```
> static esp_err_t [...]

This code not works for me. 
I called read_server_time() after network steering on my Router device, no response. Then i also bind and config report this cluster , but no success.

xieqinan, 2025-03-07:
@mundevx,

This issue is closed. You can submit a new issue to describe your question and reference this one.