espressif/esp-zigbee-sdk #181: How/Where to set the make/model etc data in the Zigbee HA Light On/off ?

Tokn59, 2023-12-15:
### Question

Hi, using this example, where do I set the name, model and other data to "broadcast" so it can be recognised ? Thank You fro providing a clear example with those definitions !!!!

xieqinan, 2023-12-18:
@Tokn59 ，

Regarding this issue, I believe a more detailed description is necessary. Do you expect the data packet to be sent via the broadcasting way? Is the `name` or `model` an attribute of the basic cluster?

Tokn59, 2023-12-18:
> @Tokn59 ，
> 
> Regarding this issue, I believe a more detailed description is necessary. Do you expect the data packet to be sent via the broadcasting way? Is the `name` or `model` an attribute of the basic cluster?

Yes those should be part of the basic cluster. Meanwhile I managed to define and successfully configure them !

xieqinan, 2023-12-28:
Hello @Tokn59 ,

If you intend to broadcast the ZCL commands, you can follow the code below:

Broadcast the ON/OFF toggle command:

```
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0xffff;
        cmd_req.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command to address(0x%x) endpoint(%d)", cmd_req.zcl_basic_cmd.dst_addr_u.addr_short, on_off_light.endpoint);
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
        
```

xieqinan, 2024-03-18:
Hello @Tokn59 ,

Please refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c#L136

xieqinan, 2024-04-15:
@Tokn59 ，

If the issue had been fixed, please consider closing it. Welcome to discuss any question for esp-zigbee-sdk.

closed 2024-07-26 by chshu