espressif/esp-zigbee-sdk #182: esp_zb_zcl_read_attr_cmd_req do not work with aqara Zigbee 1.2 devices

RVC69, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2

### esp-zigbee-lib version.

1.0.6

### esp-zboss-lib version.

1.0.6

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

Expected zb_action_handler callback when calling esp_zb_zcl_read_attr_cmd_req on zigbee 1.2 and 3.0 devices__

### What is the actual behavior?

no callback using esp_zb_zcl_read_attr_cmd_req  on aqara zigbee 1.2 devices

### Steps to reproduce.

**1) this code works with ikea tradfri switch device**
// -------------------------------------------------------------------------------------------------------------------
void switch_state(void) {
    uint8_t index = devices_find_index_str(ON_OFF_SWITCH_IKEA_1, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);
    if (index == 0xff) return;
        ESP_LOGI(TAG, "Reading state from local endpoint (0x%x), distant endpoint(0x%x), address(0x%x), clusterId(0x%x), IEEE address: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x", 
                    ha_devices[index].coordinator_endpoint, ha_devices[index].endpoint, ha_devices[index].short_addr,ha_devices[index].cluster_id,
                    ha_devices[index].ieee_addr[7], ha_devices[index].ieee_addr[6], 
                    ha_devices[index].ieee_addr[5], ha_devices[index].ieee_addr[4],
                    ha_devices[index].ieee_addr[3], ha_devices[index].ieee_addr[2], 
                    ha_devices[index].ieee_addr[1], ha_devices[index].ieee_addr[0]);
        static esp_zb_zcl_read_attr_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.src_endpoint = ha_devices[index].coordinator_endpoint; 
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        //cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
        cmd_req.zcl_basic_cmd.dst_endpoint = ha_devices[index].endpoint; 
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = ha_devices[index].short_addr; 

        cmd_req.clusterID = ha_devices[index].cluster_id;
        uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID};
        cmd_req.attr_number = sizeof(attributes) / sizeof(uint16_t);;
        cmd_req.attr_field = attributes;                                                   
		esp_zb_zcl_read_attr_cmd_req( &cmd_req);
}
Got : 
I (576198) ZIGBEE: Send 'on_off toggle' from local endpoint(0x3), distant endpoint(0x1), address(0x21a1), clusterId(0x6), IEEE address: 14:b4:57:ff:fe:c5:08:d7
I (578198) ZIGBEE: Reading state from local endpoint (0x3), distant endpoint(0x1), address(0x21a1), clusterId(0x6), IEEE address: 14:b4:57:ff:fe:c5:08:d7
W (578218) ZIGBEE: Zigbee action(0x1000) callback
W (578218) ZIGBEE: ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID 0x1000
I (578218) ZIGBEE: Received attribute response to local endpoint (0x3) from distant endpoint(0x1) address(0x21a1) cluster(0x6)
I (578228) ZIGBEE: Received attribute response: attribute(0x0), type(0x10), value(0)

**2) Same code on aqara 1.2 temperature sensor, correctly bound, not working (no action handler callback)**
// -------------------------------------------------------------------------------------------------------------------
void temperature_state(void) {
    uint8_t index = devices_find_index_str(TEMPERATURE_MEASUREMENT_AQUARA_1, ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT);
    if (index == 0xff) return;
        ESP_LOGI(TAG, "Reading state from local endpoint (0x%x), distant endpoint(0x%x), address(0x%x), clusterId(0x%x), IEEE address: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x", 
                    ha_devices[index].coordinator_endpoint, ha_devices[index].endpoint, ha_devices[index].short_addr,ha_devices[index].cluster_id,
                    ha_devices[index].ieee_addr[7], ha_devices[index].ieee_addr[6], 
                    ha_devices[index].ieee_addr[5], ha_devices[index].ieee_addr[4],
                    ha_devices[index].ieee_addr[3], ha_devices[index].ieee_addr[2], 
                    ha_devices[index].ieee_addr[1], ha_devices[index].ieee_addr[0]);
        static esp_zb_zcl_read_attr_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.src_endpoint = ha_devices[index].coordinator_endpoint; 
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        //cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
        cmd_req.zcl_basic_cmd.dst_endpoint = ha_devices[index].endpoint; 
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = ha_devices[index].short_addr;

        cmd_req.clusterID = ha_devices[index].cluster_id; 
        uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID};
        cmd_req.attr_number = sizeof(attributes) / sizeof(uint16_t);
        cmd_req.attr_field = attributes;                                     
		esp_zb_zcl_read_attr_cmd_req( &cmd_req);
}
I (10198) ZIGBEE: Reading state from local endpoint (0x1), distant endpoint(0x1), address(0x1dd), clusterId(0x402), IEEE address: 00:15:8d:00:02:82:ee:2d

**Got no action handler callback**

xieqinan, 2023-12-18:
Hello @RVC69 ,

Regarding these issues, I believe the command `esp_zb_zcl_read_attr_cmd_req()` is being used correctly by you. However, it is necessary for you to utilize the sniffer to confirm whether the packet of `read_cmd_req` sent by the `ESP` device and the packet of `read_cmd_resp` responded by the `Aqara` are being transmitted or not.

RVC69, 2023-12-18:
Hello, Thanks for the answer, using sniffer and wireshark, binding new device and reading aqara temperature sensor, gives the following logs. Comments are welcomed. 
####################################################################################################################################
ESP IDF LOGS
I (3240) ZBOSS: zcl/zcl_main.c:67   >>zcl_init
I (3240) ZBOSS: zcl/zcl_reporting.c:112   >>zb_zcl_init_reporting_ctx
I (3250) ZBOSS: zcl/zcl_reporting.c:120   Use 23 buffer for reporting
I (3250) ZBOSS: zcl/zcl_reporting.c:134   <<zb_zcl_init_reporting_ctx
I (3260) ZBOSS: zcl/zcl_main.c:94   <<zcl_init
I (3260) ZBOSS: zll/zll_sas.c:105   > zb_zll_process_sas
I (3270) ZBOSS: zll/zll_sas.c:42   > zb_zll_sas_startup_attr
I (3280) ZBOSS: zll/zll_sas.c:61   < zb_zll_sas_startup_attr
I (3280) ZBOSS: zll/zll_sas.c:99   < zb_zll_sas_security_attr
I (3290) ZBOSS: zll/zll_sas.c:113   < zb_zll_process_sas
I (3300) ZBOSS: common/zb_init_default.c:183   FFD build
I (3300) ZBOSS: common/zb_init_default.c:187   sizes: g_zb 19924 sched 1312 bpool 9856 nwk 1908 aps 3904 addr 660 zdo 836
I (3310) ZBOSS: common/zb_init_default.c:196   sec 2
I (3320) ZBOSS: common/zb_init_default.c:198   zcl 600
I (3330) ZBOSS: common/zb_init_default.c:201   zll 380
I (3330) ZBOSS: common/zb_init_default.c:207   nvram 140
I (3340) ZBOSS: common/zb_init_default.c:210   buttons 140
I (3340) ZBOSS: common/zb_init_default.c:212   err_ind 4
I (3350) ZBOSS: common/zb_init_default.c:218   scheduler q size 60
I (3360) ZBOSS: common/zb_init_default.c:221   g_mac 800 g_imac 28
I (3360) ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 60 ZB_NWK_IN_Q_SIZE 30 ZB_MAC_PENDING_QUEUE_SIZE 15 ZB_APS_DST_BINDING_TABLE_SIZE 64 ZB_APS_BIND_TRANS_TABLE_SIZE 16 ZB_N_APS_RETRANS_ENTRIES 20
I (3380) ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 32 ZB_IEEE_ADDR_TABLE_SIZE 32 ZB_NEIGHBOR_TABLE_SIZE 32 ZB_NWK_ROUTING_TABLE_SIZE 5 ZB_APS_DUPS_TABLE_SIZE 32
I (3400) ZBOSS: commissioning/bdb/bdb_finding_binding.c:60   > zb_bdb_finding_binding_init_ctx
I (3410) ZBOSS: commissioning/bdb/bdb_finding_binding.c:64   < zb_bdb_finding_binding_init_ctx
I (3420) ZBOSS: zcl/zcl_diagnostics_commands.c:109   >>zb_zcl_diagnostics_sync_counters(), cb_param 0, cb 0x0x0      
I (3430) ZBOSS: zcl/zcl_diagnostics_commands.c:118   Synchronization is scheduled!
I (3440) ZBOSS: zcl/zcl_diagnostics_commands.c:128   <<zb_zcl_diagnostics_sync_counters()
I (3450) ZBOSS: zdo/zdo_app_prod_conf.c:144   no production config block found
I (3450) ZBOSS: zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling to application
I (3470) ZIGBEE: Zigbee stack running
I (3470) ZBOSS: zcl/zcl_diagnostics_commands.c:74   >>zb_zcl_diagnostics_sync_counters_cb(), param 25
I (3480) ZBOSS: zcl/zcl_diagnostics_commands.c:76   status 0x0
I (3490) ZBOSS: zcl/zcl_diagnostics_commands.c:101   <<zb_zcl_diagnostics_sync_counters_cb()
I (3500) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:340   dev type 0, joined 1, ext_pan_id 1, authenticated 1  
I (3510) ZBOSS: zdo/zdo_formation.c:78   zb_nlme_network_formation_confirm 26 status 0
W (3520) ZIGBEE: Network(0x5c56) closed, devices joining not allowed.
I (3520) RGB_LED: RGB Led R(0) G(0) B(0) Blink (0)
I (3630) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 0
I (3630) ZIGBEE: Start network formation
I (3630) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 11 comm status 0
I (3640) ZIGBEE: Formed network successfully (ieee_address: 40:4c:ca:ff:fe:4a:c6:e4, PAN ID: 0x5c56, Channel:13)
I (4250) ZIGBEE: Network(0x5c56) is open for 180 seconds
I (4250) RGB_LED: RGB Led R(0) G(0) B(255) Blink (250)
I (4350) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 0
I (4350) ZIGBEE: Joined network successfully (Extended PAN ID: 40:4c:ca:ff:fe:4a:c6:e4, PAN ID: 0x5c56, Channel:13, Short Address: 0x0000)
I (8200) ZIGBEE: Device not found IEEE address: 00:15:8d:00:02:82:ee:2d
I (28200) ZIGBEE: Device not found IEEE address: 00:15:8d:00:02:82:ee:2d
I (48200) ZIGBEE: Device not found IEEE address: 00:15:8d:00:02:82:ee:2d
I (49930) ZBOSS: nwk/nwk_cr_join.c:1269   association: status 0, address 0x18af, device 00:15:8d:00:02:82:ee:2d, mac_iface_idx 0
I (50650) ZBOSS: secur/bdb_secur.c:397   ACHTUNG: schedule bdb_link_key_refresh_alarm, ref 31
I (50910) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (50910) ZIGBEE: New device commissioned or rejoined (short: 0x18af) Capability 0x80
I (50960) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (50960) ZBOSS: zcl/zcl_common.c:1053   > zb_zcl_is_target_endpoint: ep_desc 0x0, profile_id 0x104
I (50960) ZBOSS: zcl/zcl_common.c:1057   ep or simple desc for it doesn't exist!
I (50970) ZBOSS: zcl/zcl_common.c:1113   < zb_zcl_is_target_endpoint, match 0
I (50980) ZBOSS: zdo/af_rx.c:459   APS pkt 11 for ep 1 is not recognized - drop
I (51760) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (51760) ZBOSS: zcl/zcl_common.c:1053   > zb_zcl_is_target_endpoint: ep_desc 0x0, profile_id 0x104
I (51770) ZBOSS: zcl/zcl_common.c:1057   ep or simple desc for it doesn't exist!
I (51770) ZBOSS: zcl/zcl_common.c:1113   < zb_zcl_is_target_endpoint, match 0
I (51780) ZBOSS: zdo/af_rx.c:459   APS pkt 12 for ep 1 is not recognized - drop
I (52280) ZIGBEE: Found new End Device: status(0), address(0x18af), endpoint(1)
[... cut ...]
221	150.605619	0x0000	Broadcast	ZigBee	59	Route Request, Dst: 0x18af, Src: 0x0000
222	150.905319	0x0000	Broadcast	ZigBee	59	Route Request, Dst: 0x18af, Src: 0x0000
223	151.225320	0x0000	Broadcast	ZigBee	59	Route Request, Dst: 0x18af, Src: 0x0000
224	153.554402		Broadcast	IEEE 802.15.4	10	Beacon Request
225	153.644666	0x0000		ZigBee	28	Beacon, Src: 0x0000, EPID: Espressif_ff:fe:4a:c6:e4
226	156.410808		Broadcast	IEEE 802.15.4	10	Beacon Request
227	156.544774	0x0000		ZigBee	28	Beacon, Src: 0x0000, EPID: Espressif_ff:fe:4a:c6:e4

RVC69, 2023-12-19:
Dear support, here under comparison of dialog between ESP32C6 and a IKEA TRAFRI ON/OFF and AQARA TEMPERATURE sensor
AQUARA never gets callback from esp_zb_zcl_read_attr_cmd_req

--------------------------------------------------------------------------------------------------------------------------------------------------------
IKEA TRADFRI ON/OFF 
I (16592) ZIGBEE: Reading state from local endpoint (0x3), distant endpoint(0x1), address(0xe707), clusterId(0x6), IEEE address: 14:b4:57:ff:fe:c5:08:d7
I (16592) ZBOSS: zcl/zcl_common.c:2209   >> zb_zcl_finish_and_send_packet_common param 2 delay 0
I (16602) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (16612) ZBOSS: zcl/zcl_common.c:1667   >>get_cluster_desc ep_desc 0x0, cluster_id 0x0 cluster_role 1
I (16622) ZBOSS: zcl/zcl_common.c:1682   <<get_cluster_desc cluster_desc 0x0
I (16632) ZBOSS: zcl/zcl_basic_commands.c:144   < zb_zcl_check_is_device_enabled ret 1
I (16642) ZBOSS: zcl/zcl_common.c:2242   extending packet to 5 bytes
I (16642) ZBOSS: zcl/zcl_common.c:1762   >>get_profile_id_by_endpoint endpoint_id = 3
I (16652) ZBOSS: zcl/zcl_common.c:1781   <<get_profile_id_by_endpoint profile_id = 0xffff
I (16662) ZBOSS: zcl/zcl_common.c:1308   cluster_needs_aps_encryption: ep 3 cluster 0x6 res 0
I (16672) ZBOSS: zcl/zcl_common.c:2450   ZCL (out): cluster: 0x6, APS security required: no
I (16682) ZBOSS: zcl/zcl_common.c:650   >> zb_zcl_register_cb param 2, cb 0x0
I (16692) ZBOSS: zcl/zcl_common.c:681   << zb_zcl_register_cb ret 0
I (16692) ZBOSS: zcl/zcl_common.c:1979   ZCL packet: buf 0x2, ref 2, len 5
I (16702) ZBOSS: zcl/zcl_common.c:1980   FC: 0x0 TSN: 0x0 CMD: 0x0
I (16732) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (16732) ZBOSS: zcl/zcl_common.c:1053   > zb_zcl_is_target_endpoint: ep_desc 0x0, profile_id 0x104
I (16742) ZBOSS: zcl/zcl_common.c:1057   ep or simple desc for it doesn't exist!
I (16752) ZBOSS: zcl/zcl_common.c:1113   < zb_zcl_is_target_endpoint, match 0
I (16752) ZBOSS: zdo/af_rx.c:459   APS pkt 3 for ep 3 is not recognized - drop
I (16772) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
************************************************************************************************************************************************************
4	7.948006	0x0000	0xe707	ZigBee	66	Data, Dst: 0xe707, Src: 0x0000
5	7.950501			IEEE 802.15.4	5	Ack
6	7.960161	0xe707	0x0000	ZigBee	53	Data, Dst: 0x0000, Src: 0xe707
7	7.962249			IEEE 802.15.4	5	Ack
8	7.964259	0xe707	0x0000	ZigBee	61	Data, Dst: 0x0000, Src: 0xe707
9	7.966602			IEEE 802.15.4	5	Ack
10	8.008216	0x0000	0xe707	ZigBee	61	Data, Dst: 0xe707, Src: 0x0000
11	8.010552			IEEE 802.15.4	5	Ack

------------------------------------------------------------------------------------------------------------------------------------------------------------
XIAOMI AQUARA TEMPERATURE SENSOR ZIGBEE 1.2
I (16401) ZIGBEE: Reading state from local endpoint (0x1), distant endpoint(0x1), address(0x18af), clusterId(0x402), 
IEEE address: 00:15:8d:00:02:82:ee:2d
I (16401) ZBOSS: zcl/zcl_common.c:2209   >> zb_zcl_finish_and_send_packet_common param 2 delay 0
I (16411) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (16421) ZBOSS: zcl/zcl_common.c:1667   >>get_cluster_desc ep_desc 0x0, cluster_id 0x0 cluster_role 1
I (16431) ZBOSS: zcl/zcl_common.c:1682   <<get_cluster_desc cluster_desc 0x0
I (16431) ZBOSS: zcl/zcl_basic_commands.c:144   < zb_zcl_check_is_device_enabled ret 1
I (16441) ZBOSS: zcl/zcl_common.c:2242   extending packet to 5 bytes
I (16451) ZBOSS: zcl/zcl_common.c:1762   >>get_profile_id_by_endpoint endpoint_id = 1
I (16461) ZBOSS: zcl/zcl_common.c:1781   <<get_profile_id_by_endpoint profile_id = 0xffff
I (16471) ZBOSS: zcl/zcl_common.c:1308   cluster_needs_aps_encryption: ep 1 cluster 0x402 res 0
I (16481) ZBOSS: zcl/zcl_common.c:2450   ZCL (out): cluster: 0x402, APS security required: no
I (16481) ZBOSS: zcl/zcl_common.c:650   >> zb_zcl_register_cb param 2, cb 0x0
I (16491) ZBOSS: zcl/zcl_common.c:681   << zb_zcl_register_cb ret 0
I (16501) ZBOSS: zcl/zcl_common.c:1979   ZCL packet: buf 0x2, ref 2, len 5
I (16511) ZBOSS: zcl/zcl_common.c:1980   FC: 0x0 TSN: 0x0 CMD: 0x0
I (17291) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (17311) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x0
I (17311) ZBOSS: zcl/zcl_common.c:1053   > zb_zcl_is_target_endpoint: ep_desc 0x0, profile_id 0x104
I (17311) ZBOSS: zcl/zcl_common.c:1057   ep or simple desc for it doesn't exist!
I (17331) ZBOSS: zcl/zcl_common.c:1113   < zb_zcl_is_target_endpoint, match 0
I (17331) ZBOSS: zdo/af_rx.c:459   APS pkt 6 for ep 1 is not recognized - drop
************************************************************************************************************************************************************
22	9.136148	0x0000	0x18af	ZigBee	58	Data, Dst: 0x18af, Src: 0x0000
23	9.136359			IEEE 802.15.4	5	Ack
24	9.141104	0x18af	0x0000	ZigBee	45	Data, Dst: 0x0000, Src: 0x18af
25	9.142927			IEEE 802.15.4	5	Ack
26	9.144792	0x18af	0x0000	ZigBee	54	Data, Dst: 0x0000, Src: 0x18af
27	9.146904			IEEE 802.15.4	5	Ack
28	9.151586	0x18af	0x0000	ZigBee	45	Data, Dst: 0x0000, Src: 0x18af
29	9.153418			IEEE 802.15.4	5	Ack
30	9.157771	0xe707	0x0000	ZigBee	77	Command, Dst: 0x0000, Src: 0xe707
31	9.160627			IEEE 802.15.4	5	Ack
32	9.163941	0x18af	0x0000	ZigBee	54	Data, Dst: 0x0000, Src: 0x18af
33	9.166061			IEEE 802.15.4	5	Ack
[... cut ...]
39	9.348721			IEEE 802.15.4	5	Ack
40	9.568516	0xe707	0x0000	ZigBee	77	Command, Dst: 0x0000, Src: 0xe707
41	9.571372			IEEE 802.15.4	5	Ack
42	12.389896	0x18af	0xe707	IEEE 802.15.4	12	Data Request
43	12.390664			IEEE 802.15.4	5	Ack
44	15.426483	0x18af	0xe707	IEEE 802.15.4	12	Data Request
45	15.427251			IEEE 802.15.4	5	Ack
46	18.460564	0x18af	0xe707	IEEE 802.15.4	12	Data Request
47	18.461332			IEEE 802.15.4	5	Ack
48	21.497710	0x18af	0xe707	IEEE 802.15.4	12	Data Request
49	21.498477			IEEE 802.15.4	5	Ack

xieqinan, 2023-12-28:
@RVC69 ,

From the Wireshark log, it appears that the Aqara Zigbee 1.2 device did not respond to the `read_attribute_response`, leading to the failure of triggering `zb_core_action_handler()`. I recommend checking the features and specifications of the Aqara device to identify and address this issues.

RVC69, 2023-12-29:
Dear xieqinan, thanks for your expertise.
best regards

closed 2023-12-29 by RVC69