espressif/esp-zigbee-sdk #183: Boot time with connection is too slow. How to achieve quick boot, send zigbee message and go back to sleep fast?

felipejfc, 2023-12-16:
### Question

So I have some door sensors here that are battery powered zigbee devices, they go to deep sleep (or at least power radio down but should be going to deep sleep given that I can keep them running for like months with 2 aaa batteries). Thing is, when I open the door, this thing will report SUPER fast, like within a second or so. In contrast, the deep sleep example in this SDK takes from 5 to 8 seconds to connect after it wakes, the README of the repo has evidence:

```
I (289) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (8749) ESP_ZB_DEEP_SLEEP: Start network steering
```

8.5 seconds from the moment zb stack initalizes to the

kelin6, 2023-12-18:
@felipejfc I sincerely apologize and forgot to update the latest debugging logs during development. I will update them in the subsequent version updates. After waking up from deep sleep, the connection time is very brief. Below is the updated log.

```
I (392) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM                                                                                                                                                                      
I (399) spi_flash: detected chip: generic                                                                                                                                                                                      
I (403) spi_flash: flash io: dio                                                                                                                                                                                               
W (407) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.                                                                                     
I (421) sleep: Configure to isolate all GPIO pins in sleep state                                                                                                                                                               
I (427) sleep: Enable automatic switching of GPIO sleep configuration                                                                                                                                                          
I (434) main_task: Started on CPU0                                                                                                                                                                                             
I (434) main_task: Calling app_main()                                                                                                                                                                                          
I (444) ESP_ZB_DEEP_SLEEP: Not a deep sleep reset                                                                                                                                                                              
I (444) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 20s                                                                                                                                                                                                                                                                                                                                                                                             
I (454) phy: phy_version: 211,0, 5857fe5, Nov  1 2023, 11:31:09                                                                                                                                                                
I (454) phy: libbtbb version: ce629d6, Nov  1 2023, 11:31:19                                                                                                                                                                   
I (464) main_task: Returned from app_main()                                                                                                                                                                                    
I (594) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL                                                                                                                                               
I (594) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized                                                                                                                                                                            
I (594) ESP_ZB_DEEP_SLEEP: Start network steering                                                                                                                                                                              
I (4514) ESP_ZB_DEEP_SLEEP: Joined network successfully (Extended PAN ID: 60:55:f9:ff:fe:f7:2f:3e, PAN ID: 0x75e7, Channel:13, Short Address: 0xcdfa)                                                                          
I (4524) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep                                                                                                                                                
I (9524) ESP_ZB_DEEP_SLEEP: Enter deep sleep    
ESP-ROM:esp32h2-20221101                                                                                                                                                                                                       
Build:Nov  1 2022                                                                                                                                                                                                              
rst:0x5 (SLEEP_WAKEUP),boot:0x1c (SPI_FAST_FLASH_BOOT)                                                                                                                                                                         
SPIWP:0xee                                                                                                                                                                                                                     
[... cut ...]
I (287) main_task: Returned from app_main()
I (297) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (297) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (967) ESP_ZB_DEEP_SLEEP: Start network steering
I (1087) ESP_ZB_DEEP_SLEEP: Joined network successfully (Extended PAN ID: 60:55:f9:ff:fe:f7:2f:3e, PAN ID: 0x75e7, Channel:13, Short Address: 0xcdfa)
I (1087) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep
I (6097) ESP_ZB_DEEP_SLEEP: Enter deep sleep

```

[image]

felipejfc, 2023-12-18:
hey @kelin6 thanks for the answer. with which set-up did you achieve this connection time? the code on the repo here without changes?

kelin6, 2023-12-19:
@felipejfc  The above log is generated using esp-zigbee-sdk version 1.0.6 without any additional code modifications. Feel free to test it.

chshu, 2024-01-08:
@felipejfc please reopen if any further quesions.

closed 2024-01-08 by chshu

susch19, 2024-02-25:
I would like to reopen this issue, since the problem still persists for me with the ESP32-C6 as an End Device.
No matter which example I try, the time between `Zigbee stack initialized` and ability to join a network is always ~3 seconds. Doesn't matter which coordinator I use. Since I don't have an ESP32-H2, I don't know if this is a C6 problem.
```
I (409) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (449) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (459) main_task: Returned from app_main()
I (469) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (469) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (3299) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (3299) ESP_ZB_DEEP_SLEEP: Device rebooted
```

If I start the ESP32-C6 without a known network, the network steering is able to start a few ms after Zigbee stack has been initialized
```
I (571) main_task: Returned from app_main()
I (573) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (573) ESP_ZB_SLEEP: Zigbee stack initialized
I (581) ESP_ZB_SLEEP: Device started up in  factory-reset mode
I (584) ESP_ZB_SLEEP: Start network steering
```

lxz946786639, 2024-08-24:
> 我想重新打开这个问题，因为使用 ESP32-C6 作为终端设备时，问题仍然存在。无论我尝试哪个示例，加入网络之间的时间始终为 ~3 秒。我使用哪个 coordinator 并不重要。由于我没有 ESP32-H2，所以我不知道这是不是 C6 的问题。`Zigbee stack initialized`
> 
> ```
> I (409) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
> I (449) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
> I (459) main_task: Returned from app_main()
> I (469) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
> I (469) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
> I (3299) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
> I (3299) ESP_ZB_DEEP_SLEEP: Device rebooted
> ```
> 
> 如果我在没有已知网络的情况下启动 ESP32-C6，则网络转向能够在 Zigbee 堆栈初始化后几毫秒内启动
> 
> ```
> I (571) main_task: Returned from app_main()
> I (573) ESP_ZB_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
> I (573) ESP_ZB_SLEEP: Zigbee stack initialized
> I (581) ESP_ZB_SLEEP: Device started up in  factory-reset mode
> I (584) ESP_ZB_SLEEP: Start network steering
> ```

The same problem occurs with ESP32H2
```
I (266) ESP_ZB_DEEP_SLEEP: Wake up from timer. Time spent in deep sleep and boot: 32272ms
I (266) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 32s

I (286) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (286) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (306) main_task: Returned from app_main()
I (316) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (316) ESP_ZB_DEEP_SLEEP: Initialize Zigbee stack
I (2696) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (2696) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep
I (2706) ESP_ZB_DEEP_SLEEP: Device rebooted
```

lxz946786639, 2024-08-24:
> 我想重新打开这个问题，因为使用 ESP32-C6 作为终端设备时，问题仍然存在。无论我尝试哪个示例，加入网络之间的时间始终为 ~3 秒。我使用哪个 coordinator 并不重要。由于我没有 ESP32-H2，所以我不知道这是不是 C6 的问题。`Zigbee stack initialized`
>
> ```
> I (409) phy_init: phy_version [...]

The same problem occurs with ESP32H2

```
I (266) ESP_ZB_DEEP_SLEEP: Wake up from timer. Time spent in deep sleep and boot: 32272ms
I (266) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 32s

I (286) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (286) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (306) main_task: Returned from app_main()
I (316) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (316) ESP_ZB_DEEP_SLEEP: Initialize Zigbee stack
I (2696) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (2696) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep
I (2706) ESP_ZB_DEEP_SLEEP: Device rebooted
```

lxz946786639, 2024-08-24:
> I would like to reopen this issue, since the problem still persists for me with the ESP32-C6 as an End Device. No matter which example I try, the time between `Zigbee stack initialized` and ability [...]

The same problem occurs with ESP32H2

```
I (266) ESP_ZB_DEEP_SLEEP: Wake up from timer. Time spent in deep sleep and boot: 32272ms
I (266) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 32s

I (286) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (286) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (306) main_task: Returned from app_main()
I (316) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (316) ESP_ZB_DEEP_SLEEP: Initialize Zigbee stack
I (2696) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (2696) ESP_ZB_DEEP_SLEEP: Start one-shot timer for 5s to enter the deep sleep
I (2706) ESP_ZB_DEEP_SLEEP: Device rebooted
```