espressif/esp-zigbee-sdk #184: Getting only 1 active endpoint instead of 4

munish-nuos, 2023-12-18:
### Question

I am using following function by espressif to add multiple endpoints with server cluster:
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *esp_zb_server_cluster_list = server_cluster();

    for(int i=0; i<MAX_EP_SIZE; i++){
       esp_zb_ep_list_add_ep(esp_zb_ep_list,
    		   esp_zb_server_cluster_list,
    		   ENDPOINTS_LIST[i],
    		   ESP_ZB_AF_HA_PROFILE_ID,
			   ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
       ESP_LOGI(TAG, "ENDPOINTS_LIST[%d]= %d",i,ENDPOINTS_LIST[i]);
    }
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);

   Here i defined uint8_t ENDPOINTS_LIST[4] = {1, 2, 3, 4};
  
But i found only 1 endpoint and you can see below "endpoint count(1)".
Found light
I (10166) ESP_ZB_SWITCH: Light Endpoint(1)
I (10196) ESP_ZB_SWITCH: Active endpoint response: status(0) and endpoint count(1)
I (10196) ESP_ZB_SWITCH: Endpoint ID List: 1
I (10256) ESP_ZB_SWITCH: Try to bind Light 1 On/Off
I (10256) ESP_ZB_SWITCH: Try to bind Light 2 On/Off
I (10256) ESP_ZB_SWITCH: Try to bind Light 3 On/Off
I (10266) ESP_ZB_SWITCH: Try to bind Light 4 On/Off
I (10286) ESP_ZB_SWITCH: Bound successfully!
I (10286) ESP_ZB_SWITCH: The light originating from address(0x0) on endpoint(1)
I (10286) ESP_ZB_SWITCH: Bound successfully!
I (10296) ESP_ZB_SWITCH: The light originating from address(0x0) on endpoint(2)
I (10296) ESP_ZB_SWITCH: Bound successfully!
I (10306) ESP_ZB_SWITCH: The light originating from address(0x0) on endpoint(3)
I (10316) ESP_ZB_SWITCH: Bound successfully!
I (10316) ESP_ZB_SWITCH: The light originating from address(0x0) on endpoint(4)

I am using below lines to bind endpoints:
 uint16_t cluster_list[] = {ESP_ZB_ZCL_CLUSTER_ID_ON_OFF};
  find_req.dst_nwk_addr = 0x0000;
  find_req.addr_of_interest = 0x0000;
  find_req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
  find_req.num_in_clusters = 1;
  find_req.num_out_clusters = 0;
  find_req.cluster_list = cluster_list;
  esp_zb_zdo_find_on_off_light(&find_req, user_find_cbx, NULL);

Where i am wrong?

munish-nuos, 2023-12-22:
I got some point in  function  esp_zb_device_register(esp_zb_ep_list);
when i watch the log "**zb_zcl_add_cluster_handlers cluster ret 0**", the endpoint1 is registered successfully. 
But other endpoints 2,3,4, i got logs, "**zb_zcl_add_cluster_handlers cluster ret -27**". I think espressif has some issue in their function "esp_zb_device_register". 
Still have issue, Please help me to resolve this issue.

xieqinan, 2023-12-28:
Hello @munish-nuos ,

Regarding the issue with the logs "zb_zcl_add_cluster_handlers cluster ret -27" for endpoints 2, 3, and 4, it is not a bug in `esp_zb_device_register()`. The log only indicates that the cluster is registered repeatedly, which does not affect the endpoint count.

I have also tested the [esp_HA_customized_light example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c#L173) using a similar approach, and it works fine. I replaced the `esp_zb_ep_list_add_ep` call with either:

```c
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT + 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT + 2, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT + 3, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
```

or

```c
uint8_t ENDPOINTS_LIST[4] = {1, 2, 3, 4};
for (int i = 0; i < 4; i++) {
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, ENDPOINTS_LIST[i], ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
    ESP_LOGI(TAG, "ENDPOINTS_LIST[%d]= %d", i, ENDPOINTS_LIST[i]);
}
```

Then, using the [esp_HA_customized_switch example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c) to send the `active_ep_request`, the log indicates a successful response with the endpoint count.

```
I (779) ESP_HA_ON_OFF_SWITCH: Active endpoint response: status(0) and endpoint count(4)
I (789) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 1
I (789) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 2
I (789) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 3
I (799) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 4
```

I am not sure what I might be missing. Could you please follow the same steps to test the issue?

munish-nuos, 2023-12-29:
> Hello @munish-nuos ,
>
> Regarding the issue with the logs "zb_zcl_add_cluster_handlers cluster ret -27" for endpoints 2, 3, and 4, it is not a bug in `esp_zb_device_register()`. The log only indicates [...]

Thank you @xieqinan for reply,
Actually i have this issue with third party gateways, like tuya and sonoff. Can you please check with these gateways.

chshu, 2024-01-08:
@mundevx The device side should already create the endpoints correctly, could it be a similar issue regarding to the specific endpoint requirement https://github.com/espressif/esp-zigbee-sdk/issues/173#issuecomment-1853336934?

Could you check this issue with the gateway vendors?

closed 2024-01-08 by chshu