espressif/esp-zigbee-sdk #185: 4 lights endpoints are not adding in Tuya Gateway

munish-nuos, 2023-12-18:
### Question

I am using this routine to add endpoints:

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *esp_zb_server_cluster_list = server_cluster();

    for(int i=0; i<MAX_EP_SIZE; i++){
       esp_zb_ep_list_add_ep(esp_zb_ep_list,
    		   esp_zb_server_cluster_list,
    		   ENDPOINTS_LIST[i],
    		   ESP_ZB_AF_HA_PROFILE_ID,
			   ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
       ESP_LOGI(TAG, "ENDPOINTS_LIST[%d]= %d",i,ENDPOINTS_LIST[i]);
    }
    esp_zb_device_register(esp_zb_ep_list);

this code working perfectely in Ewelink app(Sonoff), showing 4 light switches, but not in Smart life app(Tuya).

Please help me

munish-nuos, 2023-12-18:
If i set MAX_EP_SIZE = 3, its working. more than 3 not showing on Smart Life (Tuya) App.

chshu, 2023-12-19:
It works with Ewelink app(Sonoff), but fails with Smart life app(Tuya). Should it be some issue with Tuya app?

We haven't tested such details with these ecosystems. As you may know, although Zigbee is a standard protocol, but unfortunately there are more or less some differences among Zigbee ecosystems.

munxerpor, 2023-12-20:
> It works with Ewelink app(Sonoff), but fails with Smart life app(Tuya). Should it be some issue with Tuya app?
>
> We haven't tested such details with these ecosystems. As you may know, although Zigbee [...]

Thank you for the reply Sir,

Yes, May be some issue with Tuya app. But i am also getting another issue, I am getting only 1 active endpoint instead of 4, during sending request esp_zb_zdo_find_on_off_light after joining network. Can you help me over this.

chshu, 2024-01-08:
duplicate issue, @mundevx please check with the gataway vendor (https://github.com/espressif/esp-zigbee-sdk/issues/184#issuecomment-1880598546)

closed 2024-01-08 by chshu