espressif/esp-zigbee-sdk #186: Active endpoint issue with Tuya or Sonoff gateway

munxerpor, 2023-12-20:
### Question

Hello Everyone,

I am using Tuya or Sonoff gateway with my espressif end device.
I am using esp_zb_ep_list_add_ep to add four same endpoints as light with endpoint values 1, 2, 3, 4. 
After successfully connected & registering to network, when i call esp_zb_zdo_find_on_off_light(&find_req, user_find_cb, NULL), it provide me only 1 ACTIVE endpoint. But my code works with some custom Gateway, showing 4 ACTIVE endpoints but not with Tuya or Sonoff. Please correct me where i am wrong.

xieqinan, 2023-12-21:
Hello @munxerpor ,

Please use the `esp_zb_zdo_active_ep_req()` function to find the active endpoints in the Zigbee network.

munish-nuos, 2023-12-22:
> Hello @munxerpor ,
> 
> Please use the `esp_zb_zdo_active_ep_req()` function to find the active endpoints in the Zigbee network.

Hi @xieqinan , I am using  esp_zb_zdo_active_ep_req() , but it returns only 1 Active endpoint. As i am adding extra 3 ep in the list in case of Tuya or Sonoff gateway, but with espressif coordinator or some custom gateway, i am getting all 4 active endpoints.

Please help me.

chshu, 2024-01-08:
@munxerpor duplicate issues:
https://github.com/espressif/esp-zigbee-sdk/issues/184, https://github.com/espressif/esp-zigbee-sdk/issues/185

closed 2024-01-08 by chshu