espressif/esp-zigbee-sdk #187: After waking up from deep sleep, Zigbee initialization fails

LimeSea, 2023-12-20:
### Question

I tried to use deep sleep in zigbee. When I connected for the first time, I entered deep sleep normally. However, after waking up again, `zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL) `appeared. I saw an error in the phy initialization part above: 
`
I (268) phy: error: pll_cal exceeds 2ms!!!
I (278) phy: error: pll_cal exceeds 2ms!!!
`
The following is the environment I use：
* ESP_IDF version: 5.2.0
* zigbee version: 1.0.6
* zboss version: 1.0.6
* soc: esp32h2

The following is my code to initialize deep sleep：
```
static void zb_deep_sleep_init(void)
{
    const esp_timer_create_args_t s_oneshot_timer_args = {
            .callback = &s_oneshot_timer_callback,
            .name = "one-shot"
    };
    ESP_ERROR_CHECK(esp_timer_create(&s_oneshot_timer_args, &s_oneshot_timer));
    struct timeval now;
    gettimeofday(&now, NULL);
    int sleep_time_ms = (now.tv_sec - s_sleep_enter_time.tv_sec) * 1000 + (now.tv_usec - s_sleep_enter_time.tv_usec) / 1000;
    esp_sleep_wakeup_cause_t wake_up_cause = esp_sleep_get_wakeup_cause();
    switch (wake_up_cause) {
    case ESP_SLEEP_WAKEUP_TIMER: {
        ESP_LOGI(TAG, "Wake up from timer. Time spent in deep sleep and boot: %dms", sleep_time_ms);
        break;
    }
    case ESP_SLEEP_WAKEUP_UNDEFINED:
    default:
        ESP_LOGI(TAG, "Not a deep sleep reset");
        break;
    }
    const int wakeup_time_sec = 20;
    ESP_LOGI(TAG, "Enabling timer wakeup, %ds\n", wakeup_time_sec);
    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(wakeup_time_sec * 1000000));
}
```
The log when the problem occurs is as follows：
```
I (23) boot: ESP-IDF v5.2-dev-3903-g66992aca7a-dirty 2nd stage bootloader
I (24) boot: compile time Dec 20 2023 18:36:39
I (25) boot: chip revision: v0.2
I (29) boot.esp32h2: SPI Speed      : 64MHz
I (33) boot.esp32h2: SPI Mode       : DIO
I (38) boot.esp32h2: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 000a2800
I (82) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (89) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=110f8h ( 69880) map
I (131) esp_image: segment 1: paddr=00021120 vaddr=40800000 size=06ef8h ( 28408) load
I (141) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=789f4h (494068) map
I (293) esp_image: segment 3: paddr=000a0a1c vaddr=40806ef8 size=0719ch ( 29084) load
I (304) esp_image: segment 4: paddr=000a7bc0 vaddr=4080e0a0 size=016ech (  5868) load
I (307) esp_image: segment 5: paddr=000a92b4 vaddr=50000000 size=00064h (   100) load
I (310) esp_image: segment 6: paddr=000a9320 vaddr=50000068 size=00010h (    16) load
I (322) boot: Loaded app from partition at offset 0x10000
I (324) boot: Disabling RNG early entropy source...
I (341) cpu_start: Unicore app
W (350) clk: esp_perip_clk_init() has not been implemented yet
I (357) cpu_start: Pro cpu start user code
I (357) cpu_start: cpu freq: 96000000 Hz
I (358) cpu_start: Application information:
I (360) cpu_start: Project name:     hello_world
I (365) cpu_start: App version:      2b32028-dirty
I (371) cpu_start: Compile time:     Dec 20 2023 19:31:49
I (377) cpu_start: ELF file SHA256:  a59fe52e2...
I (382) cpu_start: ESP-IDF:          v5.2-dev-3903-g66992aca7a-dirty
I (389) cpu_start: Min chip rev:     v0.0
I (394) cpu_start: Max chip rev:     v0.99 
I (399) cpu_start: Chip rev:         v0.2
I (403) heap_init: Initializing. RAM available for dynamic allocation:
I (411) heap_init: At 40816DE0 len 000365A0 (217 KiB): RAM
I (417) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (424) spi_flash: detected chip: generic
I (428) spi_flash: flash io: dio
W (431) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (445) sleep: Configure to isolate all GPIO pins in sleep state
I (451) sleep: Enable automatic switching of GPIO sleep configuration
I (459) main_task: Started on CPU0
I (459) main_task: Calling app_main()
I (469) zigbee: Not a deep sleep reset
I (479) zigbee: Enabling timer wakeup, 20s

I (479) phy: phy_version: 211,0, 9435a66, Oct 12 2023, 11:16:11
I (489) phy: libbtbb version: f6a6425, Oct 12 2023, 11:16:27
I (499) main_task: Returned from app_main()
I (509) zigbee: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (509) zigbee: Zigbee stack initialized
I (8979) zigbee: Start network steering
I (8999) zigbee: Joined network successfully (Extended PAN ID: 44:93:cc:d7:5f:aa:d5:de, PAN ID: 0xce5e, Channel:11)
I (8999) zigbee: Start one-shot timer for 10s to enter the deep sleep
I (18999) zigbee: Enter deep sleep
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x5 (SLEEP_WAKEUP),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x1878
load:0x4083cfd0,len:0xe90
load:0x4083efd0,len:0x2e14
entry 0x4083cfda
I (24) boot: ESP-IDF v5.2-dev-3903-g66992aca7a-dirty 2nd stage bootloader
I (25) boot: compile time Dec 20 2023 18:36:39
I (26) boot: chip revision: v0.2
I (29) boot.esp32h2: SPI Speed      : 64MHz
I (34) boot.esp32h2: SPI Mode       : DIO
I (39) boot.esp32h2: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (49) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=110f8h ( 69880) map
I (57) esp_image: segment 1: paddr=00021120 vaddr=40800000 size=06ef8h ( 28408) load
I (68) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=789f4h (494068) map
I (74) esp_image: segment 3: paddr=000a0a1c vaddr=40806ef8 size=0719ch ( 29084) load
I (85) esp_image: segment 4: paddr=000a7bc0 vaddr=4080e0a0 size=016ech (  5868) load
I (91) esp_image: segment 5: paddr=000a92b4 vaddr=50000000 size=00064h (   100) 
I (99) esp_image: segment 6: paddr=000a9320 vaddr=50000068 size=00010h (    16) 
I (107) boot: Fast booting app from partition at offset 0x10000
I (113) boot: Disabling RNG early entropy source...
I (130) cpu_start: Unicore app
W (140) clk: esp_perip_clk_init() has not been implemented yet
I (146) cpu_start: Pro cpu start user code
I (147) cpu_start: cpu freq: 96000000 Hz
I (147) cpu_start: Application information:
I (149) cpu_start: Project name:     hello_world
I (154) cpu_start: App version:      2b32028-dirty
I (160) cpu_start: Compile time:     Dec 20 2023 19:31:49
I (166) cpu_start: ELF file SHA256:  a59fe52e2...
I (171) cpu_start: ESP-IDF:          v5.2-dev-3903-g66992aca7a-dirty
I (178) cpu_start: Min chip rev:     v0.0
I (183) cpu_start: Max chip rev:     v0.99 
I (188) cpu_start: Chip rev:         v0.2
I (193) heap_init: Initializing. RAM available for dynamic allocation:
I (200) heap_init: At 40816DE0 len 000365A0 (217 KiB): RAM
I (206) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (213) spi_flash: detected chip: generic
I (217) spi_flash: flash io: dio
W (221) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (234) sleep: Configure to isolate all GPIO pins in sleep state
I (241) sleep: Enable automatic switching of GPIO sleep configuration
I (248) main_task: Started on CPU0
I (248) main_task: Calling app_main()
I (258) zigbee: Wake up from timer. Time spent in deep sleep and boot: 20268ms
I (268) zigbee: Enabling timer wakeup, 20s

I (268) phy: error: pll_cal exceeds 2ms!!!
I (278) phy: error: pll_cal exceeds 2ms!!!
I (278) phy: phy_version: 211,0, 9435a66, Oct 12 2023, 11:16:11
I (278) phy: libbtbb version: f6a6425, Oct 12 2023, 11:16:27
I (298) main_task: Returned from app_main()
I (308) zigbee: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (308) zigbee: Zigbee stack initialized
W (8708) zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL)
W (18148) zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL)
W (27588) zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL)
W (37028) zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL)
```

kelin6, 2024-01-12:
@LimeSea The provided error message `I (268) phy: error: pll_cal exceeds 2ms!!! ` indicates that the RF is not functioning properly, leading to issues with Zigbee network initialization after returning from deep sleep. This is reflected in the log `W (37028) zigbee: Failed to initialize Zigbee stack (status: ESP_FAIL)`

However, based on the environment you provided, I was not able to reproduce the issue. Do you have another ESP32H2 available? You may try using a different ESPH2 and running the deep sleep test again to see if the issue persists.

LimeSea, 2024-01-12:
@kelin6 I tried re-cloning esp-idf and the problem has been solved. Maybe there was a problem with the component cloning of my esp-idf.

kelin6, 2024-01-12:
@LimeSea I'm delighted to hear that you've resolved it. That's great news!

xieqinan, 2024-03-08:
@LimeSea ,

Could you please update the esp-zigbee-sdk to v1.2.0 to test this issue? If it has been fixed, please close this issue.

closed 2024-03-08 by LimeSea