espressif/esp-zigbee-sdk #188: OOM and then crash in zboss probably related to ED aging

wisemanny, 2023-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-342-gbcf1645e44

### esp-zigbee-lib version.

 1.0.6

### esp-zboss-lib version.

 1.0.6

### Espressif SoC revision.

esp32h2

### What is the expected behavior?

No crash

### What is the actual behavior?

ED crashes and restarts

### Steps to reproduce.

The device is ED connected to Tuya gateway, 2 gang switch. Taken from customized_switch example and adjusted to work with Tuya.

```
1. It seems to start with a bunch of such messages (like 100 or so):

I (9462573) ZBOSS: common/zb_scheduler.c:865   alarms q of size 58 q size 60                                                                                                                                    I (9462583) ZBOSS: common/zb_scheduler.c:870   q ent 13 func 0x42038c7a runtime 616054 param 0                                                                                                                  I (9462583) ZBOSS: common/zb_scheduler.c:870   q ent 2 func 0x42038c7a runtime 616055 param 0                                                                                                                   I (9462593) ZBOSS: common/zb_scheduler.c:870   q ent 21 func 0x42038c7a runtime 616056 param 0                                                                                                                  I (9462603) ZBOSS: common/zb_scheduler.c:870   q ent 17 func 0x42038c7a runtime 616057 param 0                                                                                                                  I (9462613) ZBOSS: common/zb_scheduler.c:870   q ent 58 func 0x42038c7a runtime 616058 param 0                                                                                                                  I (9462623) ZBOSS: common/zb_scheduler.c:870   q ent 4 func 0x42038c7a runtime 616059 param 0        

2. Then comes

I (9463623) ZBOSS: common/zb_scheduler.c:870   q ent 44 func 0x4204e2fe runtime 619996 param 0                                                                                                                  I (9463633) ZBOSS: common/zb_scheduler.c:870   q ent 14 func 0x4202b1c2 runtime 638378 param 0                                                                                                                  I (9463643) ZBOSS: nwk/nwk_nlme.c:794   OOM state is detected. ACHTUNG: Out of memory!                                                                                                                          I (9463653) ZBOSS: common/zb_bufpool_mult.c:1607   OOM! bufs allocated {21 i-o 28} max 60                                                                                                                       I (9463653) ZBOSS: nwk/nwk_nlme.c:836   ACHTUNG: Sending OOM status                                                                                                                                             I (9463683) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9463683) ZBOSS: zdo/zdo_app.c:1004   Failed zb_buf_get_out_delayed [-1]                                                                                                                                      I (9463693) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9463693) ZBOSS: zdo/zdo_app.c:1004   Failed zb_buf_get_out_delayed [-1]                                                                                                                                      I (9463703) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9463713) ZBOSS: zdo/zdo_app.c:1004   Failed zb_buf_get_out_delayed [-1]                                                                                                                                      I (9463723) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9463723) ZBOSS: zdo/zdo_app.c:1004   Failed zb_buf_get_out_delayed [-1]                                                                                                                                      I (9463733) ZBOSS: common/zb_scheduler.c:865   alarms q of size 58 q size 60                                                                                                                                    I (9463743) ZBOSS: common/zb_scheduler.c:870   q ent 15 func 0x420711b2 runtime 616190 param 0                                                                                                                  I (9463753) ZBOSS: common/zb_scheduler.c:870   q ent 7 func 0x42038c7a runtime 616206 param 0                                                                                                                   I (9463763) ZBOSS: common/zb_scheduler.c:870   q ent 19 func 0x42038c7a runtime 616218 param 0                                                                                                                  I (9463763) ZBOSS: common/zb_scheduler.c:870   q ent 53 func 0x42038c7a runtime 616221 param 0

3. Then again runtime ... param ....
4. Then comes this:

I (9465363) ZBOSS: common/zb_scheduler.c:870   q ent 14 func 0x4202b1c2 runtime 638378 param 0                                                                                                                  I (9465383) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9465383) ZBOSS: nwk/zb_nwk_ed_aging.c:99   Failed zb_buf_get_out_delayed [-1]                                                                                                                                I (9465393) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      E (9465403) ZB_ESP_MAC: zb mac received buffer full!                                                                                                                                                            I (9465393) ZBOSS: nwk/zb_nwk_ed_aging.c:99   Failed zb_buf_get_out_delayed [-1]                                                                                                                                I (9465413) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      E (9465423) ZB_ESP_MAC: zb mac received buffer full!                                                                                                                                                            I (9465423) ZBOSS: nwk/zb_nwk_ed_aging.c:99   Failed zb_buf_get_out_delayed [-1]                                                                                                                                I (9465433) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is full                                                                                                                                      I (9465443) ZBOSS: nwk/zb_nwk_ed_aging.c:99   Failed zb_buf_get_out_delayed [-1]                                                                                                                                I (9465453) ZBOSS: common/zb_bufpool_mult.c:393   delayed buf q[0] is ful [... cut ...] a696867  T5      : 0x66656463  T6      : 0x62613938                                                                                                                          MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000                                                                                                                          MHARTID : 0x00000000    
```

Happens almost once a day or a few times a day. Let me know if you need more information.

xieqinan, 2023-12-21:
@wisemanny,

This issue appears to be triggered by numerous callbacks that are stored in memory but are not executed immediately. To pinpoint the command causing the extended processing time, it would be helpful to have the command communication list within the Zigbee network. Providing the Wireshark packet (.pcapng) for this issue would be beneficial.

Could you confirm whether the `esp_zb_scheduler_alarm()` function is called in your application?

Additionally, could you confirm whether the device has a substantial amount of data to handle?

wisemanny, 2023-12-21:
1. I will get the sniffer in a few days or a week and then can do it.
2. Yes, I do use esp_zb_scheduler_alarm as a way to update the attribute, but it is being called 1-2 times a day. The device is still in development mode and is not used often. I am observing how it works in the long run.
3. Not a lot of data at all. It is a 2-gang switch and as I said, I send commands to it 1-2 times a day now. I see that most of the time, it disconnects and reconnects to the gateway.

xieqinan, 2023-12-28:
@wisemanny ,

Using `esp_zb_scheduler_alarm()` as a timer for triggering events that will be executed after a long time is not an optimal approach. It occupies the limited-size schedule queue for an extended period, potentially leading to issues. I believe that this problem is triggered by this usage. Could you please count how many times `esp_zb_scheduler_alarm()` is called and how many times the callback registered in the scheduler is executed before the crash occurs?

wisemanny, 2023-12-29:
@xieqinan will do. Some of the notes are:

1. I do use esp_zb_scheduler_alarm in such a way - update switch attr to On, wait 3 seconds, update to Off. Here is the full code of the callback:

```
static void update_light_status_cb(uint8_t param)
  {
      bool g_switch_state = (param == 1);
      vTaskDelay(100 / portTICK_PERIOD_MS);

      ESP_LOGI(TAG, "Report back the value for endpoint %d, led: %d", param, g_switch_state);

      // Report back the value
      if (param == 1) {
          // Indicate we open the gate
          ESP_ERROR_CHECK(led_strip_set_pixel(g_led, 0, 0, 5, 0));
      } else {
          // Indiate we close the gate
          ESP_ERROR_CHECK(led_strip_set_pixel(g_led, 0, 5, 0, 0));
      }
      ESP_ERROR_CHECK(led_strip_refresh(g_led));

      esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(
              param,
              ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
              ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
              ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
              &g_switch_state,
              false);
      if(state != ESP_ZB_ZCL_STATUS_SUCCESS) {
          ESP_LOGE(TAG, "Setting switch status attribute failed!");
          return;
      }

      vTaskDelay(3000 / portTICK_PERIOD_MS);

      // TODO
      // Handle GPIOs to pull the collector low

      // Report back the status
      ESP_ERROR_CHECK(led_strip_clear(g_led));
      g_switch_state = !g_switch_state;

      // TODO
      // Handle GPIOs to pull the collector high

      ESP_LOGI(TAG, "Report back the value for endpoint %d, led: %d", param, g_switch_state);

      state = esp_zb_zcl_set_attribute_val(
              param,
              ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
              ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
              ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
              &g_switch_state,                                                                                                                                                                                                false);                                                                                                                                                                                                 if(state != ESP_ZB_ZCL_STATUS_SUCCESS) {                                                                                                                                                                            ESP_LOGE(TAG, "Setting switch status attribute after delay failed!");                                                                                                                                           return;                                                                                                                                                                                                     }                                                                                                                                                                                                           }                                                                        
```

2. I do have pcap files, but they are not linked to the crash. They reflect on the situation when the device is idling - i.e. no commands are sent to it, it just keeps the connection to the hub. Are you interested in those?

3. I will stress test it to see when and how it will crash. Will do in 1-2 weeks.

wisemanny, 2024-01-01:
OK, roughly after 15 calls it starts to send warning messages such as:

I (314988) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 6 cluster_role 1                                                                                                              I (314998) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 6 cluster_role 1                                                                                                          I (315008) ZBOSS: nwk/nwk_nlme.c:794   OOM state is detected. ACHTUNG: Out of memory!                                                                                                                           I (315018) ZBOSS: common/zb_bufpool_mult.c:1607   OOM! bufs allocated {28 i-o 5} max 60                                                                                                                         I (315018) ZBOSS: nwk/nwk_nlme.c:836   ACHTUNG: Sending OOM status                                                                                                                                              I (315038) ZBOSS: nwk/nwk_main.c:3688   Put 12 into input queue (blocked by 3)                                                                                                                                  I (315038) ZBOSS: nwk/nwk_main.c:3688   Put 11 into input queue (blocked by 3)                                                                                                                                  I (315048) ZBOSS: nwk/nwk_main.c:3688   Put 13 into input queue (blocked by 3)                                                                                                                                  I (315058) ZBOSS: nwk/nwk_main.c:3688   Put 4 into input queue (blocked by 3)                                                                                                                                   I (315068) ZBOSS: nwk/nwk_main.c:3688   Put 15 into input queue (blocked by 3)                                                                                                                                  I (315068) ZBOSS: nwk/nwk_main.c:3688   Put 18 into input queue (blocked by 3)                                                                                                                                  I (315078) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 12 from in queue                                                                                                                       I (315088) ZBOSS: nwk/nwk_main.c:3688   Put 19 into input queue (blocked by 12)                                                                                                                                 I (315098) ZBOSS: nwk/nwk_main.c:3688   Put 26 into input queue (blocked by 12)                                                                                                                                 I (315108) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 11 from in queue                                                                                                                       I (315108) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 13 from in queue                                                                                                                       I (315118) ZBOSS: nwk/nwk_nlme.c:857   OOM state ends                                                                                                                                                           I (315128) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 4 from in queue                                                                                                                        I (315138) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 15 from in queue
I (315148) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 18 from in queue                                                                                                                       I (315148) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 19 from in queue                                                                                                                       I (315158) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 26 from in queue
I (315618) hormann_door: zb_action_handler, callback_id is 0                                                                                                                                                    I (315618) hormann_door: zb_attribute_handler                                                                                                                                                                   I (315618) hormann_door: Received message: endpoint(2), cluster(0x6), attribute(0x0), data size(1)               

Before there was this:

I (229508) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 6 cluster_role 1                                                                                                              I (229508) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 6 cluster_role 1                                                                                                          I (229528) ZBOSS: nwk/nwk_nlme.c:794   OOM state is detected. ACHTUNG: Out of memory!                                 [... cut ...] mon/zb_bufpool_mult.c:1607   OOM! bufs allocated {28 i-o 5} max 60                                                                                                                         I (229538) ZBOSS: nwk/nwk_nlme.c:836   ACHTUNG: Sending OOM status                                                                                                                                              I (229558) ZBOSS: nwk/nwk_nlme.c:857   OOM state ends   

Let me know if I you want me to send more data.

xieqinan, 2024-02-06:
@wisemanny ，

Does the crashing occur again after removing the alarm?

wisemanny, 2024-02-06:
I switched to another job so I did not pay attention recently. Let's close the issue for now, when I get more details - I will reopen it.

wisemanny, 2024-02-06:
@xieqinan actually, what is the right way to update the state of the switch? I used esp_zb_scheduler_alarm called from zb_action_handler where I would update the state of the switch. What I want to do is when a switch was presssed in the app, I want to to go "on" and "off" in 3 seconds. So in the esp_zb_scheduler_alarm I would use esp_zb_zcl_set_attribute_val to change it to "on", sleep 3s and then to "off".

Now I decided to make a new rtos task. So in the zb_action_handler I start a new task using xTaskCreate, and in the task I call esp_zb_zcl_set_attribute_val. It crashes now with this error:

`
I (6781) door_opener: Report back the value for endpoint 1, state: 1
I (6781) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 6 cluster_role 1
I (6791) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 6 cluster_role 1                                                                                                            
I (6791) ZBOSS: common/zb_debug.c:64   Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/zcl/zcl_general_commands.c:607                                                                                                                                                                                                                                                                                
abort() was called at PC 0x420208cd on core 0                                                                                                                                                                   
0x420208cd: zb_assert at ??:?                                                                                                                                                 
`
What is the right way to update the attribute from a different rtos task? Should I call esp_zb_scheduler_alarm from the task?

wisemanny, 2024-02-15:
@xieqinan here is another crash. The device is ED and is idling. No commands are sent, no data received (not a sensor) - it is a switch that is not used because I am away. Here is the log:

A number of messsages like this:
```

I (44894023) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (46154203) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (46742043) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (47594703) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (48246803) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (49045193) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (49592433) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (50145813) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (50878513) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (51584523) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (52242403) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
....
I (109416823) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (110344403) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (110709013) ZBOSS: nwk/nwk_main.c:2280   Drop packet with improper protocol version: 0x5
I (110846553) ZBOSS: common/zb_debug.c:64   Assertion failed /home/xiaqilin/esp/esp-zboss/components/zboss_stack/zboss/common/zb_bufpool_mult.c:1022

abort() was called at PC 0x420232fb on core 0
0x420232fb: zb_assert at ??:?
```

Seems like out of nowhere.

xieqinan, 2024-03-11:
@wisemanny ，

This issue appears to be triggered by thread safety concerns. If the project includes other tasks utilizing the `esp_zb` API, please utilize `esp_zb_lock_acquire()` and `esp_zb_lock_release()` to mitigate potential thread safety issues.

xieqinan, 2024-04-15:
@wisemanny ，

We have already released the esp-zigbee-sdk v1.2.3, which includes some fixes for this issue. Could you please test it again with the new version?

xieqinan, 2024-06-17:
@wisemanny ,

Do you have any updates on this issue? If it has been fixed, please consider closing it.

wisemanny, 2024-06-18:
Sorry for the delay, I have put the device in garage already and it is not convenient for now to experiment with it. let's consider this as closed.

closed 2024-06-18 by wisemanny