espressif/esp-zigbee-sdk #189: It's stupid, but can someone explain me how to add the zigbee sdk in the esp-idf ( a getting started like)

Hatim27, 2023-12-24:
### Question

Hello Everyone,

Could someone kindly spare a few moments to guide me through the process of integrating the Zigbee SDK into esp idf, after the the installation of the esp-idf? Your assistance would be greatly appreciated.

I'm stucked and i'm getting a lot of random errors.

Thank you in advance.

stuart, 2023-12-25:
I found the instructions here helpful. 
https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html

Make sure you have the dependencies in the file idf_component.yml, see the examples. Then they get pulled and added in automagically to your project as managed_components.

Hatim27, 2023-12-25:
Many thanks stuart, i succeed to flash my esp32-h2.
Now i need to open a new question regarding the mfg tool.

Thanks again.

Hatim27, 2023-12-25:
Question answered

closed 2023-12-25 by Hatim27