espressif/esp-zigbee-sdk #190: How to install and use the mfg tool for manufacturing properties

Hatim27, 2023-12-25:
### Question

Hello Everyone, 
I need to set the install code and manufacturing information, for that I need the mfg tool delivered within the SDK.
When I try to install the requirements I'm getting this error : 

PS D:\EspIDF\esp-idf\tools\mfg_tool> D:\Esp-Tools\python_env\idf5.1_py3.11_env\Scripts\pip install -r requirements.txt
ERROR: Could not find a version that satisfies the requirement ctypes==1.1.0 (from versions: none)
ERROR: No matching distribution found for ctypes==1.1.0

I'm using Visual Studio Code on Windows.

Many thanks in advanced.

xieqinan, 2023-12-28:
Hello @Hatim27 ,

> PS D:\EspIDF\esp-idf\tools\mfg_tool> D:\Esp-Tools\python_env\idf5.1_py3.11_env\Scripts\pip install -r requirements.txt

I am confused by the information, and I cannot locate the directory `~/esp-idf\tools\mfg_tool` in the esp-idf repository. According to the [README](https://github.com/espressif/esp-zigbee-sdk/blob/main/tools/mfg_tool/README.md?plain=1#L20), we expect to install `requirements.txt` in `~/esp-zigbee-sdk/tools/mfg_tool`. Could you please double-check this folder?

Hatim27, 2024-01-02:
Hi @xieqinan 
I tried as suggested by Readme and i'm getting the same issue : 

 C:\Espressif\frameworks\esp-idf-v5.1\esp-zigbee-sdk-main\tools\mfg_tool>python3 -m pip install -r requirements.txt
ERROR: Could not find a version that satisfies the requirement ctypes==1.1.0 (from versions: none)
ERROR: No matching distribution found for ctypes==1.1.0

Any help please,

kelin6, 2024-01-09:
@Hatim27 You can temporarily ignore this error and directly run `python esp_zb_mfg_tool.py --csv ./multi_devices_config.csv` to test. It should only be missing the `future`module, then simply use `pip install future`. I will update the requirement file accordingly.

closed 2024-01-12 by espressif-bot