espressif/esp-zigbee-sdk #192: How to create custom cluster ID

tengulak, 2023-12-26:
### Is your feature request related to a problem?

i try to create my own custom cluster id, but there is only one function "esp_zb_zcl_attr_list_create" which supports only defined clusters. Is there any function ?

xieqinan, 2023-12-28:
Hello @tengulak ，

The custom cluster ID in `esp-zigbee-sdk` is required to be greater than 0xFC00. You can use the function `esp_zb_zcl_attr_list_create(custom_cluster_id)` to create a custom attribute list.

chshu, 2024-02-01:
@tengulak Please reopen if any follow up questions.

closed 2024-02-01 by chshu