espressif/esp-zigbee-sdk #193: Invalid Value Checking for custom clusters

Kris151, 2023-12-27:
### Question

I am writing a zigbee server with some custom clusters, and I don't know how to check if the value in command is valid or not. I can see there are some enum in device library to check the value, but I don't know how to implement similar function.

xieqinan, 2023-12-28:
@Kris151 ,

Could you provide some examples of the values for the custom cluster that you expect to check? As far as I am concerned, the value of the custom cluster is defined by the users and can be verified by the users.

Kris151, 2024-01-02:
It is kinda like the system mode attribute in thermostat cluster. I have the enum for the attribute, e.g 0-5, and if the client asks to write attribute with value 0x06, it should return 0x87, invalid value. However, as far as I implement the custom cluster following the example, I don't know how can I check the value and return 0x87 to the client side if the value is out of range.

As I know is the value is set in the attribute when the message is received. So I check the value in attribute handler and set the attribute back to its original value if it is not valid. But the client server will still receive 0x00, which is success. How can I check the value before the function set the value in attribute and return 0x87 if it is out of range?

xieqinan, 2024-01-02:
Hello, @Kris151 ,
 
I believe that the developer will receive a 0x86 (unsupported attribute) response when attempting to write an attribute that is not set. The stack will handle the above process accordingly.

Kris151, 2024-01-02:
Hi @xieqinan ,
I think I did not explain it clearly, what I meant is an attribute type is enum, and the enum contains value 0-5 (similar to thermostat system mode attribute). And when client is trying to write the attribute with value 0x06, which should be out of range (as the enum only contain value 0-5), so the client should receive message 0x87 and 0x06 should not be set in that attribute.

When I try with thermostat cluster, and I write the system mode attribute(attribute type: enum with value 0-9) with value 0x10, the client side receive 0x87. I want the same thing to do with my custom attribute (type: enum). 

Although I created the enum for my custom attribute, I have no ideas how to use it to check if the new value is valid or not.

xieqinan, 2024-02-06:
@Kris151 ,

Understood, I misunderstood your previous message. Currently, the custom cluster does not support checking the written value. However, support for the `writing_value_check` callback will be added soon.

closed 2024-02-23 by espressif-bot

xieqinan, 2024-02-23:
Hello,

The custom cluster `writing_attr_check` and `check_value` callback had been added in [esp-zigbee-sdk v1.1.2](esp_zb_zcl_custom_cluster_handlers_update).