espressif/esp-zigbee-sdk #194: esp_zb_zcl_set_attribute_val not working for battery attribute

mundevx, 2023-12-29:
### Question

Hi,

I have issue regarding updating battery status over cluster **ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG** and attributes ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID ,  ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_PERCENTAGE_REMAINING_ID.

I register this cluster successfully. But esp_zb_zcl_set_attribute_val for changing battery status is not working. It responds with error code 0x88.

xieqinan, 2024-01-02:
@mundevx ，

The interpretation of error code 0x88 can be found in [zcl_common.h](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L202). This error suggests that the attribute is marked as "only_read" and cannot be modified. To address this, you can attempt to set the `false` to `esp_zb_zcl_set_attribute_val()` to perform no any additional checks to set the attribute.

mundevx, 2024-01-02:
> @mundevx ，
>
> The interpretation of error code 0x88 can be found in [...]

It solves the issue. Thank you!

closed 2024-01-08 by chshu