espressif/esp-zigbee-sdk #195: How to set battery status attribute

mundevx, 2023-12-29:
### Question

Hi,

I want to set battery status attribute. how i can do it.
i used functions esp_zigbee_publish_battery_power and esp_zigbee_write_attribute_request but these are only giving me compiler errors as implicit declaration. what function have to use to set battery status attribute over power cluster.

xieqinan, 2024-01-02:
@mundevx ，

The `esp_zigbee_publish_battery_power` and `esp_zigbee_write_attribute_request` are not APIs within the esp-zigbee-sdk, and their usage may result in compiler errors. Instead, the esp-zigbee-sdk offers the `power_config cluster` for managing battery status. For detailed information, you can consult the [power_config](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_power_config.h) documentation. Additionally, the SDK provides the `esp_zb_power_config_cluster_create()` function to create a standard power_config cluster, and `esp_zb_power_config_cluster_add_attr()` to add optional attributes to the power_config cluster.

mundevx, 2024-01-02:
> @mundevx ，
>
> The `esp_zigbee_publish_battery_power` and `esp_zigbee_write_attribute_request` are not APIs within the esp-zigbee-sdk, and their usage may result in compiler errors. Instead, the [...]

Ok, i will try your suggestion and will revert back to you.

chshu, 2024-01-08:
Closing, the issue has been addressed in https://github.com/espressif/esp-zigbee-sdk/issues/194

closed 2024-01-08 by chshu