espressif/esp-zigbee-sdk #196: How to create a esp32h2 zigbee scene switch configuration

mundevx, 2024-01-02:
### Question

Hi,

I want to create a scene switch, with 3 scenes in a single switch but my configuration is not showing scene button on gateway app.
My code for creating scene switch endpoint is given below:

esp_zb_cluster_list_t * esp_zb_cluster_list = get_server_cluster();
esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ONOFF_SWITCH_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_SCENE_SELECTOR_DEVICE_ID);
esp_zb_device_register(esp_zb_ep_list);

And configuration server cluster is given below:

esp_zb_cluster_list_t * get_server_cluster(){
	uint8_t test_attr;
    test_attr = 0;

    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);

    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);

    /* group cluster create with fully customized */
    uint16_t group_id = ESP_ZB_ZCL_GROUPS_NAME_SUPPORT_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &group_id);
    /* scenes cluster create with standard cluster + customized */
    uint8_t scene_count = SCENE_COUNT;
    uint8_t scene_id = 1;
    uint8_t current_scene = CURRENT_SCENE;
    uint8_t name_support = NAME_SUPPORT;
    uint8_t valid_id = SCENE_VALID;
    uint8_t mode_scene = 1;

    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);

    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);

    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    return esp_zb_cluster_list;
}

You can review my code, and suggest me where i am wrong. Thank you! I want to show the single scene button on gateway app.

chshu, 2024-03-08:
@mundevx From your multiple issues regarding scene usage, it seems you have made good progress on this topic.

Is the scene table size (https://github.com/espressif/esp-zigbee-sdk/issues/234) the last issue for you on the scene topic? Could you close other issues that are already addressed?

xieqinan, 2024-03-08:
@mundevx ,

> /* scenes cluster create with standard cluster + customized */
> uint8_t scene_count = SCENE_COUNT;
> uint8_t scene_id = 1;
> uint8_t current_scene = CURRENT_SCENE;
> uint8_t name_support = NAME_SUPPORT; [...]

suggestion:
```
uint8_t scene_count = SCENE_COUNT;
uint8_t scene_id = 1;
uint8_t current_scene = CURRENT_SCENE;
uint8_t name_support = NAME_SUPPORT;
uint8_t valid_id = SCENE_VALID;
uint8_t mode_scene = 1;

esp_zb_scenes_cluster_cfg_t scene_cfg = 
{
    .scenes_count = SCENE_COUNT,
    .current_scene = CURRENT_SCENE,
    .scene_valid = SCENE_VALID,
    .name_support = NAME_SUPPORT,
    .current_group = 0,
};
esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(&scene_cfg);
esp_zb_scenes_cluster_add_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_CURRENT_SCENE_ID, &scene_id);

```

The issue below appears to be related to this one. Could you please integrate them into one? It would be helpful for us to address the same question together.

https://github.com/espressif/esp-zigbee-sdk/issues/222
https://github.com/espressif/esp-zigbee-sdk/issues/248

mundevx, 2024-03-09:
> @mundevx ,
>
>
> suggestion:
>
> ```
> uint8_t scene_count = SCENE_COUNT;
> uint8_t scene_id = 1;
> uint8_t current_scene = CURRENT_SCENE;
> uint8_t name_support = NAME_SUPPORT;
> uint8_t valid_id = SCENE_VALID; [...]
Thanks you @xieqinan ,

Let me check and run this code, then i will get back to you.

xieqinan, 2024-04-15:
@mundevx，

Do you have any updates regarding to the issue?

closed 2024-07-26 by chshu