espressif/esp-zigbee-sdk #197: Listening for Attribute Reporting in Zigbee Gateway Example

ayushsharma82, 2024-01-02:
How can we listen for attribute reporting in zigbee gateway example under this repo? 

I have a use-case where device will periodically report attribute  to coordinator. The coordinator (zigbee gateway example) should be able to listen to these attribute reports.

xieqinan, 2024-01-08:
@ayushsharma82 ,

You can refer to the [customized device example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L91) to implement your requirement.

The steps are as follows:

First, it is necessary to implement the client side of the cluster that will report to the gateway in the gateway example.

Second, it is necessary to check whether the `use-case` can send the attribute report to the gateway or not. You should verify if the destination address, endpoint ID, and cluster ID can be matched correctly by the gateway in the report information.

Third, you can configure the attribute report using the `esp_zb_zcl_config_report_cmd_req()` according to your requirement.

xieqinan, 2024-02-06:
@ayushsharma82 ,

Please refer to https://github.com/espressif/esp-zigbee-sdk/issues/219#issuecomment-1926585961

chshu, 2024-03-08:
@ayushsharma82 please reopen if you have any further questions.

closed 2024-03-08 by chshu