espressif/esp-zigbee-sdk #198: Function for reporting scene status to coordinator

mundevx, 2024-01-04:
### Question

I need a function that will report scene attribute or send scene status to coordinator or gateway. how to use ESP_ZB_ZCL_CMD_SCENES_ADD_SCENE_RESPONSE command.

mundevx, 2024-01-05:
> Hello @mundevx ,
> 
> The attributes of the scene cluster do not possess the ability for reporting, as indicated in Zigbee ZCL specification 3.7.2.2.1 section.
> 
> The `ESP_ZB_ZCL_CMD_SCENES_ADD_SCENE_RESPONSE` is a command used for responding to the `ESP_ZB_ZCL_CMD_SCENES_ADD_SCENE` command and cannot be called directly.
> 
> You may attempt to use the [get_scene_membership](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L1982) function to obtain membership on the coordinator side.

Ok, i will check and will respond to you.

mundevx, 2024-01-13:
Hi xieqinan,

How i can create Scene switch using espressif code. when i click a switch, it will call the scene on coordinator side.

xieqinan, 2024-02-06:
@mundevx ,

The `esp_zb_on_off_switch_ep_create()` function can be used to create a scene switch. The ESP-Zigbee-SDK provides the following functions to support the scene cluster, which you can refer to in order to implement your project.

```
esp_zb_zcl_scenes_add_scene_cmd_req();
esp_zb_zcl_scenes_remove_scene_cmd_req();
esp_zb_zcl_scenes_remove_all_scenes_cmd_req();
esp_zb_zcl_scenes_view_scene_cmd_req();
esp_zb_zcl_scenes_store_scene_cmd_req();
esp_zb_zcl_scenes_recall_scene_cmd_req();
esp_zb_zcl_scenes_get_scene_membership_cmd_req();
```

chshu, 2024-03-08:
@mundevx closing as the question is answered.

closed 2024-03-08 by chshu