espressif/esp-zigbee-sdk #199: Provide many more "common" examples as samples of standard Zigbee devices in this esp-zigbee-sdk repository?

Hedda, 2024-01-04:
### Is your feature request related to a problem?

Please consider provide many "common" examples as samples apps of standard Zigbee devices this esp-zigbee-sdk repository.

To compare Silicon Labs provides a collection of applications and device examples for different protocols including for Zigbee:

https://github.com/SiliconLabs/zigbee_applications

https://github.com/SiliconLabs/application_examples

Texas Instruments provides these examples as sample applications (including thermostat, doorlock, and more):

https://github.com/TexasInstruments/simplelink-zstack-examples/tree/main/examples/rtos/LP_CC2652PSIP/zstack

https://github.com/TexasInstruments/simplelink-zstack-examples/tree/main/examples/rtos

For the esp-matter (Espressif's SDK for Matter) you are at least providing many more examples as app samples:

https://github.com/espressif/esp-matter/tree/main/examples

Most importantly it provides an "all_device_types_app" though personally I think it is nicer with seperate examples broken out:

https://github.com/espressif/esp-matter/tree/main/examples/all_device_types_app/main

Currently there is only a "light_driver" as Zigbee End-Device example and a "switch_driver" as Zigbee Home Automation example:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/common

https://github.com/espressif/esp-idf/tree/master/examples/zigbee

I think that it would be a good idea to include more sample code with examples of other commonly Zigbee devices, such as ex:

* Zigbee Router (Zigbee 3.0 Router with option to set TX power)
* sleepy devices (ex. temperature and humidity sensor, preferably with battery monitoring and battery procentage sensor).
* Room Air Conditioner (ex. HVAC climate control like a thermostat)
* Fan (ex. an percentage, or fan preset mode low/medium/high)
* Chime (ex. Zigbee Doorbell or Siren, with duration and volume level input).
* Binary sensors (ex. Light Sensor, Leak Sensor, Smoke detector, PIR Human sensor, like maybe occupancy or motion detected on/off, door and window sensor open/close, battery low/medium/high).
* Door lock

### Describe the solution you'd like.

More example, preferably collected in a single place for all protocols with ports for each protocol.

xieqinan, 2024-01-08:
@Hedda ,

Thanks for your suggestions; they are very relevant, and we will take them into consideration. Before it,  if you have specific examples that you are unsure how to implement based on the current example, please feel free to create an issue, and we will provide a solution as soon as possible.

chshu, 2024-01-08:
Link https://github.com/espressif/esp-zigbee-sdk/issues/205#issuecomment-1880811875

chshu, 2024-07-26:
Closing this issue, please file separate issues for any specific examples you are looking for.

closed 2024-07-26 by chshu

mastersf, 2025-06-19:
Chu, could you please provide examples on binary sensors (eg. PIR) and sleepy devices as outlined by user earlier

mastersf, 2025-06-20:
could you please add more examples for binary sensor (specifically PIR / motion), light lux, battery?