espressif/esp-zigbee-sdk #200: Sleepy End device example - i2c fails after wakeup

darianbjohnson, 2024-01-04:
### Question

I have gotten the sleepy end device code to work on my ESP32H2 - however, I cannot use a led driver (connected via i2c) after the board wakes from sleep.  Is there a lock that I need to keep the i2c connection active?

kelin6, 2024-01-05:
@darianbjohnson  I'm deeply sorry to inform you that the recovery and backup of I2C sleep are not yet supported in the IDF. We are currently discussing the support for I2C sleep internally.

darianbjohnson, 2024-01-06:
Ok, thanks for the update

Suxsem, 2024-01-11:
@kelin6 is the RMT affected by the same problem when using light sleep?
i'm using the led_strip component but i'm not able to properly control the rgb led when wake up from light sleep, esp32-c6

kelin6, 2024-01-11:
@Suxsem Yes, light sleep for RMT is not yet supported on the esp32c6/esp32h2 chips.

> @kelin6 is the RMT affected by the same problem when using light sleep? i'm using the led_strip component but i'm not able to properly control the rgb led when wake up from light sleep, esp32-c6

darianbjohnson, 2024-01-12:
@kelin6 is there an estimated time/release for when this will be resolved?

darianbjohnson, 2024-03-28:
Note - for anyone with this problem I get around this issue by removing the Bus - i2c_del_master_bus(bus_handle) -  and the Device -i2c_master_bus_rm_device(device_handle) - prior to sleep. And I re-create them on wake-up.

kelin6, 2024-04-23:
@darianbjohnson @Suxsem I2C sleep retention support has been added in [Support i2c sleep retention on esp32c6/h2](https://github.com/espressif/esp-idf/commit/cf521b60ea28d5f0b840f2b56ccf13eefe64f273), while RMT support is still in progress.

darianbjohnson, 2024-06-26:
Actually, I was able to get example shown in [test_i2c_sleep_retention](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/test_apps/i2c_test_apps/main/test_i2c_sleep_retention.c) to work. But, I cannot get this to work when using zigbee sleep (which uses automatic sleep). Any ideas?

lxz946786639, 2024-06-29:
> Actually, I was able to get example shown in [...]

I also encountered the same problem

lxz946786639, 2024-06-30:
> Note - for anyone with this problem I get around this issue by removing the Bus - i2c_del_master_bus(bus_handle) - and the Device -i2c_master_bus_rm_device(device_handle) - prior to sleep. And I re-create them on wake-up.

According to this method, i2c work ok. I think this is currently the only solution for automatic sleep.

chshu, 2024-09-29:
@darianbjohnson @lxz946786639 Do you have any pending issues on the I2C retention after wakeup?

darianbjohnson, 2024-09-29:
I2C does not work me on wake-up, unless I remove the Bus and re_create the Bus on wakeup.

lxz946786639, 2024-10-16:
> I2C does not work me on wake-up, unless I remove the Bus and re_create the Bus on wakeup.

me too.

chshu, 2024-12-19:
@darianbjohnson @lxz946786639 The I2C retention is supported since: https://github.com/espressif/esp-idf/commit/27b2f7a10b662fdbe4991e12ba9c04870083a453, please double check with the latest code base, if you still encounter any issue regarding I2C retention, please file an issue at https://github.com/espressif/esp-idf/issues

closed 2024-12-19 by chshu