espressif/esp-zigbee-sdk #201: OTA Client increasing size issue

mundevx, 2024-01-05:
### Question

On increasing size of OTA_UPGRADE_MAX_DATA_SIZE,  ota functionality not works. 
Default value of OTA_UPGRADE_MAX_DATA_SIZE = 64, and if change it to greater tha 64 like 128 or 512, ota functionality not works.

xieqinan, 2024-01-08:
@mundevx ,

The maximum data size for OTA is currently limited to below 64 in the current version. We are working on providing an improved solution to enhance the OTA speed.

xieqinan, 2024-08-29:
@mundevx ,

The maximum size of an OTA upgrade is now supported up to 255 (limited by the ZCL specification) in esp-zigbee-sdk v1.5.0. Please use the [OTA example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota) for testing.

If the issue has been resolved, please consider closing it.

mundevx, 2024-08-30:
Thank you very much @xieqinan

closed 2024-09-29 by chshu