espressif/esp-zigbee-sdk #202: unable to write a "large" string

diazmanuel, 2024-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### esp-zigbee-lib version.

1.0.7

### esp-zboss-lib version.

1.0.7

### Espressif SoC revision.

ESP32-H6 and ESP32-H2

### What is the expected behavior?

I'm trying to write a string attribute with a length of like 150-200 characters. The behavior I expect is that it always arrives and the content is correct

### What is the actual behavior?

The current behavior is that sometimes the write callback on the destination device is not activated or when it is activated the content is truncated or damaged

### Steps to reproduce.

just send a string with a 150-200 characters

### More Information.

I think it may be due to an internal buffer in the sender or in the receiver, is there any limitation on the number of characters? reading the zigbee 3.0 documentation I did not find a limit on the maximum length

xieqinan, 2024-01-08:
@diazmanuel ,

The upcoming version will support the frame fragment feature, which should meet the above requirements. It is scheduled for release later this week.

diazmanuel, 2024-01-15:
Hello, @xieqinan,
I have seen the last update they made adding what you mentioned, but I don't understand how to use it to solve the problem I have of not being able to send a long string, could you tell me how to solve the problem?

xieqinan, 2024-01-23:
@diazmanuel ，

The simple implementation is as followed:

```
void *esp_zb_zcl_custom_array_copy_malloc(esp_zb_zcl_attr_type_t type, void *array, uint32_t elem_size, uint16_t elem_count)
{
    uint8_t *value = NULL;
    uint8_t offset = 0;
    switch (type) {
    case ESP_ZB_ZCL_ATTR_TYPE_LONG_OCTET_STRING:
    case ESP_ZB_ZCL_ATTR_TYPE_LONG_CHAR_STRING:
    case ESP_ZB_ZCL_ATTR_TYPE_ARRAY:
    case ESP_ZB_ZCL_ATTR_TYPE_16BIT_ARRAY:
    case ESP_ZB_ZCL_ATTR_TYPE_32BIT_ARRAY:
        value = malloc(sizeof(uint16_t) + elem_size * elem_count);
        memcpy(value + offset, &elem_count, sizeof(uint16_t));
        offset += sizeof(uint16_t);
        break;
    case ESP_ZB_ZCL_ATTR_TYPE_OCTET_STRING:
    case ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING:
        value = malloc(sizeof(uint8_t) + elem_size * elem_count);
        memcpy(value + offset, &elem_count, sizeof(uint8_t));
        offset += sizeof(uint8_t);
        break;
    default:
        ESP_LOGW(TAG, "The data type(0x%x) is not array or string, failed to create it", type);
        value = NULL;
        break;
    }
    memcpy(value + offset, array, elem_count * elem_size);
    return value;
};

void esp_zb_zcl_custom_array_free(uint8_t *array)
{
    if (array) {
        free(array);
    }
}

static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        esp_zb_zcl_custom_cluster_cmd_req_t custom_req;

        uint8_t message[1024] = {};

        for (int i = 0; i < sizeof(message) / sizeof(uint8_t); i++) message[i] = i & 0xFF;

        uint8_t *value =
            esp_zb_zcl_custom_array_copy_malloc(ESP_ZB_ZCL_ATTR_TYPE_ARRAY, message, sizeof(message[0]), sizeof(message) / sizeof(uint8_t));
        custom_req.zcl_basic_cmd.dst_endpoint = light_node.endpoint;
        custom_req.zcl_basic_cmd.dst_addr_u.addr_short = light_node.short_addr;
        custom_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        custom_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT; // ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        custom_req.cluster_id = CUSTOM_CLUSTER_ID;
        custom_req.custom_cmd_id = 0x01;
        custom_req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
        custom_req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
        custom_req.data.type = ESP_ZB_ZCL_ATTR_TYPE_ARRAY;
        custom_req.data.value = value;
        ESP_EARLY_LOGI(TAG, " Transfer(%d) * * * * * * * * * * * * * * * * * *", sizeof(message));
        esp_zb_zcl_custom_cluster_cmd_req(&custom_req);
        esp_zb_zcl_custom_array_free(value);
    }
}
```

diazmanuel, 2024-01-24:
hi @xieqinan ,Thanks for the reply,
could you share the code on how to read it? 
Also, it is not possible to use the read and write command request functions with a long string?

xieqinan, 2024-01-25:
@diazmanuel ，

The writing and reading are the same with the custom cluster request, only if the data type of attribute is supported.

diazmanuel, 2024-01-31:
but why can't I declare an attribute of type string in a custom cluster and write and read it with the esp_zb_zcl_write_attr_cmd_req() and esp_zb_zcl_read_attr_cmd_req() functions? The length of the string according to the Zigbee 3.0 standard is supposed to be at least 255 characters, but in practice when I try to write or read something longer than 40 characters it starts to have errors.

I leave you a screenshot of the zigbee documentation

[image]

Is it possible to solve this problem so that the complete string can be used as indicated in the documentation?

xieqinan, 2024-03-18:
@diazmanuel ,

Large data for custom clusters, both reading and writing, is now supported in esp-zigbee-sdk v1.2.1. You can test this functionality using the `esp_zb_zcl_read_attr_cmd_req()` and `esp_zb_zcl_write_attr_cmd_req()` APIs.

diazmanuel, 2024-03-18:
> @diazmanuel ,
> 
> Large data for custom clusters, both reading and writing, is now supported in esp-zigbee-sdk v1.2.1. You can test this functionality using the `esp_zb_zcl_read_attr_cmd_req()` and `esp_zb_zcl_write_attr_cmd_req()` APIs.

hello, thanks you, we will test this update after fixing the error with the custom clusters #285

diazmanuel, 2024-03-18:
> @diazmanuel ,
> 
> Large data for custom clusters, both reading and writing, is now supported in esp-zigbee-sdk v1.2.1. You can test this functionality using the `esp_zb_zcl_read_attr_cmd_req()` and `esp_zb_zcl_write_attr_cmd_req()` APIs.

Hello, we still haven't been able to get it to work, depending on the message, some arrive well and others don't.
We carry out the following test, sending different messages and printing the response.
In some cases the messages arrive fine (this is almost always the case with short message)
It also happens that the system restarts when long messages arrive, anyway I couldn't find the logic why it fails, I leave the tests below so you can analyze it better. both the endpoint and the gateway are running version 1.2.1 on idf 5.1.2

Message 1:
"This message has a length of 105: COMPUTER, MOUSE, DESK, CHAIR, FAN, LIGHT, NOTEBOOK, BOTTLE, PLUG, GLASS"
Arrives Message 1:
"E (288557) APP: len: 105 data: This message has a length of 105: COMPUTER, MOUSE, DESK, CHAIR, FAN, LIGHT, NOTEBOOK, BOTTLE, PLUG, GLASS "
the code crashed?: YES
Complete LOG Message 1:
```
I (607) cpu_start: Unicore app
I (607) cpu_start: Pro cpu up.
W (617) clk: esp_perip_clk_init() has not been implemented yet
I (623) cpu_start: Pro cpu start user code
I (624) cpu_start: cpu freq: 96000000 Hz
I (624) cpu_start: Application information:
I (626) cpu_start: Project name:     firmware_boiler_zigbee_G2
I (633) cpu_start: App version:      57fc143-dirty
I (638) cpu_start: Compile time:     Mar 18 2024 11:31:09
I (644) cpu_start: ELF file SHA256:  6477b87dfdecd40b...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is dd9e1137f1e42cb1b5688d48547ad1725c8138bb5ef0f9814f90f64ff1615239
I (650) cpu_start: ESP-IDF:          v5.1.3-dirty
I (656) cpu_start: Min chip rev:     v0.0
I (660) cpu_start: Max chip rev:     v0.99 
I (665) cpu_start: Chip rev:         v0.1
I (670) heap_init: Initializing. RAM available for dynamic allocation:
I (677) heap_init: At 4081FED0 len 0002D4B0 (181 KiB): D/IRAM
I (684) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (691) spi_flash: detected chip: generic
I (695) spi_flash: flash io: dio
I (703) sleep: Configure to isolate all GPIO pins in sleep state
I (706) sleep: Enable automatic switching of GPIO sleep configuration
I (713) coexist: coex firmware version: 77cd7f8
I (719) app_start: Starting scheduler on CPU0
I (723) main_task: Started on CPU0
I (727) main_task: Calling app_main()
E (741) AZURE: dato 
E (741) AZURE: dato 
E (741) task_wdt: esp_task_wdt_init(593): TWDT already initialized
I (744) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (753) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (762) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (772) taskZigbee: zigbee Handler Task Initialized
I (777) taskZigbee: Zigbee Handler Task Created
I (791) phy: phy_version: 230,2, 9aae6ea, Jan 15 2024, 11:17:12
I (792) phy: libbtbb version: 944f18e, Jan 15 2024, 11:17:25
I (822) taskZigbee:  Zigbee Task Created
W (825) taskZigbee: ZDO signal: 23, status: -1
I (826) taskZigbee: Zigbee signal: SKIP_STARTUP
I (1771) taskZigbee: Start network steering
W (1800) taskZigbee: ZDO signal: 54, status: 0
I (1804) taskZigbee: Zigbee signal: DEVICE_REBOOT
I (1804) taskZigbee: Zigbee stack initialized
I (1892) taskZigbee: ieee_cb address is: 60:55:f9:00:00:f6:f0:78
I (1893) taskZigbee: local address is: 74:4d:bd:ff:fe:60:9b:dd
I (1895) taskZigbee: local short address is: 0xf982
I (1909) taskZigbee: bind_cb status:0
W (1923) taskZigbee: ZDO signal: 50, status: 0
I (2129) BLE_INIT: Using main XTAL as clock source
I (2135) BLE_INIT: ble controller commit:[217f1bf]
I (2138) NimBLE_SPP_BLE_PRPH: BLE Host Task Started
I (2320) taskZigbee: Zigbee signal: DEVICE_REBOOT
I (2321) taskZigbee: Zigbee stack initialized
W (12977) taskZigbee: ZDO signal: 50, status: 0
[... 406 lines cut ...]
40812900: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40812920: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40812940: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40812960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40812980: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408129a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x408129c4
408129c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x408129e4
408129e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 6477b87dfdecd40b
Warning: checksum mismatch between flashed and built applications. Checksum of built application is dd9e1137f1e42cb1b5688d48547ad1725c8138bb5ef0f9814f90f64ff1615239

Rebooting...
```

Results:
Messages 1: Len 105, arrived correctly and crash
Messages 2: Len 135, arrived correctly and crash
Messages 3: Len 41, arrived correctly and do not crash
Messages 4: Len 53, arrived truncated and do not crash
Messages 5: Len 230, did not arrive and crash
 
Each message was sent multiple times giving the same results, meaning that the messages that are truncated do so systematically and in the same place.

Could you reopen this issue to follow up? thank you

xieqinan, 2024-03-19:
Hi, @diazmanuel ,

I have created a simple example to test this issue, which is attached here. Could you please also verify it?
[custom_cluster_test.zip](https://github.com/espressif/esp-zigbee-sdk/files/14644915/custom_cluster_test.zip)

diazmanuel, 2024-03-20:
hello, @xieqinan 
I was testing the example that you sent me and yes, it works correctly without any type of error, so to continue with the testing, I made a modification to bring it closer to what I have. On the endpoint side I simply added another way to print the data that came to me in string format:
```
            for (int i = 0; i < message->attribute.data.size; i++) {
                printf("0x%2x ", *(uint8_t *)(message->attribute.data.value + i));
            }
            printf("\n");
            printf("%s",(char *)message->attribute.data.value);
            printf("\n");
            ESP_LOGE(TAG,"%s",(char *)message->attribute.data.value);
```
On the other hand, in the coordinator,  I replaced the way in which the character string is generated as follows:
```
        case 1: {
            // uint8_t value[CUSTOM_STRING_MAX_SIZE];
            // value[0] = CUSTOM_STRING_MAX_SIZE - 1;
            // for (int i = 1; i < CUSTOM_STRING_MAX_SIZE; i++) {
            //     value[i] = i;
            // }
            char *msg="This message has a length of 230: COMPUTER, LAMP, FOCUS, HOTMIGA, KEYBOARD, GUITAR, BOTTLE, GLASS, CELL PHONE, DESK, WATER, BATHROOM, FIRE, TRUCK, ONE, TWO, THREE, FOUR, FIVE, SIX, SEVEN, EIGHT, NINE, TEN, ELEVEN, TWELVE, THIRTEEN";
            char *value;
            value = (void *)malloc(sizeof(uint8_t) * (strlen(msg) +  1));
            memset( value,0,sizeof(uint8_t) * (strlen(msg) +  1));
            ZBM_Set_Zigbee_Attr_String(value,strlen(msg),msg);
            esp_zb_zcl_attribute_t attrs = {2, {ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, sizeof(value), value}};
            esp_zb_zcl_write_attr_cmd_t cmd_req = {
                .zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT,
                .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
                .clusterID = CUSTOM_CLUSTER_ID,
                .attr_number = 1,
                .attr_field = &attrs,
            };
            esp_zb_zcl_write_attr_cmd_req(&cmd_req);
            ESP_LOGI(TAG, "Send 'write large data(%d)' command", sizeof(value));
            flag = 2;
            free(value);
            break;
```
```
bool ZBM_Set_Zigbee_Attr_String(void *attr_data, uint16_t attr_len, char *s)
{
  bool res = false;

  if(NULL == attr_data || 0 == attr_len || NULL == s)
  {
    return res;
  }

  ((uint8_t*)attr_data)[0] = (uint8_t)attr_len;

  if(0 < attr_len) {
    memcpy(&((uint8_t*)attr_data)[1], s, attr_len);
  }

  res = true;
  return res;  
}
```
After sending the message from the coordinator, the endpoint receives the following message

```
I (18027) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(10), cluster(0xfc00), attribute(0x2), data size(231)
0xe6 0x54 0x68 0x69 0x73 0x20 0x6d 0x65 0x73 0x73 0x61 0x67 0x65 0x20 0x68 0x61 0x73 0x20 0x61 0x20 0x6c 0x65 0x6e 0x67 0x74 0x68 0x20 0x6f 0x66 0x20 0x32 0x33 0x30 0x3a 0x20 0x43 0x4f 0x4d 0x50 0x55 0x54 0x45 0x52 0x2c 0x20 0x4c 0x41 0x4d 0x50 0x2c 0x20 0x46 0x4f 0x43 0x55 0x53 0x2c 0x20 0x48 0x4f 0x54 0x4d 0x49 0x47 0x41 0x2c 0x20 0x4b 0x45 0x59 0x42 0x4f 0x41 0x52 0x44 0x2c 0x20 0x47 0x55 0x49 0x54 0x41 0x52 0x2c 0x20 0x42 0x4f 0x54 0x54 0x4c 0x45 0x2c 0x20 0x47 0x4c 0x41 0x53 0x53 0x2c 0x20 0x43 0x45 0x4c 0x4c 0x20 0x50 0x48 0x4f 0x4e 0x45 0x2c 0x20 0x44 0x45 0x53 0x4b 0x2c 0x20 0x57 0x41 0x54 0x45 0x52 0x2c 0x20 0x42 0x41 0x54 0x48 0x52 0x4f 0x4f 0x4d 0x2c 0x20 0x46 0x49 0x52 0x45 0x2c 0x20 0x54 0x52 0x55 0x43 0x4b 0x2c 0x20 0x4f 0x4e 0x45 0x2c 0x20 0x54 0x57 0x4f 0x2c 0x20 0x54 0x48 0x52 0x45 0x45 0x2c 0x20 0x46 0x4f 0x55 0x52 0x2c 0x20 0x46 0x49 0x56 0x45 0x2c 0x20 0x53 0x49 0x58 0x2c 0x20 0x53 0x45 0x56 0x45 0x4e 0x2c 0x20 0x45 0x49 0x47 0x48 0x54 0x2c 0x20 0x4e 0x49 0x4e 0x45 0x2c 0x20 0x54 0x45 0x4e 0x2c 0x20 0x45 0x4c 0x45 0x56 0x45 0x4e 0x2c 0x20 0x54 0x57 0x45 0x4c 0x56 0x 0 0x 0 0x 0 0x 0 0x a 0x42 0x52 0x54 0x45 0x45 0x4e 
�This message has a length of 230: COMPUTER, LAMP, FOCUS, HOTMIGA, KEYBOARD, GUITAR, BOTTLE, GLASS, CELL PHONE, DESK, WATER, BATHROOM, FIRE, TRUCK, ONE, TWO, THREE, FOUR, FIVE, SIX, SEVEN, EIGHT, NINE, TEN, ELEVEN, TWELV
E (18147) ESP_ZB_ON_OFF_LIGHT: �This message has a length of 230: COMPUTER, LAMP, FOCUS, HOTMIGA, KEYBOARD, GUITAR, BOTTLE, GLASS, CELL PHONE, DESK, WATER, BATHROOM, FIRE, TRUCK, ONE, TWO, THREE, FOUR, FIVE, SIX, SEVEN, EIGHT, NINE, TEN, ELEVEN, TWELV
```

It was possible to observe that the string of characters is cut, and analyzing the print by bytes it is seen that some bytes at the end are found as split (0x45 0x4c 0x56 **0x 0 0x 0 0x 0 0x 0 0x a** 0x42 0x52 0x54 0x45 0x45 0x4e) , something strange that I cannot explain, and I do not understand how this case can occur but I interpret it as that the message was corrupted, which results in the last value read by the print or the log function being 0x, which it interprets as \0, meaning the end of the chain. This explains why I noticed that the chain was breaking, it could also explain why it sometimes crashed in my code but this requires further analysis.

Another thing I noticed in your example is that if I modify the length of the message you originally sent, changing the CUSTOM_STRING_MAX_SIZE parameter to 150, the program crashes, I understand that because it is of type string it should be able to send up to 255 as specified by the standard, I don't know if there is a limitation on this topic

Anyway here you can see that the data is being corrupted.
I also attach the modified example. In addition to the changes I mentioned above, I also had to modify the dependencies.lock file since the idf versions did not match, and the coordinator is an ESP32-C6.

[custom_test_mod.zip](https://github.com/espressif/esp-zigbee-sdk/files/14668518/custom_test_mod.zip)

diazmanuel, 2024-04-04:
hello, @xieqinan 
do you have any update about this issue? 
Could you open it again so that it can continue following?

xieqinan, 2024-04-07:
@diazmanuel ,

Could you update the esp-idf version to v5.1.3 for testing again? I also will test the examples provided by you and give you feedback ASAP.

xieqinan, 2024-04-07:
@diazmanuel ,

> It was possible to observe that the string of characters is cut, and analyzing the print by bytes it is seen that some bytes at the end are found as split (0x45 0x4c 0x56 0x 0 0x 0 0x 0 0x 0 0x a [...]

The issue has been confirmed as a stack issue, and we will fix it in the next release. The root cause of this issue is that the I/O buffer does not have sufficient capacity for the input data, resulting in data being overwritten at the end.

> Another thing I noticed in your example is that if I modify the length of the message you originally sent, changing the CUSTOM_STRING_MAX_SIZE parameter to 150, the program crashes, I understand [...]

Have you made any modifications to the code I provided? I tested it with a length equal to 150 case, and it worked successfully.

diazmanuel, 2024-04-08:
hello @xieqinan, 

> The issue has been confirmed as a stack issue, and we will fix it in the next release. The root cause of this issue is that the I/O buffer does not have sufficient capacity for the input data, resulting in data being overwritten at the end.

I'm glad to hear you were able to identify the problem.

> Have you made any modifications to the code I provided? I tested it with a length equal to 150 case, and it worked successfully.
No, the only changes I made were those I mentioned above, and I also used an H2 chip in one case and a C6 in the other. I think the only difference may be the versions of the SDK. In a previous comment I attached the modified project, there you can check the dependencies file. Anyway, I'll try it again and let you know if the error continues to appear.

diazmanuel, 2024-04-12:
Thanks, we tried it and now it works correctly

closed 2024-04-12 by diazmanuel

kgkask, 2024-05-24:
I am wondering if it can be used with Arduino IDE, I am new to both Zigbee and ESP32H2 in general and I would like to used it for some projects, so is it possible .

kgkask, 2024-06-19:
@xieqinan 
I used this example and it works fine,  however I tried to adjust it to suits my case, where I want to send the string from the ED to the Coordinator when I do that and adjust everyThing accordingly the zigbee coordinator do not receive any thing. a help I will be appreciated. 

I actually in the next step I want to carry a message with the write_resp() to change the ED mode sleep/awake. 

here is my codes: 

End Device: 
  `#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

/* Switch configuration */
#define GPIO_INPUT_IO_TOGGLE_SWITCH GPIO_NUM_9
#define PAIR_SIZE(TYPE_STR_PAIR)    (sizeof(TYPE_STR_PAIR) / sizeof(TYPE_STR_PAIR[0]))
#define CUSTOM_CLUSTER_ID 0xFC00
#define CUSTOM_STRING_MAX_SIZE 127
#define LED_PIN RGB_BUILTIN

typedef enum {
  SWITCH_ON_CONTROL,
  SWITCH_OFF_CONTROL,
  SWITCH_ONOFF_TOGGLE_CONTROL,
  SWITCH_LEVEL_UP_CONTROL,
  SWITCH_LEVEL_DOWN_CONTROL,
  SWITCH_LEVEL_CYCLE_CONTROL,
  SWITCH_COLOR_CONTROL,
} switch_func_t;

typedef enum {
  SWITCH_IDLE,
  SWITCH_PRESS_ARMED,
  SWITCH_PRESS_DETECTED,
  SWITCH_PRESSED,
  SWITCH_RELEASE_DETECTED,
} switch_state_t;

typedef struct {
  uint8_t pin;
  switch_func_t func;
} switch_func_pair_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

/* Zigbee configuration */
#define INSTALLCODE_POLICY_ENABLE       false    /* enable the install code policy for security */
#define ESP_ZB_PRIMARY_CHANNEL_MASK ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK /* Zigbee primary channel mask use in the example */
//#define ESP_ZB_PRIMARY_CHANNEL_MASK     (1l << 18)  /* Zigbee primary channel mask use in the example */
#define HA_ESP_LIGHT_ENDPOINT           10    /* esp light bulb device endpoint, used to process light controlling commands */
#define ED_AGING_TIMEOUT                ESP_ZB_ED_AGING_TIMEOUT_64MIN
#define ED_KEEP_ALIVE                   8000    /* 3000 millisecond */

/* Default Coordinator config */
//RADIO_MODE_NATIVE, main differnce between IDF and Arduino IDE starting of the command.
#define ESP_ZB_ZED_CONFIG()                                                                 \
  {                                                                                         \
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED, .install_code_policy = INSTALLCODE_POLICY_ENABLE, \
    .nwk_cfg = {                                                                            \
      .zed_cfg =                                                                            \
        {                                                                                   \
          .ed_timeout = ED_AGING_TIMEOUT,                                                   \
          .keep_alive = ED_KEEP_ALIVE,                                                      \
        },                                                                                  \
[... 728 lines cut ...]
  ESP_ERROR_CHECK(esp_zb_start(false));
  esp_zb_main_loop_iteration();
}

/********************* Arduino functions **************************/
// DONE converting
void setup() {
  //Serial.begin(115200)
  // Init Zigbee
  esp_zb_platform_config_t config = {
    .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
    .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
  };
  //ESP_ERROR_CHECK(nvs_flash_init());
  ESP_ERROR_CHECK(esp_zb_platform_config(&config));
  // Init RMT and leave light OFF
  neopixelWrite(LED_PIN, 0, 0, 0);
  // Start Zigbee task
  xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

void loop() {

}
`