espressif/esp-zigbee-sdk #203: ESP32C6 crash/reboot when reporting attribute (esp_zb_zcl_report_attr_cmd_req)

kylemwagner83, 2024-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### esp-zigbee-lib version.

0.9.0

### esp-zboss-lib version.

0.7.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Update a cluster attribute

### What is the actual behavior?

Crashes and reboots the SOC

### Steps to reproduce.

Using the HA_on_off_light as a starting point, I'm able to connect to my Zigbee network, but as soon as I add the esp_zb_zcl_report_attr_cmd_req function, the ESP crashes and reboots.

Zigbee Main Code
```
static void esp_zb_task(void *params)
{
    // Define custom mfg/model
    char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};
    char modelid[] = {14, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'S', 'e', 'n', 's', 'o', 'r'};

    // Define attributes
    int *tempParam = 2300;
    void* tempvoid = &tempParam;

    // Initialize Zigbee stack
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    // Create genBasic cluster/attribute list
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    
    
    // Add & set custom mfg/model attributes
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);

    // Create temperature cluster/attribute list
    esp_zb_attribute_list_t *esp_zb_temperature_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, tempvoid);

    // Create cluster list
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();

    // Add clusters/attribute lists to cluster list
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list, esp_zb_temperature_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    // Create endpoint list
[... 186 lines cut ...]
408188c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408188e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40818900: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40818920: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40818940: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: e33bde163873bc10

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd4c
load:0x4086e610,len:0x2dd4
load:0x40875728,len:0x17d0
entry 0x4086c410

```

If I comment out the the zb_update_temperature command, it succssfully connects to the network, and the temperature attribute is readable from the coordinator.

xieqinan, 2024-01-08:
Hello @kylemwagner83 ,

Could you please verify whether the `ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_ID` and `ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_ID` attributes have been added to the `temperature_meas_cluster`? Furthermore, ensure that the `temperature_value` is within the range of `ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_ID` to `ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_ID`.

kylemwagner83, 2024-01-08:
Properly adding the min and max values fixed the issue, thank you!!

closed 2024-01-08 by kylemwagner83