espressif/esp-zigbee-sdk #204: Support for ZCL flow measurement cluster

Bobstin, 2024-01-07:
### Is your feature request related to a problem?

ZCL defines a cluster for flow measurement (0x0404). This cluster does not appear to be currently supported.

### Describe the solution you'd like.

Implement the cluster and associated attributes for flow measurement.

### Describe alternatives you've considered.

Currently I am using a generic analog input cluster, which largely works but is not what the standard says I should use.

xieqinan, 2024-01-08:
@Bobstin ,

Noted. We will add cluster support in the future; however, it may require some time, and we appreciate your patience.

jacksonrwoodard, 2024-04-02:
Bobstin can you send your code in here because I am trying to figure out how to make the analog input cluster work, but I cannot find any examples on how to get this to work.

Bobstin, 2024-04-06:
This is more of a demo that I have been building than anything else, but the analog input cluster does work with it: https://github.com/Bobstin/AutomatedBrewery/blob/master/ESPZigbeeDemo/zigbee_cluster_demo/main/esp_zb_light.c

jacksonrwoodard, 2024-04-07:
> This is more of a demo that I have been building than anything else, but the analog input cluster does work with it: https://github.com/Bobstin/AutomatedBrewery/blob/master/ESPZigbeeDemo/zigbee_cluster_demo/main/esp_zb_light.c

What GPIO pin are you using to get the analog input results?

jacksonrwoodard, 2024-04-07:
> > This is more of a demo that I have been building than anything else, but the analog input cluster does work with it: https://github.com/Bobstin/AutomatedBrewery/blob/master/ESPZigbeeDemo/zigbee_cluster_demo/main/esp_zb_light.c
> 
> What GPIO pin are you using to get the analog input results?

Or how can I set which GPIO pin the data is being read from?

xieqinan, 2024-04-15:
@Bobstin ,

The [flow measurement cluster](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_flow_meas.h) has been supported in the esp-zigbee-sdk v1.2.3. Could you please test it on your side? If it works as expected, please consider closing this issue.

closed 2024-07-26 by chshu