espressif/esp-zigbee-sdk #205: Add support for many more ZCL Clusters out-of-the-box

Hedda, 2024-01-08:
### Is your feature request related to a problem?

Zigbee Cluster defines many more clusters that does not appear to be currently supported in ESP-Zigbee-SDK, so wondering if possible to make a generic bulk feature request that ask you to add all or at least many more ZCL Clusters out-of-the-box?

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html#zcl-clusters

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/index.html

That cluster list is not up to date with the new Zigbee Cluster Library Specification R8 (Revision 8), sometimes also refered to as as ZCLr8 or ZCLv8.

https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf

Most if not all of the holes of missing ZCL clusters are however already supported by popular Zigbee gateway implementations, such as the relativly popular Home Assistant's built-in ZHA integration which is based on the zigpy library and is more than likley going to be used as a testbed by most developers for testing Zigbeee compatibility:

https://github.com/zigpy/zigpy/tree/dev/zigpy/zcl/clusters

Example of missing ZCL clusters :

0x0409 pH Measurement
0x0408 Soil Moisture
0x0407 Leaf Wetness
0x0404 Flow Measurement
0x0401 Illuminance Level Sensing
0x0301 ballast configuration
0x0203 dehumidification
0x0002 Device Temperature
0x0010 binary_output
0x0011 binary_value
0x0012 multistate_input
0x0013 multistate_output

There also missing most Substance Measured that can be used for measurement in air quality testers and drinking water testers or pool/spa/hottub testers (where you currently only have 0x040d Carbon Dioxide + 0x042a = PM2.5 measurement):

0x040c Carbon Monoxide (CO) 
0x040e Ethylene (CH2) 
0x040f Ethylene Oxide (C2H4O) 
0x0410 Hydrogen (H) 
0x0411 Hydrogen Sulfide (H2S) 
0x0412 Nitric Oxide (NO) 
0x0413 Nitrogen Dioxide (NO2) 
0x0414 Oxygen (O2) 
0x0415 Ozone (O3)
0x0416 Sulfur Dioxide (SO2) 
0x0417 Dissolved Oxygen (DO) 
0x0418 Bromate
0x0419 Chloramines
0x041a Chlorine
0x041b Fecal coliform & E. Coli
0x041c86 Fluoride
0x041d Haloacetic Acids
0x041e Total Trihalomethanes
0x041f Total Coliform Bacteria
0x0420 Turbidity
0x0421 Copper
0x0422 Lead
0x0423 Manganese
0x0424 Sulfate
0x0425 Bromodichloromethane
0x0426 Bromoform
0x0427 Chlorodibromomethane
0x0428 Chloroform
0x0429 Sodium
0x042B Formaldehyde

Then there is all the Smart Energy Clusters which are used in energy meters (where you currently only have 0x0702 = Metering) though understand that not all are possible as for example Sub-GHz require a such radio:

0x0700 Price
0x0701
0x0702 Metering
0x0703 Messaging
0x0704 Tunneling
0x0705 Prepayment
0x0707 Calendar
0x0708 Device Management
0x0709 Events
0x070B Sub-GHz
0x0800 Key Establishment
0x0b01 Meter Identification

PS: Best might be for you to go through and update the whole cluster list to match the new Zigbee Cluster Library Specification R8 (Revision 8) and add all missing ZCL Clusters?

https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf

### Describe the solution you'd like.

For highest flexibility would preferably see you add support for all Zigbee Cluster identifiers in Zigbee Cluster Library Specification R8 (Revision 8) but would be happy to see you start by adding as many of those examples as possible to begin with.

chshu, 2024-01-08:
@Hedda Thank you for the feature request. We are actively enhancing the ESP Zigbee SDK with additional features.

Rather than incorporating all the missing features at once, we aim to prioritize and support the features essential to real customer projects. Please don't hesitate to reach out to our business team at https://www.espressif.com/en/contact-us/technical-inquiries. Share details about your project schedule and priority, allowing us to better address your specific needs.

chshu, 2024-07-26:
Closing this issue, please file separate issues for any missing clusters that are required for your projects.

closed 2024-07-26 by chshu