espressif/esp-zigbee-sdk #206: Zigbee2MQTT using the Zigbee gateway example.

djorr5, 2024-01-08:
### Question

I would like to know if it would be possible to create an example of the Zigbee Gateway using the thread router board talking to Home Assistant. The use case is to control a Zigbee GPO (power point) across a WAN from HA. I currently have something setup using the HA Switch example on the C6 to control a commercial GPO and also using a S3 as a Zigbee2MQTT with a low level interface to the C6 board for on/off control. However I was wondering if it could be achieved with the Thread router board and the Gateway example talking to Home Assistant. 

### Additional context.

It doesn't neceessarily have to be the Gateway example either. Could even be the Zigbee2MQTT on the Thread router board talking over UART to the H2 to control the GPO. But the current problem I have is that the state is currently not being reported back when I control on or off to the GPO.

chshu, 2024-01-08:
AFAIK, the [zigpy](https://github.com/zigpy/zigpy) is required to work with [ZHA](https://www.home-assistant.io/integrations/zha/), we will have a ESP NCP solution with corresponding zigpy-xx to work with home assistant.

zigpy doesn't work on the ESP Thread BR board, but instead you need to implement the protocol between the gateway and home assistant (which was done in zigpy) on your own. Which I think should be possible, if you understand the detailed information of Home Assistant cloud interface.

chshu, 2024-03-08:
@djorr5 Let us know if you have any follow up questions on this topic.

closed 2024-03-08 by chshu