espressif/esp-zigbee-sdk #207: Support for device type classes for both HA profile and ZLL profile out-of-the-box

Hedda, 2024-01-08:
### Is your feature request related to a problem?

ESP-Zigbee-SDK currently do not define support for all device types in the Zigbee Home Automation (ZHA / Zigbee HA) profile nor in the Zigbee Light Link (ZLL) profile, so wondering if possible to make a generic bulk feature request that asks you to add all or at least most HA and ZLL device type classes out-of-the-box?

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html#ha-automation-device-types

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/ha/esp_zigbee_ha_standard.html

Most if not all of the holes of missing HA and ZLL device type classes are however already supported by popular Zigbee gateway implementations, such as the relatively popular Home Assistant's built-in ZHA integration which is based on the zigpy library and is more than likely going to be used as a testbed by most developers for testing Zigbee compatibility:

https://github.com/zigpy/zigpy/blob/dev/zigpy/profiles/zha.py

https://github.com/zigpy/zigpy/blob/dev/zigpy/profiles/zll.py

Example of HA device types for Zigbee HA (Zigbee Home Automation) profiles supported in zigpy but not supported yet in ESP-Zigbee-SDK:

# Generic

    LEVEL_CONTROL_SWITCH = 0x0001
    LEVEL_CONTROLLABLE_OUTPUT = 0x0003
    SCENE_SELECTOR = 0x0004
    REMOTE_CONTROL = 0x0006
    COMBINED_INTERFACE = 0x0007
    RANGE_EXTENDER = 0x0008
    SIMPLE_SENSOR = 0x000C
    CONSUMPTION_AWARENESS_DEVICE = 0x000D
    HOME_GATEWAY = 0x0050
    SMART_PLUG = 0x0051
    WHITE_GOODS = 0x0052
    METER_INTERFACE = 0x0053
	
# Lighting

    ON_OFF_LIGHT = 0x0100
    DIMMABLE_LIGHT = 0x0101
    COLOR_DIMMABLE_LIGHT = 0x0102
    ON_OFF_LIGHT_SWITCH = 0x0103
    DIMMER_SWITCH = 0x0104
    COLOR_DIMMER_SWITCH = 0x0105
    LIGHT_SENSOR = 0x0106
    OCCUPANCY_SENSOR = 0x0107

# ZLO device types

    ON_OFF_BALLAST = 0x0108
    DIMMABLE_BALLAST = 0x0109
    ON_OFF_PLUG_IN_UNIT = 0x010A
    DIMMABLE_PLUG_IN_UNIT = 0x010B
    COLOR_TEMPERATURE_LIGHT = 0x010C
    EXTENDED_COLOR_LIGHT = 0x010D
    LIGHT_LEVEL_SENSOR = 0x010E

# Closure

    WINDOW_COVERING_DEVICE = 0x0202
    WINDOW_COVERING_CONTROLLER = 0x0203

# HVAC

    HEATING_COOLING_UNIT = 0x0300
    PUMP = 0x0303
    PUMP_CONTROLLER = 0x0304
    PRESSURE_SENSOR = 0x0305
    FLOW_SENSOR = 0x0306
    MINI_SPLIT_AC = 0x0307

# Intruder Alarm Systems

    IAS_CONTROL = 0x0400  # IAS Control and Indicating Equipment
    IAS_ANCILLARY_CONTROL = 0x0401  # IAS Ancillary Control Equipment
    IAS_ZONE = 0x0402
    IAS_WARNING_DEVICE = 0x0403

# ZLO device types, continued

    COLOR_CONTROLLER = 0x0800
    COLOR_SCENE_CONTROLLER = 0x0810
    NON_COLOR_CONTROLLER = 0x0820
    NON_COLOR_SCENE_CONTROLLER = 0x0830
    CONTROL_BRIDGE = 0x0840
    ON_OFF_SENSOR = 0x0850
	
	
# ZLL device types (some of which overlap with HA device types but with "light" instead of "switch")

    ON_OFF_LIGHT = 0x0000
    ON_OFF_PLUGIN_UNIT = 0x0010
    DIMMABLE_LIGHT = 0x0100
    DIMMABLE_PLUGIN_UNIT = 0x0110
    COLOR_LIGHT = 0x0200
    EXTENDED_COLOR_LIGHT = 0x0210
    COLOR_TEMPERATURE_LIGHT = 0x0220
    COLOR_CONTROLLER = 0x0800
    COLOR_SCENE_CONTROLLER = 0x0810
    CONTROLLER = 0x0820
    SCENE_CONTROLLER = 0x0830
    CONTROL_BRIDGE = 0x0840
    ON_OFF_SENSOR = 0x0850

PS: Off-topic but FYI, you also need to fully support Zigbee Green Power Power (GPP) to make Zigbee 3.0 compliant devices that can be certified, so at least all ZGP Proxy device types needs to be supported as well:

https://github.com/zigpy/zigpy/blob/dev/zigpy/profiles/zgp.py

chshu, 2024-01-08:
Link https://github.com/espressif/esp-zigbee-sdk/issues/205#issuecomment-1880811875

chshu, 2024-07-26:
Closing this issue, please file separate issues for any particular device types are required for your projects.

closed 2024-07-26 by chshu