espressif/esp-zigbee-sdk #208: Add support for ZLL (Zigbee Light Link) application profile and its device classes

Hedda, 2024-01-08:
### Is your feature request related to a problem?

Generic request to please also add full support for ZLL (Zigbee Light Link) profile (PROFILE_ID = 49246) and all of its device classes:

The main difference between ZLL (Zigbee Light Link) profile and ZHA (Zigbee Home Automation) profile ZLL that it specifially define devices as "lights" instead "switches" for some devices classes that otherwise match the same device class ID as ZHA.

ZLL does however have some lighting related extensions that is not included in Zigbee HA, which is by the way why Philips/Signify primarly uses the Zigbee Light Link profile in its Philips Hue series products and it is what their Philips Hue Bridge supports directly.

In theory, using the ZLL profile ID will make it easier to make a Hue compatible device that can be paired with Philips Hue Bridge:

https://developers.meethue.com/zigbee-3-0-support-in-hue-ecosystem/

https://www.makeuseof.com/other-smart-bulbs-work-with-philips-hue/

So real world use case is that you can only ever join a Zigbee Light Link device directly to the official Philips Hue Bridge v1 and v2.

https://www.philips-hue.com/en-us/explore-hue/works-with

For reference, compare device classes for ZLL verses ZHA supported in zigpy library used in Home Assistant's Zigbee integration:

https://github.com/zigpy/zigpy/blob/dev/zigpy/profiles/zll.py

https://github.com/zigpy/zigpy/blob/dev/zigpy/profiles/zha.py

```
class DeviceType(t.enum16):
    ON_OFF_LIGHT = 0x0000
    ON_OFF_PLUGIN_UNIT = 0x0010
    DIMMABLE_LIGHT = 0x0100
    DIMMABLE_PLUGIN_UNIT = 0x0110
    COLOR_LIGHT = 0x0200
    EXTENDED_COLOR_LIGHT = 0x0210
    COLOR_TEMPERATURE_LIGHT = 0x0220

    COLOR_CONTROLLER = 0x0800
    COLOR_SCENE_CONTROLLER = 0x0810
    CONTROLLER = 0x0820
    SCENE_CONTROLLER = 0x0830
    CONTROL_BRIDGE = 0x0840
    ON_OFF_SENSOR = 0x0850
```

PS: ZLL was originally part of Zigbee Lighting & Occupancy Device Specification, so it also covers scenarios for Occupancy sensors.

### Describe the solution you'd like.

Be able to make ZLL (Zigbee Light Link) profile compatible devices that can be paired directly to the official Philips Hue Bridge.

chshu, 2024-01-08:
Link: https://github.com/espressif/esp-zigbee-sdk/issues/205#issuecomment-1880811875

chshu, 2024-03-08:
@Hedda The touchlink examples are supported now:
https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink.

The ZLL seems an out of date spec, which was replaced by the touchlink in Zigbee 3.0. Do you think the touchlink feature is enough to cover your request?

Bosemani, 2024-03-19:
Hai @Hedda 
Did you tried to add device to Philips Hue?  Using touchlink feature how to add to hue gateway?

xieqinan, 2024-04-02:
@Bosemani 

Please refer to https://github.com/espressif/esp-zigbee-sdk/issues/168#issuecomment-1952370582

chshu, 2024-07-26:
touchlink feature is supported: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink

closed 2024-07-26 by chshu