espressif/esp-zigbee-sdk #210: Error when requesting basic cluster attribute on a ZLL network

spark404, 2024-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1288-g5524b692ee

### esp-zigbee-lib version.

1.0.7

### esp-zboss-lib version.

1.0.7

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

ESP32-C6 device sends a Read Attribute Response for the requested attribute.

### What is the actual behavior?

The packet is ignored:

```
I (5715) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: 12:e9:46:98:08:21:e0:7e, PAN ID: 0x8d8e, Channel:15)
I (5925) ZBOSS: zcl/zcl_common.c:585   >>get_endpoint_desc 0x40816434 0x4081f6ac ep_id 11
I (5925) ZBOSS: zcl/zcl_common.c:590   device_ctx 0x4081f6ac, ep_count 1
I (5925) ZBOSS: zcl/zcl_common.c:592   ep_id 11
I (5935) ZBOSS: zcl/zcl_common.c:602   <<get_endpoint_desc desc 0x4081f584
I (5935) ZBOSS: zcl/zcl_common.c:1053   > zb_zcl_is_target_endpoint: ep_desc 0x4081f584, profile_id 0x104
I (5945) ZBOSS: zcl/zcl_common.c:1108   ep profile 0xc05e doesn't match cmd profile 0x104!
```

### Steps to reproduce.

This is a bit hard to reproduce. I'm connecting my ESP32-C6 to a ZLL network (Philis HUE). Joining the network is not a problem, however once the simple descriptor is read the bridge starts requesting attribute values. That results in the error message above.

I believe the error results from the implementation specifics for ZLL. As described in the following snippet: 
```
The ZLL profile ID is 0xC05E. However, for the purposes of interoperability, ZCL commands addressing clusters other than ZLL com- missioning or those commands in the commissioning utility command set of the ZLL commissioning cluster, use the ZHA profile identifi- er, 0x0104. The ZLL commissioning cluster has a cluster ID of 0x1000.
```
Source: https://www.silabs.com/documents/public/user-guides/ug103-09-fundamentals-zll.pdf, section 3.1

### More Information.

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_VERBOSE, ESP_ZB_TRACE_SUBSYSTEM_ZCL );

    esp_zb_init(&zb_nwk_cfg);
    esp_zb_enable_joining_to_distributed(true);
    esp_zb_secur_TC_standard_distributed_key_set(zll_hue_trustcenter_key);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    // ON/OFF Device supports Basic / Identify / Groups / Scenes / On/Off / Level Control
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_level_cluster_cfg_t level_cfg = {
            .current_level = 0x0,
    };
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_basic_cluster_create(&light_cfg.basic_cfg);
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_identify_cluster_create(&light_cfg.identify_cfg);
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_groups_cluster_create(&light_cfg.groups_cfg);
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(&light_cfg.scenes_cfg);
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&light_cfg.on_off_cfg);
    esp_zb_attribute_list_t *esp_zb_level_cluster = esp_zb_level_cluster_create(&level_cfg);

    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(cluster_list, esp_zb_level_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(esp_zb_ep_list, cluster_list, 11, ESP_ZB_AF_ZLL_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
    ESP_ERROR_CHECK(esp_zb_device_register(esp_zb_ep_list));

    ESP_ERROR_CHECK(esp_zb_start(false));

    // This never returns
    esp_zb_main_loop_iteration();

    ESP_LOGW("esp_zb_task", "stopped");
}
```

xieqinan, 2024-01-23:
@spark404 ,

I believe the error is originating from the line `esp_zb_ep_list_add_ep(esp_zb_ep_list, cluster_list, 11, ESP_ZB_AF_ZLL_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);`. As mentioned in the snippet, "The ZLL commissioning cluster has a cluster ID of 0x1000." You can refer to the [touchlink](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_touchlink/touchlink_switch/main/esp_touchlink_switch.c#L188) example in the esp-zigbee-sdk for guidance.

xieqinan, 2024-04-02:
@spark404，

The esp-zigbee-sdk has been updated to v1.2.2. You can test it again. Please feel free to discuss any questions regarding the esp-zigbee-sdk. If the issue has been resolved, please close it.

closed 2024-07-26 by chshu