espressif/esp-zigbee-sdk #211: Add support for particulate matter cluster PM1.0, PM4.0 and PM10.

deviashin, 2024-01-09:
### Is your feature request related to a problem?

Hello!
Should we expect other types of particulate matter other than esp_zigbee_zcl_pm2_5_measurement.h?
My project uses Honeywell's HPMA115C0-004 sensor which provides data such as PM1.0, PM2.5, PM4.0 and PM10.

xieqinan, 2024-02-06:
@deviashin ,

The terms "PM1.0, PM2.5, PM4.0, and PM10" do not appear to be defined in the specification. If necessary, you can use a custom cluster to support these measurements.

deviashin, 2024-02-12:
I should have paid attention to this before making the request. I agree, there are no such clusters in the ZCL standard. Same as AQI. Well, let's hope they add it someday.

closed 2024-02-12 by deviashin